TheImagingSource/tiscamera #24: General output Pin in Linux

peqc1300, 2015-02-27:
I would like to know in what way, if possible, I can control the General output Pin of the board monochrome camera? Programatically if possible.

TIS-Stefan, 2015-03-02:
It is possible. You need the UVC extensions for accessing them.

peqc1300, 2015-03-02:
I do have one question. Is there really any difference between the trigger, privacy and trigger Global reset flags? It seems they do the same thing.

TIS-Stefan, 2015-03-03:
I must admit, I do not know, what "privacy" is. I only see it being provided by the V4L2 driver. It is not a property of the camera.

The "Global Reset", better "Global Reset Shutter mode" is a special trigger mode, which depends on the used sensor. The DFK DMK 72 has it. Usually the sensor runs as rolling shutter sensor. That means, the sensor is exposed line by line in sequence. If you have enabled the trigger mode and then enabled the "Global Shutter Reset mode", then all lines of the sensor a light sensitive at the same point of time, like a global shutter sensor. Then the sensor continues to expose line by line. In the situation, where you use trigger and Global Shutter Reset Mode and you have enough light, the image will become brighter to the bottom, because the lower the line, the longer is is light sensitive.

Global Reset Shutter mode runs in combination with enabled trigger only. Global Reset Shutter Mode and trigger mode are different properties.

peqc1300, 2015-03-05:
I understand now. I have used the privacy flag thinking it was the trigger. The odd thing is. It does. If I send a pulse to the trigger input with privacy flag on it will take the picture. Thanks I will be using the trigger mode you provided.

closed 2015-03-05 by peqc1300