TheImagingSource/tiscamera #25: Errors loading USB2 camera XML description file

morefigs, 2015-03-24:
When calling `sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0` the following output is generated:

	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980920 (V4L2_CID_USER_BASE + 32)
	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980924 (V4L2_CID_USER_BASE + 36)
	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980926 (V4L2_CID_USER_BASE + 38)
	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980927 (V4L2_CID_USER_BASE + 39)
	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980928 (V4L2_CID_USER_BASE + 40)
	[libwebcam] Unknown V4L2 user control ID encountered: 0x00980930 (V4L2_CID_USER_BASE + 48)
	Importing dynamic controls from file tisEUVC.xml.
	tisEUVC.xml: error: video0: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
	tisEUVC.xml: error: video0: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
	tisEUVC.xml: error: video0: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
	tisEUVC.xml: error: video0: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
	tisEUVC.xml: error: video0: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
	tisEUVC.xml: error: video0: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
	tisEUVC.xml: error: video0: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
	tisEUVC.xml: error: video0: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
	tisEUVC.xml: error: video0: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
	tisEUVC.xml: error: video0: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
	tisEUVC.xml: error: video0: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)

When attempting to use this camera with `guvcview` the program's trigger checkbox is ignored by the camera, hence cannot use triggering with this camera.

The camera is DFK 22BUC03 running on Raspberry Pi B+. Calling `v4l2-ctl --all` shows that the camera supports `trigger (bool)`.

morefigs, 2015-03-24:
Issue seems to have been solved by changing firmware from `dmk22uc03_135.euvc` to `dmk22uc03_163_uvc.euvc`.

How can I tell which is the best firmware to use? This website says to use `dmk22uc03_135.euvc` for Dxx 22UC cameras, but that didn't work so well. There are also firmware files for D**F**K 22UC03 cameras (like mine), wouldn't that be better suited?

TIS-Stefan, 2015-03-24:
Hello

I usually use the firmware with highest number.

aidanlandsberg, 2015-05-06:
Hi,

I too have experienced this exact same issue. Upon calling `<sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0>` I obtain the following error (note i require to use the DFM 22BUC03-ML in trigger mode):

 `<[libwebcam] Unknown V4L2 user control ID encountered: 0x00980924 (V4L2_CID_USER_BASE + 36)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980926 (V4L2_CID_USER_BASE + 38)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980927 (V4L2_CID_USER_BASE + 39)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980928 (V4L2_CID_USER_BASE + 40)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980930 (V4L2_CID_USER_BASE + 48)
Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml: error: video1: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
aidan@aidan-iMac:~/Documents/Scripsie/Camera/tiscamera/data/uvc-extensions$ sudo uvcdynctrl -i tisEUVC.xml -d /dev/video1
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980924 (V4L2_CID_USER_BASE + 36)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980926 (V4L2_CID_USER_BASE + 38)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980927 (V4L2_CID_USER_BASE + 39)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980928 (V4L2_CID_USER_BASE + 40)
[libwebcam] Unknown V4L2 user control ID encountered: 0x00980930 (V4L2_CID_USER_BASE + 48)
Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml: error: video1: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)
tisEUVC.xml: error: video1: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 17: File exists)>`

I am not at all familiar with these devices and any advice will be greatly appreciated. Updating the firmware as mentioned above didn't resolve the issue.

morefigs, 2015-05-06:
Which firmware are you using? And OS version, etc?

aidanlandsberg, 2015-05-08:
I am using the dmk22uc03_135.euvc on Ubuntu 14.04. Upon loading the dmk22uc03_163_uvc.euvc firmware, it seems that the xml file is indeed loaded with the Trigger and Trigger Global Reset Shutter variables being mapped. Other mapping errors still exist though. Is this normal?

TIS-Stefan, 2015-05-08:
You are right, you need the dmk22uc03_163_uvc.euvc  firmware and must install the UVC Extension files. Global Reset shutter is not available for this camera, because it is already a global shutter camera.
As far as the properties work, ignore the error messages

aidanlandsberg, 2015-05-08:
All right, thanks so much for the support. Much appreciated

closed 2015-05-22 by TIS-Stefan