TheImagingSource/tiscamera #26: Does DFK21BU04 camera support trigger and exporsure function?

catrinne, 2015-04-08:
hello?

i'm trying to make an application which take a snap shot triggered by external signal.

i'm using a respberry pi and DFK21BU04.

trigger and exporsure function are required.

bellow shows the functions that it support.

however there are no trigger function and exporsure function.

are there no way to make my application except buying a new camera?

pi@raspberrypi ~/opencv $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : DFx 21BU04
        Bus info      : usb-bcm2708_usb-1.2
        Driver version: 3.18.7
        Capabilities  : 0x84000001
                Video Capture
                Streaming
                Device Capabilities
        Device Caps   : 0x04000001
                Video Capture
                Streaming
Priority: 2
Video input : 0 (Input 4: ok)
Format Video Capture:
        Width/Height  : 640/480
        Pixel Format  : 'GREY'
        Field         : None
        Bytes per Line: 640
        Size Image    : 307200
        Colorspace    : Unknown (00000000)
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=63 step=1 default=-8193 value=0
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=128
                            hue (int)    : min=0 max=359 step=1 default=-8193 value=180
   white_balance_component_auto (bool)   : default=1 value=1
    white_balance_red_component (int)    : min=0 max=95 step=1 default=32 value=54 flags=inactive
   white_balance_blue_component (int)    : min=0 max=95 step=1 default=32 value=49 flags=inactive
                          gamma (int)    : min=1 max=500 step=1 default=57343 value=100
                           gain (int)    : min=260 max=1023 step=1 default=57343 value=260

TIS-Stefan, 2015-04-13:
Hello

First of all, you must use the extra UDEV rules for the camera, if you not already did:
https://github.com/TheImagingSource/tiscamera/wiki/UDEV-Rules

Then you need the uvc extension units:
https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units

After doing so, please see, which parameters you receive.

catrinne, 2015-04-18:
thank you for the your answer
i tried but i got a same result.

pi@raspberrypi ~/tiscamera/src $ cd tiscamera
-bash: cd: tiscamera: No such file or directory
pi@raspberrypi ~/tiscamera/src $ sudo cp data/udev/80-theimagingsource-cameras.rules /etc/udev/rules.d/
cp: cannot stat `data/udev/80-theimagingsource-cameras.rules': No such file or directory
pi@raspberrypi ~/tiscamera/src $ sudo service udev restart
[ ok ] Stopping the hotplug events dispatcher: udevd.
[ ok ] Starting the hotplug events dispatcher: udevd.

pi@raspberrypi ~/tiscamera/src $ cd ..
pi@raspberrypi ~/tiscamera $ cd data
pi@raspberrypi ~/tiscamera/data $ ls
firmware  udev  uvc-extensions
pi@raspberrypi ~/tiscamera/data $ cd uvc-extensions/
pi@raspberrypi ~/tiscamera/data/uvc-extensions $ sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090E (V4L2_CID_USER_BASE + 14)
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090F (V4L2_CID_USER_BASE + 15)
Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml: error: video0: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video0: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
pi@raspberrypi ~/tiscamera/data/uvc-extensions $

pi@raspberrypi ~/tiscamera/data/uvc-extensions $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : DFx 21BU04
        Bus info      : usb-bcm2708_usb-1.2
        Driver version: 3.12.28
        Capabilities  : 0x84000001
                Video Capture
                Streaming
                Device Capabilities
        Device Caps   : 0x04000001
                Video Capture
                Streaming
Priority: 2
Video input : 0 (Input 4: ok)
Format Video Capture:
        Width/Height  : 640/480
        Pixel Format  : 'YUYV'
        Field         : None
        Bytes per Line: 1280
        Size Image    : 614400
        Colorspace    : Unknown (00000000)
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=63 step=1 default=-8193 value=0
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=128
                            hue (int)    : min=0 max=359 step=1 default=-8193 value=180
   white_balance_component_auto (bool)   : default=1 value=1
    white_balance_red_component (int)    : min=0 max=95 step=1 default=32 value=32 flags=inactive
   white_balance_blue_component (int)    : min=0 max=95 step=1 default=32 value=32 flags=inactive
                          gamma (int)    : min=1 max=500 step=1 default=57343 value=100
                           gain (int)    : min=260 max=1023 step=1 default=57343 value=260

TIS-Stefan, 2015-04-21:
Sorry for the delay of my answer. Did you apply the the UVC Extension? Your output does not look like that.
https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units

I did the very same on my Raspberry PI 1 and have all parameters, that I expect, such as "Trigger Mode"

catrinne, 2015-04-22:
thank you for your answer.

i think i arleady applied the UVC Extenstion.
i use usb2 camera so i execute below instruction/

pi@raspberrypi ~/tiscamera/data $ cd uvc-extensions/
pi@raspberrypi ~/tiscamera/data/uvc-extensions $ sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0

do you have same camera with me?
please let me know your camera model.

thanks.

TIS-Stefan, 2015-04-22:
I use the permanent solution on my Linux systems:
sudo mkdir /usr/share/uvcdynctrl/data/199e
sudo cp <tiscamera>/data/uvc-extensions/*.xml /usr/share/uvcdynctrl/data/199e

This works fine, at least, when Linux is restarted.

catrinne, 2015-04-28:
Sorry for the delay of my response. i update firmware to version 1027.
finally, it works. thanks.
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : DFx 21BU04
        Bus info      : usb-bcm2708_usb-1.4
        Driver version: 3.18.7
        Capabilities  : 0x84000001
                Video Capture
                Streaming
                Device Capabilities
        Device Caps   : 0x04000001
                Video Capture
                Streaming
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height  : 640/480
        Pixel Format  : 'YUYV'
        Field         : None
        Bytes per Line: 1280
        Size Image    : 614400
        Colorspace    : Unknown (00000000)
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=63 step=1 default=-8193 value=0
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=128
                            hue (int)    : min=0 max=359 step=1 default=-8193 value=180
   white_balance_component_auto (bool)   : default=1 value=1
    white_balance_red_component (int)    : min=0 max=95 step=1 default=32 value=32 flags=inactive
   white_balance_blue_component (int)    : min=0 max=95 step=1 default=32 value=32 flags=inactive
                          gamma (int)    : min=1 max=500 step=1 default=57343 value=100
                           gain (int)    : min=260 max=1023 step=1 default=57343 value=260
                  exposure_auto (menu)   : min=0 max=3 default=0 value=3
              exposure_absolute (int)    : min=1 max=300000 step=1 default=333 value=333 flags=inactive

however, trigger function is remained... --

TIS-Stefan, 2015-04-28:
Hello Catrinne

I would like to send a program sample to you, but I can not upload anything here.

Could you please create a contact at http://www.theimagingsource.com/en_US/company/contact/

The sample is gstreamer and QT4 based. It compiles on Raspberry PI.

Stefan

catrinne, 2015-05-06:
i already created contact. thanks.

closed 2015-05-22 by TIS-Stefan