TheImagingSource/tiscamera #27: Use Software trigger

Asubayo, 2015-04-21:
Hello,

I have some troubles to use aravis software trigger with long exposure time on a DMK23G445 and a Basler ace1300gm. I would like to capture single images on the both cameras with the maximum available exposure time. 

On the DMK, the maximum exposure time is 30 seconds. 
With the Basler, it's 10 seconds. 

For now, I can grab single image with 10 seconds exposure time on the basler only if I use sleep(1) before to emit a software trigger and I don't understand why I need to wait 1 second. It doesn't works with the DMK if I wait 1 second before to emit the software trigger. If I don't wait before to send software trigger, I can only grab single frames if the maximum exposure time is 5 seconds. If I set the exposure time to >5, 10, 20, 30, it doesn't work.

I am doing the following things in my program : 
- set gain, exposure, fps (the minimum for the DMK is 3.75), format ...
- create a stream with arv_camera_create_stream() and the following parameters : 
ARV_GV_STREAM_PACKET_RESEND_NEVER
ARV_GV_STREAM_SOCKET_BUFFER_AUTO
packet-timeout = 40000
frame-retention = 200000
- push a new arvbuffer in the stream with arv_stream_push_buffer()
- set acquisition more to continuous with ARV_ACQUISITION_MODE_CONTINUOUS
- configure camera to use software trigger with arv_camera_set_trigger(camera, "Software")
- start acquisition with arv_camera_start_acquisition(camera)
- sleep(1) if I use the basler camera
- send software trigger with arv_camera_software_trigger(camera)
- get the arvbuffer from the stream with arv_stream_timeout_pop_buffer(stream, exposuretimevalue (us) + 1sec)

I don't understand what's wrong and why I can't get a complete buffer from the stream.
If you are familiar with aravis or software triggers, could you give some hints please ?

Thank you very much !

TIS-Stefan, 2015-04-21:
Could you please give me a sample program source code, so I can try this on my own?

Thank you in advance!

Asubayo, 2015-04-21:
Here is quick example. I use opencv to get and display the image but the lines can be commented. 

void CameraGigeSdkAravis::grabTest(){

	ArvCamera  *cam;
	ArvPixelFormat  format;
        ArvStream *stream_;
        int  width_;
        int height_;
        double fps_;
        double gainMin_, gainMax_;
        unsigned int payload_;
        const char *pixel_format_string_;
        double exposureMin_, exposureMax_;
        const char *caps_string_;
        int  gain_;
        double exp_;
        guint64 nbCompletedBuffers_;
        guint64 nbFailures_;
        guint64  nbUnderruns_;

        namedWindow( "Display window", WINDOW_AUTOSIZE );

        /// List connected cameras
        arv_update_device_list();

		int n_devices = arv_get_n_devices();

		cout << "************** DETECTED CAMERAS *************** " << endl;
		cout << "*" << endl;

		for(int i = 0; i < n_devices; i++){

			cout << "* -> [" << i << "] " << arv_get_device_id(i) << endl;

		}

		cout << "*" << endl;
		cout << "************************************************ " << endl;

        /// Camera to use
        int cameraID = 2;

        /// Parameters
        double e = 1000000; // exposition time to use
        double g = 15; // gain to use
        bool shiftImageBits = true; // 1 for DMK, 0 for BASLER

        /// Get camera name
        string deviceName = "";

		for(int i = 0; i< n_devices; i++){

			if(cameraID == i){

				deviceName = arv_get_device_id(i);
				break;

			}
		}

		if(deviceName != ""){

            cout << "Device : " << deviceName << endl;

            cam = arv_camera_new(deviceName.c_str());

            if(cam != NULL){

                cout << "Connection success to the camera." << endl;
[... 186 lines cut ...]

                   }

                    arv_stream_get_statistics(stream_, &nbCompletedBuffers_, &nbFailures_, &nbUnderruns_);

                    cout << "Completed buffers = " << (unsigned long long) nbCompletedBuffers_	<< endl;
                    cout << "Failures          = " << (unsigned long long) nbFailures_           << endl;
                    cout << "Underruns         = " << (unsigned long long) nbUnderruns_          << endl;

                    // Stop acquisition.
                    arv_camera_stop_acquisition(cam);

                    // TriggerMode off.
                    arv_device_set_string_feature_value(arv_camera_get_device(cam), "TriggerMode" , "Off");

                    g_object_unref(stream_);
                    g_object_unref(cam);

                }

            }

        }

	}

TIS-Stefan, 2015-04-23:
Sorry for the delay of my answer, but I must admit, I am somewhat lost here. I worked with the gstreamer aravis source only, not directly with aravis. 

Therefore I would like you to create the issue on the Aravis github at https://github.com/issues

Asubayo, 2015-04-23:
Ok I will ask to the aravis project, thank you anyway.

TIS-Stefan, 2015-04-27:
Dear Ausobayo

I found time to handle your problem. 

If you use the camera in triggered mode, you should always use a stream callback as shown in the "aravis/tests/arvexample.c".

Also you should wait a time period, after you called 

arv_camera_start_acquisition(cam);		

as you did for Basler cam.

some sample code:
[code]
void StreamCallback (ArvStream *stream, ApplicationData *data)
{
	data->buffer_count++;
	cout << data->buffer_count << endl;
}

		arv_camera_set_acquisition_mode(cam, ARV_ACQUISITION_MODE_CONTINUOUS);

		arv_camera_set_trigger(cam, "Software");

		g_signal_connect (stream, "new-buffer", G_CALLBACK (StreamCallback), &data);

		arv_stream_set_emit_signals (stream, TRUE);
		
		// Start acquisition.
		arv_camera_start_acquisition(cam);		

		while( data.buffer_count < 10 )
		{
			sleep(1);
			cout << "Trigger" << endl;
			arv_camera_software_trigger(cam);
		}
		 // Stop acquisition.
		arv_camera_stop_acquisition(cam);
[/code]

In the StreamCallback function the image is received and you can handle it,

closed 2015-05-07 by TIS-Stefan