TheImagingSource/tiscamera #29: Neximage 5MP support

fmadec, 2015-05-04:
Hi,

I am trying to make a Neximage 5MP working on Raspberry PI2.
I was wondering if this camera is supported by your software, if so, which firmware should I use?

Thanks
F.

TIS-Stefan, 2015-05-05:
Hello

This is a Celestron support case. Please ask the Celestron support for this.

dmk72 ones, I would use.

fmadec, 2015-05-05:
Hi,
none of the dmk72 seems to work, the command returns nothing, no error, just nothing.
I will try to contact Celestron without hope because they unfortunately only talk about windows .

TIS-Stefan, 2015-05-05:
The firmware you want to install was not intended to be used with the Neximage series.

However, did you follow the documentation at
https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-Cameras?
The firmware should be "dmk72uc02_162_uvc.euvc". However, I am not sure, which camera your camera exactly is.

fmadec, 2015-05-05:
Yes, I understand that the firmware was not for the Celestron one, in fact during my tries to make it working I discovered that the celestron camera was a tis one, that the reason way I am trying to use your software.

I did follow your documentation, I tried first on a raspberry PI 2 with raspbian without success.
I switched to an other PC and the firmware update works but after few seconds of grabbing images (with guvcview), it's stop and says: "could not grad image (select timeout): device temporary unavailable "

By the way my camera is the Clestron neximage 5 MP:
CMOS sensor (1/2.5") Micron MTP9P031
2.2 microns pixel size
2592x1944 pixels

Thanks for trying to help me.

TIS-Stefan, 2015-05-05:
I did not test on a Raspberry PI2 on my own. However, try lower frame rates.

closed 2015-05-07 by TIS-Stefan

fmadec, 2015-05-07:
Hi,
sorry for the late answer, but a lower frame rates simply gives more time before the freeze....

TIS-Stefan, 2015-05-07:
No problem being late, but I have no context. I do not know, whether you run on Linux or Windows

fmadec, 2015-05-07:
I an trying to make it running on linux

TIS-Stefan, 2015-05-07:
Did you try a lower frame rate?

fmadec, 2015-05-07:
Yes I tried without sucess