TheImagingSource/tiscamera #30: Missing resolutions for DMM 42BUC03-ML

tommises, 2015-05-07:
I flashed the DMM 42BUC03-ML camera with the following firmware: dmk42uc03_143.euvc 
Is it the right one?

The product page [1] lists a few different resolutions, but some of them are missing from the camera firmware. The missing resolutions are 1280x720 and 1024x768. 
[1] http://www.theimagingsource.com/en_US/products/oem-cameras/usb-cmos-mono/dmm42buc03ml/

:~$  v4l2-ctl -d 7 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'GREY'
        Name        : Greyscale (8-bit)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 720x480
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)

Thanks,

Tom

TIS-Stefan, 2015-05-08:
You are right, these resolutions are not available with the current firmware. I have no deadline for a next release.

closed 2015-05-08 by TIS-Stefan

tommises, 2015-05-24:
I have just tried to use 720x480 and 640x480 and neither of them work. Depending on the selected frame rate, either no data is sent or the sent frames are corrupted.

I must say this is quite disappointing, the product page specifies a number of different resolutions and I ended with only one.