TheImagingSource/tiscamera #32: Incorrect exposure control in DMM 42BUC03-ML

tommises, 2015-05-16:
Camera: DMM 42BUC03-ML
Firmware: dmk42uc03_143.euvc

The V4L2 specification gives 1/10000 sec as a base unit for absolute exposure.
See here (V4L2_CID_EXPOSURE_ABSOLUTE): 
http://www.linuxtv.org/downloads/legacy/video4linux/API/V4L2_API/spec-single/v4l2.html

It seems it is not implemented this way in the camera. Below are brightness values of an object measured for different exposure settings.
[exposure_absolute] [measured brightness]  
1   50
2   68
3   100
4   120
5   150
6   171
7   200
8   225

So 6/10000 (->171) should give twice the brightness of 3/10000 (->100), but this is not the case. It makes more sense, if 1 is in fact 2, 2 is 3, and so forth. Then:
1 -> 2 -> 2/10000 -> 50
3 -> 4 -> 4/10000 -> 100

Is that how it works? In that case, it is not possible to set 1/10000 sec which is the shortest exposure time specified for this camera. My application requires such short exposure times, so I would like to clarify and know what exposure time I am actually using, and I would also like to be able to set 1/10000 sec.

Neoo, 2015-05-22:
Does the element tiswhitebalance is available on Gstreamer-1.0 now ?

TIS-Stefan, 2015-05-22:
No,

TIS-Stefan, 2016-11-23:
Now the answer is "Yes", because we updated the tiscamera repository.

closed 2016-11-23 by TIS-Stefan