TheImagingSource/tiscamera #33: Compatibility With GStreamer-1.0

Neoo, 2015-05-22:
Does the element tiswhitebalance is available with Gstreamer-1.0

TIS-Stefan, 2015-05-22:
Nom currently not.

Neoo, 2015-05-25:
Thank you a lot.
What can I do for the gsttiswhitebalance to work with Gstreamer-1.0 ?

TIS-Stefan, 2015-05-26:
>What can I do for the gsttiswhitebalance to get it to work with Gstreamer-1.0 ?
Well, you have the source code of the 0.10 version, therefore you only need to enhance the GStreamer interface functions to 1.0.  Since I did not work with 1.0 on my own before, I must admit, I do not know, what to do. But I guess, the low level on byte functions doing the white balance, wont change.

tkaiso, 2016-08-29:
any progress since then ?   does TIS has the planning date to fix this issue ?  gst-0.1 is getting out of date and obsolete from my point of view. 
I ran the following command and got this error, 
$ sudo  GST_DEBUG=totem:4,typefind:5 GST_PLUGIN_PATH=/usr/local/TCAM_INSTALL_GST_1_0 gst-launch-1.0 v4l2src ! video/x-bayer,width=1280,height=720 ! tcamwhitebalance ! queue ! bayer2rgb ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-1.0:19749): GLib-GObject-CRITICAL **: g_object_get: assertion 'G_IS_OBJECT (object)' failed
Caught SIGSEGV
#0  0x00007f5bf34c3e8d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f5bf39f939c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f5bf39f9722 in g_main_loop_run ()
#3  0x00007f5bf3f548e9 in gst_bus_poll ()
#4  0x00000000004046f8 in ?? ()
#5  0x0000000000403808 in ?? ()
#6  0x00007f5bf33e9830 in __libc_start_main (main=0x403130, argc=14, 
#7  0x0000000000403d19 in ?? ()
Spinning.  Please run 'gdb gst-launch-1.0 19749' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:04.342455867
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

TIS-Stefan, 2016-08-29:
You can try the beta-release branch. It contains GStreamer 1.0 modules.

tkaiso, 2016-08-29:
Yes, I did.  this was the result with next.   My cam is DFK 23UX249.   thanks. 
$ git branch -a
* beta-release
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/beta-release
  remotes/origin/documentation-2.0
  remotes/origin/master

tkaiso, 2016-08-29:
I forgot to say,  My HEAD is almost the latest as shown below. 
$ git log
commit bf76267a0975ec53e9c8ecd085b09dba6750e3fe
Author: Edgar Thier <edgar.thier@theimagingsource.com>
Date:   Mon Jun 13 08:46:04 2016 +0200

    Enable videoconvert as final bin element

TIS-Stefan, 2016-08-29:
Well, I have to admit, I do not have much more on my own currently.

tkaiso, 2016-08-30:
when do you plan to release this beta version as production version ? 
DFK 23UX249 is your product and this problem is getting critical from user's stand point. someone working for TIS needs to fix it.  I hope you to put high priority on this issue.  otherwise, it is inevitable for us to give up using it. 

Q1) haven't you reproduced this problem at your site ?   (I think this can be reproduced at any site, but I am not sure.)

If we have time,  we might be able to try to dig into this problem to fix it, but to do that, we need more information with this code from TIS.   can we talk to the developer of this code ?    

thanks

TIS-Stefan, 2016-08-30:
I must admit, currently I have to timing plan for the free software you downloaded from the beta-release branch.  
Why do you need to use GStreamer 1.0? 

And yes, I have some trouble too, but not a segmentation fault. I simply have no image, but I do not know why. I am not the programmer of these modules.

TIS-Edgar, 2016-11-16:
Support for Gstreamer-1.0 has been released.

closed 2016-11-16 by TIS-Edgar