TheImagingSource/tiscamera #34: Motion blur for short exposure times in DMM 42BUC03-ML

tommises, 2015-05-26:
Unfortunately, another problem with this camera, this time a critical one.

We are experiencing a strange problem with light apparently "leaking" into the pixels, which results in kind of motion blur appearing in the case of short exposure times.

Please refer to this dataset: https://drive.google.com/open?id=0B1XDXh7oI3etM08zN1VPdDNOUms

There are three datasets there, each for different exposure time: 1/10000sec, 1/5000sec and 1/500sec. There is also a marked up image.

The camera was being rotated from side to side during the capturing. For 1/10000sec and 1/5000sec, there is motion blur visible on both sides of the circular pattern. The pattern is well exposed, with the exposure consistent with the rest of the frame, while the motion blur is just a ghost. For 1/500sec, this phenomenon is not visible.

It was tested with both the Windows and Linux firmware. Resolution was 1280x960 @15fps.

For reference, we also tried a smartphone camera, which used 1/3000sec and did not show any such problems.

I have the following theory for the mechanism how this happens. The light "leaks" into the pixels outside the desired exposure time. The blur smudge is quite long, so the corresponding time is in the same range as the frame rate (1/15sec). For short exposures, it results in the appearance of the ghost. For longer exposures, the difference in amount of light between the "leaking" and the actual exposure is larger and the ghost does not appear.

There is also another strange behavior, it does not seem to be related to the problem above, but maybe it is an indication of some general problems with the sensor. In some areas of the image, there are dark pixel rows every other row. It is marked up in the image in the dataset.

As the camera can experience such quick movements in our application and short exposure times are necessary, this problem renders the camera inadequate for our application at this time. I would appreciate a quick response, as we will need to make a decision whether there is a resolution here or we need to move on and try another product, which would be disappointing. 

Thanks,

Tom

TIS-Stefan, 2015-05-26:
Hello Tom

What you see is an effect of Shutter Efficiency of Global shutter sensors. The sensor manufacturer describes this effect in following PDF:
https://www.aptina.com/products/technology/Aptina_Global-Shutter-WhitePaper.pdf

So using a longer exposure time, between 1/500 and 1/1000 will give a better result.

Another good description of this phenomena can be found at:
https://www.cvl.isy.liu.se/education/undergraduate/tsbb09/lasmaterial/Amtel-Journal-Issue6.pdf

A rolling shutter camera wont have such an effect, but it has the "curtain" effect on horizontal movements.

BR
Stefan

tommises, 2015-05-27:
Hi Stefan,

Thanks - I had no idea about shutter efficiency.

After familiarizing myself with the linked documents, I realized increasing frame rate would help, which I confirmed by comparing 7fps with 25fps. In that case, we could use a lower resolution with higher frame rate. It is unfortunate that the lower resolutions in DMM 42BUC03-ML are broken, as I reported in the recent comment to Issue #30. Any chance this will get fixed?

Alternatively, we could order DMM 22BUC03-ML, which has smaller resolution.
* I would like to know which resolutions and frame rates are **actually** available with the Linux firmware for this camera.
* Is the shutter efficiency in this lower-resolution sensor similar? As far as I can tell, this sensor is an older design, so I would not like to end up changing the camera, just to learn that despite a faster readout time, there is no improvement due to lower shutter speed efficiency.

Thanks,

Tom

TIS-Stefan, 2015-05-28:
Tom

Sorry for the delay of my answer. These are the video formats and frame rates avaialble in Linux of the DxK 22:

	Format 0 :  Greyscale 8-bit (Y800)  1
		744x480
			60/1
			30/1
			25/1
			15/1
		640x480
			60/1
			30/1
			25/1
			15/1
		320x240
			141/1
			120/1
			80/1
			60/1
			30/1
			25/1
			15/1

I also used a very short exposure time and 60 fps frame rate. I did not see the issue in my test.

Stefan

tommises, 2015-05-30:
Stefan - Thanks for the resolutions and for doing the test.

closed 2015-06-01 by TIS-Stefan