TheImagingSource/tiscamera #35: Sample code in c/cpp for USB2.0 in LINUX?

dsr10june, 2015-05-29:
Could you please tell me where is the sample code for Linux operating system using a 2.0 camera.
OR How can I modify the sample code of GIGE cam so that it works with usb2.0 also.

TIS-Stefan, 2015-05-29:
Hello

There is no specific sample code, because you can get all of them in the V4L2 and GStreamer documentation.

dsr10june, 2015-05-30:
Thanks for the reply.

Is there any added advantage of using V4L2/GStreamer over OPENCV? (I guess Opencv make use of v4l2)?

TIS-Stefan, 2015-06-01:
The advantage is in using our auto exposure and auto white balance GStreamer moduls. 
Also the OpenCV implementation for video seems to handle RGB24 video formats only. Therefore, you wont get usable images using OpenCV only.

JoseSacristan, 2015-06-08:
HI Stefan,

are you telling to us, that we can't take usable images with the opencv::VideoCapture?

similar to (these easy opencv code works with my webcam):

cv::VideoCapture cap(0); 
cv:Mat frame;
cap >> frame;
cv::imwrite("./shot.jpg",frame);

TIS-Stefan, 2015-06-08:
Yes. This is what I know, because OpenCV does not handle the grayscale video format of the camera. As far as I know, you can see this in the OpenCV source code too. Please correct me, if I am wrong here.

JoseSacristan, 2015-06-29:
HI Stefan,

Finally I have my USB 3.0 camera (DFK 23UM021) working in OpenCV. Sometimes the biggest problems have easy solutions:

cv::VideoCapture cap(0);
cv:Mat frame,  image;
cap >> frame;
cv::Mat raw(frame.rows, frame.cols, CV_8UC1, frame.data);
cv::cvtColor(raw, image, CV_BayerGB2BGR);
cv::imwrite("./shot.jpg", image);

TIS-Stefan, 2015-06-30:
Hi Jose

That looks well.

Stefan

closed 2015-06-30 by TIS-Stefan