TheImagingSource/tiscamera #36: USB Color CMOS DFK AFU050-L.34: Device not found!

emepetres, 2015-06-15:
Hi! I'm trying to install a DFK AFU050-L.34 under Manjaro x64 or Debian Jessie x86.

When I plugged it in the device is detected (the following lines are from dmesg):
[ 1314.428152] usb 1-3: new high-speed USB device number 4 using ehci-pci
[ 1316.105359] usb 1-3: New USB device found, idVendor=199e, idProduct=8209
[ 1316.105376] usb 1-3: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[ 1316.105388] usb 1-3: Product: DFK AFU050-L34
[ 1316.105401] usb 1-3: Manufacturer: The Imaging Source Europe GmbH
[ 1316.105412] usb 1-3: SerialNumber: 21510555

I have successfully compiled and in installed code from tiscamera/src/ and followed instructions from the wiki (https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-Cameras)
But the command
$ sudo ./euvccam-fw -p
failed with the error message
Device not found.

Also /dev/video0 doesn't exist.

As this is a CMOS camera I understand I don't need the udev rules, anyway I have tried without success.

What I'm missing? Thanks for your help!

TIS-Stefan, 2015-06-17:
Hello

>What I'm missing? Thanks for your help!

An USB 3.0 controller. The camera will not work on USB 2.0

>$ sudo ./euvccam-fw -p

Is, as documented, for USB 2.0 cameras only. You have an USB 3.0 camera.

emepetres, 2015-06-17:
The camera is USB 2.0 -> http://www.theimagingsource.com/en_US/products/integrated-optics-cameras/usb-cmos-color/dfkafu050l34/

Anyway If anyone is interested in the solution, I had response from ImageSource. In their own words: 

"“Please download attachments from: ftp://tis_ftp:Imaging199@ftp.theimagingsource.com/Imaging/13d193/libafu050.tgz

the DFK AFU050-L34 is a little special. It does not deliver a UVC-compliant data stream. Therefore, it must be accessed with a software library specific for that type of camera. You can download that library from the FTP link above.

This library will be integrated into the software package at github/tiscamera at a later point in time, but unfortunately is has not been done yet”

TIS-Stefan, 2015-06-17:
outch you are right. I mixed it with the DFK AFU130. Sorry for that. The DFK AFU050-L34 is not UVC compatible. For this an extra library is needed. Please create a request at 
http://www.theimagingsource.com/en_US/company/contact/

I will answer this tomorrow.

emepetres, 2015-06-19:
Thank you for your answers Stefan, I left the request and now I am waiting the response :-)

TIS-Stefan, 2015-06-19:
I have sent the lib files to you.

micdobro, 2015-08-17:
I have the same camera. Contacted TIS as you suggested, although a total silence in all docs is a bit of a deal-breaker. Can the lib itself be put into git? It is a brilliant little camera. Also - is there anything that can prevent me from getting the camera working on a RPi? 
As for other things - I will be happy to contribute back the doc on getting this little devil up and running.

TIS-Stefan, 2015-08-18:
I answered by email and sent the libafo050 sources to you.

closed 2015-08-18 by TIS-Stefan