TheImagingSource/tiscamera #38: Forcing DHCP state with camera-ip-conf

cmarmo, 2015-07-24:
Hello,

I'm having problems with the DHCP configuration of a DMK23G445. The camera is connected to a switch, the switch is connected to a ADSL box. No subnets are configured, I'm expecting that when the network goes up after a cut-off (the network is not very reliable... :( ) the camera will ask the box DNS for an IP address in the correct range (192.168.X.X). This is not the case: the camera obtain a strange address (169.254.X.X DHCP server is not found?).
camera-ip-conf-cli info tell me that DHCP is off and Current Gateway is 0.0.0.0
Using camera-ip-conf-cli allowed me to force dhcp and gateway only if the IP address is specified.
For the project I'm involved in, soon, the number of cameras I will manage will increase and fixing the IP address for each camera avoiding superposition with local network is not easy in an automatic way.
Do you think it could be possible (and interesting for you) to just force the gateway and DHCP use with camera-ip-conf-cli?
Thanks for listening,
Chiara

TIS-Stefan, 2015-07-24:
Hi

169.254 is an auto link local IP address, that is used, if no DHCP server is found but a network device is set to DHCP. That is TCP/IP standard. Therefore, I guess, either your DHCP does not speak with the camera or the camera does not find your DHCP server. 

Maybe it is a configuration problem of your ADSL box? (Another question: Does it have GigE? If not, you wont get images from the camera, so the box is the completely wrong device to use. But I see the point to have internet and camera at the same point of time.)
You may check, whether there is a device or MAC restriction in your box.

Forcing DHCP wont work, because the camera already is on DHCP.

Better is using a GigE Switch.

cmarmo, 2015-07-24:
Hi,
We are using GigE switch.
Yes, this is probably a problem on the box side: I'm trying to compare consistently the box configurations, but as the problem is linked to remote network or power cut-off I'm not completely in control....
That's why I'm looking for a tool forcing the camera to retry the connection with a specific server...
Thanks for your explanations!
Chiara

TIS-Stefan, 2015-07-24:
Hello Chiara

I must understand, I do not really the problem. In my camera network I have a DHCP server, that runs on my computer. If a camera is connected to this network, it automatically gets an new IP address

Same with company network. 

Also: Never use two DHCP servers in one network.

cmarmo, 2015-07-24:
Thanks for your answers and suggestions!
I will work on them.
Best regards,
Chiara

TIS-Stefan, 2015-07-24:
Just a hint: If the switch and the camera are powered at the same point of time, the camera uses the auto link local IP address, instead of waiting for the switch booting. Maybe this is happens on your side.

closed 2015-07-24 by TIS-Stefan

cmarmo, 2015-07-24:
Yep, I will try to re-arrange the connection scheme.
Thanks! :)
Chiara