TheImagingSource/tiscamera #39: Cannot capture images in OpenCV (DMK 42BUC03)

ccrb, 2015-08-10:
Hello all, 

I have a DMK 42BUC03 camera and I want to grab images using OpenCV in Linux. 
I used this sample code on my Lubuntu14.04 (https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e), and 
$make 
these files.

But it returns:

/tmp/ccS7nNEK.o: In function `ShowImage(char*, _IplImage*, int)':
/home/ccrb/Downloads/test/main.cpp:10: undefined reference to `cvNamedWindow'
/home/ccrb/Downloads/test/main.cpp:11: undefined reference to `cvShowImage'
/home/ccrb/Downloads/test/main.cpp:12: undefined reference to `cvWaitKey'
/home/ccrb/Downloads/test/main.cpp:13: undefined reference to `cvDestroyWindow'
/tmp/ccS7nNEK.o: In function `main':
/home/ccrb/Downloads/test/main.cpp:31: undefined reference to `cvNamedWindow'
/home/ccrb/Downloads/test/main.cpp:37: undefined reference to `cvCreateImage'
/home/ccrb/Downloads/test/main.cpp:38: undefined reference to `cvCreateImage'
/home/ccrb/Downloads/test/main.cpp:61: undefined reference to `cvCvtColor'
/home/ccrb/Downloads/test/main.cpp:63: undefined reference to `cvShowImage'
/home/ccrb/Downloads/test/main.cpp:64: undefined reference to `cvWaitKey'
/home/ccrb/Downloads/test/main.cpp:73: undefined reference to `cvDestroyWindow'
collect2: error: ld returned 1 exit status
make: *** [opencv-test] Error 1

on my terminal...
Can anyone tell me how to solve this question?

I want capture images with my Lubuntu and can use opencv libraries(by C/C++) to process images...

thanks all,
ccrb

TIS-Stefan, 2015-08-10:
Is is possible, the OpenCV dev libs are not installed on your computer?

ccrb, 2015-08-10:
@TIS-Stefan 
I have done this page(http://milq.github.io/install-opencv-ubuntu-debian/), and it can works by sample code.
Should I also do (https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-Cameras) on my new PC?
I have already done on another PC. And it still can't $make on my new PC.
 
Thank you so much, Stefan.

ccrb, 2015-08-10:
Hello all,

I change in my Makefile :
	$(CXX) -I /usr/local/include/opencv2 -Wall -g -O0 -o opencv-test main.cpp v4ldevice.cpp $(shell pkg-config opencv --libs libv4l2)
and it can make and capture successful!

closed 2015-08-10 by TIS-Stefan