TheImagingSource/tiscamera #41: Not complete frames with v4l2 and opencv

Asubayo, 2015-08-13:
Hello,

I have a DMK Camera (DMx 31AU03.AS) that I would to control under linux (Debian). I have tried your code example https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e but I don't understand why the grabbed frames are not complete. I have tried to change the FPS value but it changes nothing to the result. Here is what I always get : 

[image]

Do you have an idea of the problem ?

Thank you very much !

labreu, 2015-08-13:
may I see your main code? maybe I can help

Asubayo, 2015-08-13:
Hello labreu,

My main code is the same as the example : 

int main ()
{

   	IplImage* pOpenCVImage;

    CvSize ImageSize;
    unsigned char* ImageBuffer = NULL;
    int wKey = -1;

    ImageSize.width = 1024;
    ImageSize.height = 768;

    cvNamedWindow( (char*)"Camera", 1 );

    printf("Program started\n");

    pOpenCVImage = cvCreateImage(ImageSize , IPL_DEPTH_8U, 1 ); // Grayscale
    
    open_device((char*)"/dev/video0");
    init_device(ImageSize.width, ImageSize.height);

    printf("Start capturing\n");

    start_capturing();

    while(wKey == -1 )
    {
        ImageBuffer = snapFrame();

        if( ImageBuffer != NULL )
        {

			memcpy( pOpenCVImage->imageData, ImageBuffer, pOpenCVImage->imageSize);

            cvShowImage( (char*)"Camera",pOpenCVImage);

            wKey = cvWaitKey(10);
			
        }
        else
        {
            printf("No image buffer retrieved.\n");
            break;
        }
    }

    cvDestroyWindow( (char*)"Camera" );
    stop_capturing();
    uninit_device();
    close_device();

    printf("Program ended\n");

    return 0;

}

labreu, 2015-08-13:
try this

if( ImageBuffer != NULL )
{
             pOpenCVImage->imageData = (char*)ImageBuffer;
             currentImg = Mat(pOpenCVImage);    //convert IplImage to Mat type
             imshow("Window name", currentImg);
             wKey = cvWaitKey(10);
}

Asubayo, 2015-08-13:
I have the same problem with this modification.

TIS-Stefan, 2015-08-14:
Hello

Please step into the funcition
void init_mmap (void)

and evaluate the value of "buf.length" at:
 buffers[n_buffers].length = buf.length;

The returned value should be 1024*768 = 786432.

Asubayo, 2015-08-14:
Hello Stefan,

Yes the value of buf.length is correct. 
I get the value 786432

TIS-Stefan, 2015-08-14:
do you see a complete image in guvcview?

Asubayo, 2015-08-17:
I have some troubles with guvcview. I have installed the last version 2.0 but there was a segmentation fault when I launched it. I have now installed the version 1.5.3 but I am not sure it works properly, I can adjust the properties of the camera but I just get black images.

Asubayo, 2015-08-17:
I have launched guvcview with a different capture method (-r option with value 2) and I get frames  like : 
[image]

Asubayo, 2015-08-17:
Ok, if I changed the fps value to 15, I get complete frames  : 

[image]

TIS-Stefan, 2015-08-17:
Is is possible, the camera is connected to an USB 3.0 controller? In this case, the image data transfer is interrupted. I never saw this on Linux, but on Windows. I did not keep in mind, the Windows driver drops these incomplete frames, therefore, I thought, simply no image was provided at all. Sorry for that.

So : on USB 3.0 controllers, your camera provides half frame rate only. It is in the hardware of the camera and can not be changed. On USB 2.0 controllers, the camera will run with full frame rate.

closed 2015-08-17 by TIS-Stefan