TheImagingSource/tiscamera #43: How to force change RGB to GRAY?

ccrb, 2015-08-16:
Hello all,
    I have a DMK 42BUC03 camera and I want to change its RGB data to GRAY.

 cvCvtColor(pOpenCVImage,pColorCVImage,CV_BayerGB2RGB); // Create a color image from the raw data 

Can I just change parameter in this function or I should grab picture first and change RGB to GRAY..
Hope I can get GRAY pics to process as real time.

TIS-Stefan, 2015-08-17:
You do not need
cvCvtColor(pOpenCVImage,pColorCVImage,CV_BayerGB2RGB); //

this line, because it creates color RGB from the incoming bayer format. However, only color cameras have bayer raw data. Your monochrome camera has no Bayer Raw, therefore you receive 1byte per pixel gray scale.
This should answer your question.

closed 2015-08-17 by TIS-Stefan

ccrb, 2015-08-17:
But if I remove this line, I just can get one whole black image.
And if I want do threshold from that RGB image, I can't get real threshold image.

cvThreshold(pColorCVImage,pColorCVImage,128,255,CV_THRESH_BINARY);

[image]

I tried to add some code about RGB to GRAY and do threshold again, but it waste some speed.

[image]