TheImagingSource/tiscamera #45: @TIS-Stefan Do you have some C sample code Not c++ to get video stream?

gonewith, 2015-08-27:
@TIS-Stefan
I am using DFM 72BUC02-ML in ubuntu 14.04. I find here from google
I have asked technical supporter  to update the firmware with dmk72uc02_162_uvc.euvct.
Now I can get color pictures using the code you posted before at here: https://gist.github.com/10f04501f49b6b3bf75e but there is a problem: the color is not correct,for example, orange become blue ... I think Maybe I need change some parameters to adjust it in the code, so It 's not a big problem?

I have another important question: Do you have some C sample code Not c++ to get video stream? if you have not ,Do I need write it from V4L2 for my company embedded device(not need GUI to display the Live image,only get the data to memory  or save the picture to jpg files)?

gonewith, 2015-08-27:
Hi @TIS-Stefan 
I find the your code here: fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_GREY;the camera output format  is 'Greyscale 8-bit (Y800)  is it right?
I use the following code I found on internet, it directly create two .jpg file using the memory data got from the camera. but I can open the file with the any picture viewer to see it.

So How can I generate .jpg file I can see it for demo ? do I need use some tools(or functions) to transfer GREY format to jpg. I can not directly save memory data to jpg files?

please have a look the following code:

//http://bbs.chinaunix.net/thread-4120327-1-1.html

#include <stdio.h>
#include <string.h>
#include <errno.h>
#include <stdlib.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <assert.h>
#include <linux/videodev2.h>

typedef struct
{
        void *start;
        int length;
}BUFTYPE;

BUFTYPE *user_buf;
int n_buffer = 0;

//打开摄像头设备
int open_camer_device()
{
        int fd;

        if((fd = open("/dev/video0",O_RDWR | O_NONBLOCK)) < 0)
        {
                perror("Fail to open");
                exit(EXIT_FAILURE);
        }
        
        return fd;
}

int init_mmap(int fd)
{
        int i = 0;
        struct v4l2_requestbuffers reqbuf;

        bzero(&reqbuf,sizeof(reqbuf));
        reqbuf.count = 4;
        reqbuf.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
        reqbuf.memory = V4L2_MEMORY_MMAP;
        
        //申请视频缓冲区(这个缓冲区位于内核空间，需要通过mmap映射)
        //这一步操作可能会修改reqbuf.count的值，修改为实际成功申请缓冲区个数
        if(-1 == ioctl(fd,VIDIOC_REQBUFS,&reqbuf))
        {
                perror("Fail to ioctl 'VIDIOC_REQBUFS'");
                exit(EXIT_FAILURE);
        }
        
        n_buffer = reqbuf.count;
        
        printf("n_buffer = %d\n",n_buffer);

        user_buf = calloc(reqbuf.count,sizeof(*user_buf));
        if(user_buf == NULL){
                fprintf(stderr,"Out of memory\n");
                exit(EXIT_FAILURE);
[... 271 lines cut ...]
        }

        return;
}

int main()
{
        int fd;       
                         
        fd = open_camer_device();

        init_camer_device(fd);
        
        start_capturing(fd);
        
        mainloop(fd);
        
        stop_capturing(fd);

        uninit_camer_device(fd);

        close_camer_device(fd);

        return 0;
}

TIS-Stefan, 2015-09-15:
Hi

Sorry for the delay of my answer. I have an C and OpenCV sample code, that does the same, your source code does. I would like to send it to you, therefore, you may contact me at http://www.theimagingsource.com/en_US/company/contact/

The video format we use is " V4L2_PIX_FMT_GREY". It is set in the init_device() function.

```C
    CLEAR(fmt);

    fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    if (force_format)
    {
        fmt.fmt.pix.width       = width;
        fmt.fmt.pix.height      = height;
        fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_GREY;
        fmt.fmt.pix.field       = V4L2_FIELD_NONE;

        if (-1 == xioctl(fd, VIDIOC_S_FMT, &fmt))
            errno_exit("VIDIOC_S_FMT");

        /* Note VIDIOC_S_FMT may change width and height. */
    }
    else
    {
        /* Preserve original settings as set by v4l2-ctl for example */
        if (-1 == xioctl(fd, VIDIOC_G_FMT, &fmt))
            errno_exit("VIDIOC_G_FMT");
    }
```

The debayering is done using OpenCV functions.
```C
cvCvtColor(pOpenCVImage,pColorCVImage,CV_BayerGB2RGB); // Create a color image from the 
```

The `CV_BayerGB2RGB` and related defines specify the bayer pattern

Stefan

closed 2015-09-15 by TIS-Stefan

gonewith, 2015-09-15:
Hi Stefan,

I got it.

Thank you very much.

Regard.

Jin(Peter)Zhang