TheImagingSource/tiscamera #46: While I connected two more same camera, /dev/video* lost!!

ccrb, 2015-09-05:
Hello all,

I have several DMK 42BUC03 camera and I want to grab images by Linux.
I used this sample code on my Lubuntu14.04 (https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e), 
and everything is fine by using only one camera.
But two question occurred.

Today I want try two same camera and I can find video0&1 at first.
But after I restart my PC, I can't find any /dev/video* !!!
How can I solve this question?

And my second question is: 
How I can get capture by using two more same camera(42BUC03).
I tried to change code in main.cpp (open_device((char*)"/dev/video0");), but it can't get two camera capture at the same time before I lost my /dev/video*
Can anyone tell me how can I get several capture by same camera?

thanks all,
-ccrb

ccrb, 2015-09-05:
Here's information by $./euvccam-fw -p

Manufacturer String: The Imaging Source Europe GmbH
Product String: DMK 42BUC03
Serial Number String: 24514192
PID: 8208
Flags: 0
EEProm size: 32768

I still can't get capture from my device now...
Even I create a node to use,

$sudo mknod /dev/video0 c 81 0
$sudo chmod 666 /dev/video0

Can anyone tell me how can I solve this question?
I have reinstall my PC(ubuntu14.04.3) and still can't get video* on /dev ......

thanks all,
ccrb

ccrb, 2015-09-05:
It seems happened this question after I git new clone on github.
https://github.com/TheImagingSource/tiscamera/tree/master/tools/euvccam-fw
It changed yesterday and I used this new ./euvccam-fw -m for my camera.
It can't change mode to uvc mode, and can't let Flags be 1.
I don't know if this is problem or not...

Please help me to get capture.

thanks

TIS-Edgar, 2015-09-07:
Hi,

Let's start with getting your devices working properly.

What does the command 'dmesg' say when you connect a camera?

Ideally it would be something like:

    [ 3634.269253] usb 3-14: new high-speed USB device number 5 using xhci_hcd
    [ 3635.996841] usb 3-14: config index 0 descriptor too short (expected 392, got 308)
    [ 3635.997125] usb 3-14: New USB device found, idVendor=199e, idProduct=8302
    [ 3635.997127] usb 3-14: New USB device strings: Mfr=2, Product=1, SerialNumber=3
    [ 3635.997128] usb 3-14: Product: DMK 22BUC03
    [ 3635.997129] usb 3-14: Manufacturer: The Imaging Source Europe GmbH
    [ 3635.997130] usb 3-14: SerialNumber: 21110463
    [ 3635.997497] uvcvideo: Found UVC 1.00 device DMK 22BUC03 (199e:8302)
    [ 3636.932874] input: DMK 22BUC03 as /devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14:1.0/input/input18

What does it say for your camera?

ccrb, 2015-09-07:
Hi Edgar, 

Here it is.

[  217.648252] usb 1-2: USB disconnect, device number 8
[  224.156623] usb 1-2: new high-speed USB device number 9 using xhci_hcd
[  224.641966] usb 1-2: New USB device found, idVendor=199e, idProduct=8208
[  224.641975] usb 1-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  224.641979] usb 1-2: Product: DMK 42BUC03
[  224.641982] usb 1-2: Manufacturer: The Imaging Source Europe GmbH
[  224.641986] usb 1-2: SerialNumber: 24514192

TIS-Edgar, 2015-09-08:
Ok,

The problem is that your cameras do not register as uvc devices.

You need to follow the steps described in [our wiki](https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras).

Simply write  dmk42uc03_143.euvc to your camera.

After a reconnect the line

    [ 3635.997497] uvcvideo: Found UVC 1.00 device DMK 42BUC03 (199e:8308)

should appear. Now you have  a /dev/video* node that allows camera interaction.

Please tell me if there where any problems.

Best regards,

Edgar

ccrb, 2015-09-08:
I know that steps for camera.

But my problem is still can't get UVC device.

sudo ./euvccam-fw -u ../../data/firmware/usb2/dmk42uc03_143.euvc 
firmware basename: 'files/dfk42uc02_[0-9]*.euvc'
->using file: '../../data/firmware/usb2/dmk42uc03_143.euvc'
->backup file: 'backups/backup_1.euvc'
Upload firmware version: 143
Creating backup file...
Current config: Vendor = 'The Imaging Source Europe GmbH', Product = 'DMK 42BUC03', Serial = '24514192'
Uploading image file...
0k written
1k written
2k written
3k written
4k written
5k written
6k written
7k written
8k written
9k written
10k written
11k written
12k written
13k written
14k written
15k written
Updating strings...
!!!IMPORTANT!!!
Please save this application and the backup files for future reference!
A backup of the old firmware was created here: backups/backup_1.euvc
To return to the original firmware, use the following command:
./euvccam-fw -u backups/backup_1.euvc

After done this and reconnect camera, 
$dmesg still return:

[  210.081848] usb 1-2: USB disconnect, device number 8
[  214.505534] usb 1-2: new high-speed USB device number 9 using xhci_hcd
[  214.991490] usb 1-2: New USB device found, idVendor=199e, idProduct=8208
[  214.991497] usb 1-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[  214.991501] usb 1-2: Product: DMK 42BUC03
[  214.991504] usb 1-2: Manufacturer: The Imaging Source Europe GmbH
[  214.991506] usb 1-2: SerialNumber: 24514192

And I can't switch to UVC mode:

Mode setting option is removed from this version. 
This setting is not required for latest firmware files

And Flags still be "0" after I done this tips.(It seems will be "1" before...)

How can I do for next...
 
Thank you, Edgar.

All the best,

ccrb

TIS-Edgar, 2015-09-08:
It seems like our firmware developer forgot to publish a firmware file where the uvc flag is set by default. I am terribly sorry for the inconvenience.
You can use <tiscamera>/tools/firmware-update to also interact with usb2 cameras.

    firmware-update -d 24514192 -m u

Should set the uvc flag as expected. I will inform our firmware developer of this problem.

Best regards,

Edgar

ccrb, 2015-09-08:
Ahh, so nice for hear that.

It can switch to UVC mode now!
>Product String: DMK 42BUC03
Serial Number String: 24514192
PID: 8308
Flags: 1
EEProm size: 32768

Thanks a lot, Edgar!!

Btw, How can I use several same camera on the same time?

It seems just need to change device's number but I don't know where should change.

Thank you for your response very much.

All the best,
ccrb

TIS-Edgar, 2015-09-08:
That is good to hear!

To use multiple cameras simultaneously you need to open all wanted device nodes.
 That basically means you need to change the device [in this line](https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e#file-main-cpp-L42).
To identify which camera has which device node you should use udev.
[Here is a gist that shows you how to iterate the devices.](https://gist.github.com/TIS-Edgar/7d9cfc346a36d6062d36) 

Best Regards,

Edgar

ccrb, 2015-09-08:
Hello Edgar,

Thank you so much for help me solve my camera UVC mode problem and hint about get my device node by udev.

Now, I know how to find my device automatically.
I"ll try to show several camera frame by use that sample v4l2 code tomorrow.

Thank you very much for your time.

Wish you all the best,

ccrb

TIS-Edgar, 2015-09-08:
No problem.
If you have any other problems, please let us know.

Best regards,

Edgar

closed 2015-09-08 by TIS-Edgar