TheImagingSource/tiscamera #47: VIDIOC_DQBUF error 5, Input/output error

ccrb, 2015-09-09:
Hello, all

I tried just change 
`open_device((char*)"/dev/video1");`
and make two different opencv-test output to use.
But while I use one named "opencv-test1"(video0), and open another terminal to execute "opencv-test2"(video1), one of them will shut down and return 
>VIDIOC_DQBUF error 5, Input/output error

How can I get two frames easily and can be used for process.....

thanks a lot

All the best,

ccrb

TIS-Edgar, 2015-09-10:
Hi,

Please excuse the late reply.

To make sure I understand your setup correctly:

You took [this gist](https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e) and simply changed the device node so that you have a executable for each camera.

On what hardware are you working?
My first instinct would be that one camera interferes with the data transfer of the other.

Could you attach both cameras and post the output of 'lsusb -t'? 

Best regards,

Edgar

ccrb, 2015-09-10:
Hello Edgar,

Thanks for your help again!

Here's the output on terminal after I type 'lsusb -t' 
>/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 3: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 1: Dev 9, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 1: Dev 9, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 2: Dev 8, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 2: Dev 8, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/3p, 12M
        |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
        |__ Port 3: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M
        |__ Port 3: Dev 6, If 2, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 3: Dev 6, If 3, Class=Application Specific Interface, Driver=, 12M
    |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 3, If 2, Class=Human Interface Device, Driver=bcm5974, 12M

My hardware now is: macbookpro 13" (use ubuntu14.04.3 for test), 
and I"ll use same code on my target board(Cubieboard4, also used ubuntu 14.04 OS).
Do you this hardware can connect more than three "42BUC03" camera and used for image process?
Each frame I use is 1280x200(ROI by OpenCV).

I"ll come back after three hours.
Thank you.

All the best,
ccrb

TIS-Edgar, 2015-09-10:
They are on the same bus, but the data volume should still be manageable.
What framerate are you using? It should make no difference, but still. [You can use this function to set it.](https://gist.github.com/TIS-Edgar/e2bd764e5594897d1835)

One possibility is that your laptop does not supply enough power for both cameras.
There are a few things you can try.
Disable autosuspend.

    sudo sh -c 'echo -1 >/sys/module/usbcore/parameters/autosuspend'

If that does not work you should try a usb hub with power supply. That will eliminate power problems as a possible error source.

If the problem then still exists we will have to dig deeper.

    sudo sh -c 'echo 0xffffffff > /sys/module/uvcvideo/parameters/trace'

This command will enable debug output for the uvc driver. You can the read the messages with dmesg. A copy of the output would be helpful.

Best regards,

Edgar

ccrb, 2015-09-11:
I tried `sudo sh -c 'echo -1 >/sys/module/usbcore/parameters/autosuspend'`, and it can use two camera on my Macbook now.
But on my target board, it doesn't have autosuspend in this folder.
It returns 
>sh: 1: cannot create /sys/module/usbcore/parameters/autosuspend: Directory nonexistent

Does this mean I can't disable autosuspend on my embedded system?
I also not find autosuspend in   
>/sys/bus/usb/devices/usb*/power

I have done this. `sudo sh -c 'echo 0xffffffff > /sys/module/uvcvideo/parameters/trace'`
But it's not help for me.
Maybe I should buy one usb hub to connect all my cam on one usb port?

I'm tried to solve this, do you have any suggest?

All the best,
ccrb

TIS-Edgar, 2015-09-11:
Hi,

If the suspend command does not work, you stll have the option of setting kernel parameters.
This will set the options once at boot time and is less flexible. For your embedded system it may still be worth a try.
To do this edit/create the file
/etc/modprobe.d/usb-autosuspend.conf

and append the line:

    options usbcore autosuspend=-1

After that a reboot should set the option. If that does not work I would have to have a closer look to at your embedded system and its configuration.

A usb hub with external power supply should also do the trick.

Best regards,

Edgar

ccrb, 2015-09-11:
Hello, 

I tried to create "usb-autosuspend.conf", and reboot my embedded system.

And it works! But sometimes it may fail.

I just bought a usb 2.0 hub one hour ago, and while I connect my cam, it can create video node!
But while I use it on v4l2sample code. It happened error and return `error 5, Input/output error` again!!
Seems I bought a not good usb-hub.......

Btw, can I just use two camera by one executable file?(It seems I should match each device in `snapFrame()`)
And I haven't use framerate function, but I can get each camera frame, should I set framerate to use?

Thank you very much for your help!! 

All the best,
ccrb

TIS-Edgar, 2015-09-14:
Hi,

You said it sometimes fails. Is there something different in those cases? Does dmesg say something? Did you reboot between them?

A possibility we are looking into is that it has something to do with the processor going into deep idle
states when it is not supposed to.

To check whether this assumption is correct, please try the procedure described at
https://access.redhat.com/articles/65410 (citing):

The first is a big hammer approach. By booting with the kernel command line argument
processor.max_cstate=1 and idle=poll the system will never enter a C-state other than zero and will not even use the MWAIT mechanism to temporarily halt in the idle routine. While this will provide the fastest scheduler response time, it is very wasteful of power and will generate heat that requires the air conditioning system to run (and waste more power).  This method is not recommended for general use, but can be quite handy for testing whether your application is being affected by cstates and idle transition latency. Booting your kernel and adding the commandline options:

    processor.max_cstate=1 idle=poll

If that is the case we can find solutions to only disable the cstate when the cameras are running.

Concerning your second question:
Yes, you can open multiple devices in one executable.
The only thing that differentiates between the devices is the file descriptor, thus there simply switching between them at runtime is all you need to do.

The framerate is important when dealing with exposure times. The higher the framerate the shorter your highest possible exposure time will be. This depends on your use case, so there is no general rule for this. However explicitly setting it will ensure that your camera always uses the same settings (even after reboots, etc.) so it will probably prevent some bad surprises.

Best regards,

Edgar

TIS-Stefan, 2015-09-15:
Hi ccrb

I would like to know, which camera models you use. Knowing this, we can create two simple Gstreamer pipelines and try to run them in parallel. If this works, we can continue with OpenCV.
 
Thank you in advance.

Stefan

ccrb, 2015-09-18:
Hello Edgar,

Sorry for the late reply.

Thanks for all your helps on my problem.

For now, I can get images on each camera on their own execute.
I"ll try to make them in one execute and set camera's frame rate next week! This two week I need to finish another work.

And I have one question, due to the camera I used is 42BUC-series, it can accept trigger signal.
Cause I haven't set framerate before, which way is better for my case? I"ll use 24fps for camera cause I have use gige-cam to detect my object before.

I'm very appreciated for all your support!

All the best, 

ccrb

ccrb, 2015-09-18:
Hello Stefan, 

Sorry for the late reply.

I"ll use more than 7 "42BUC03" to detect my object, and use image processing by OpenCV.

Thanks for all your help.

All the best,

ccrb

closed 2015-09-22 by TIS-Stefan

ccrb, 2015-09-25:
Hello Edger,

I found why I still have this problem today.

I found two of my camera "Serial Number String" are same!!!

How can I reset my camera serial number if i have lost my backup file.

All the best,

ccrb