TheImagingSource/tiscamera #48: How to change camera serial number?

ccrb, 2015-09-28:
Hello all,

I found why I still have VIDIOC_DQBUF error 5 today.

I found two of my camera "Serial Number String" are same!!!

How can I reset my camera serial number if i have lost my backup file.

All the best,

ccrb

TIS-Stefan, 2015-09-28:
Hi

The serial number in a camera can not be changed. Also our production is not able to write the same serial number into different devices. However, that is, how I expected it working. From this, I think, there is another problem.
Please check, whether the Linux shows the same serial numbers for both cameras, if only one of them is connected, not both together.

ccrb, 2015-09-28:
Hello Stefan,

I got serial number on terminal from both of them print:
 
>Product String: DMK 42BUC03
Serial Number String: 24514192

It's wired but actually happened on two of my video device...

All the best,

ccrb

closed 2016-11-16 by TIS-Edgar