TheImagingSource/tiscamera #49: DFK AFU050-L34 in Ubuntu 14.04

simonfm86, 2015-10-19:
I want to install the cameras DFK AFU050-L34 in Ubuntu 14.04, when I connect I get the message:

[924.771161] usb 3-6: new high-speed USB device using xhci_hcd number 8
[926.335959] usb 3-6: New USB device found, idVendor = 199E, idProduct = 8209
[926.335962] usb 3-6: New USB device strings: Mfr = 2, Product = 1, SerialNumber = 3
[926.335963] usb 3-6: Product: DFK AFU050-L34
[926.335965] usb 3-6: Manufacturer: The Imaging Source Europe GmbH
[926.335965] usb 3-6: SerialNumber: 20,510,272

I have the udev rules and UVC extension units defined.

When I try to detect the camera:
<b>sudo -p ./uvc cam-fw</b>

I get the message:
<b>Device not found.</b>

What's happening? Thanks for your help!

retornaz, 2015-11-02:
Hi
I got the same issue and euvccam-fw say also Device not found.
How to put this camera in "UVC" mode ?

TIS-Stefan, 2015-11-03:
The DFK AFU050-L34 is not UVC compatible and therefore not in the list of the affected cameras for the euvcfwcam-fw tool
For using this camera, you need an extra library, which is currently not on the git hub. Please create a request at http://www.theimagingsource.com/en_US/company/contact/ and will send the code to you.

I ask my programmers to add the libafu050 to the git hub here.

closed 2015-11-03 by TIS-Stefan