TheImagingSource/tiscamera #51: Unable to create GST pipeline

retornaz, 2015-11-02:
Hello

I'm using a DFM 72BUC02-ML camera in UVC mode:
usb 1-3: New USB device found, idVendor=199e, idProduct=8307
usb 1-3: New USB device strings: Mfr=2, Product=1, SerialNumber=3
usb 1-3: Product: DFM 72BUC02-ML
usb 1-3: Manufacturer: The Imaging Source Europe GmbH
usb 1-3: SerialNumber: 44510125
uvcvideo: Found UVC 1.00 device DFM 72BUC02-ML (199e:8307)

And launch gstreamer with the following pipeline:
gst-launch-0.10 v4l2src device=/dev/video1 ! video/x-raw-gray,width=1280,height=720,framerate=30/1 ! tiscolorize ! tiswhitebalance ! tis_auto_exposure ! bayer2rgb ! ffmpegcolorspace ! queue ! xvimagesink

But if fail with the following error:
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Could not negotiate format

Here is the full debug log:
0:00:00.000134142 13849      0x234fa00 INFO                GST_INIT gstquery.c:107:_gst_query_initialize: init queries
0:00:00.000511876 13849      0x234fa00 INFO                GST_INIT gstmessage.c:73:_gst_message_initialize: init messages
0:00:00.000584003 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:350:_gst_plugin_initialize: registering 0 static plugins
0:00:00.000631393 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:255:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.000636480 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:257:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.000644344 13849      0x234fa00 INFO            GST_REGISTRY gstregistry.c:1648:ensure_current_registry: reading registry cache: /home/philippe/.gstreamer-0.10/registry.x86_64.bin
0:00:00.005089819 13849      0x234fa00 INFO            GST_REGISTRY gstregistrybinary.c:614:gst_registry_binary_read_cache: loaded /home/philippe/.gstreamer-0.10/registry.x86_64.bin in 0.004438 seconds
0:00:00.005114383 13849      0x234fa00 INFO            GST_REGISTRY gstregistry.c:1518:scan_and_update_registry: Validating plugins from registry cache: /home/philippe/.gstreamer-0.10/registry.x86_64.bin
0:00:00.005833651 13849      0x234fa00 INFO            GST_REGISTRY gstregistry.c:1610:scan_and_update_registry: Registry cache has not changed
0:00:00.005841785 13849      0x234fa00 INFO            GST_REGISTRY gstregistry.c:1683:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.005845117 13849      0x234fa00 INFO                GST_INIT gst.c:805:init_post: GLib runtime version: 2.42.2
0:00:00.005847780 13849      0x234fa00 INFO                GST_INIT gst.c:807:init_post: GLib headers version: 2.42.1
0:00:00.005863251 13849      0x234fa00 INFO            GST_PIPELINE gstparse.c:334:gst_parse_launch_full: parsing pipeline description 'v4l2src device=/dev/video1 ! video/x-raw-gray,width=1280,height=720,framerate=30/1 ! tiscolorize ! tiswhitebalance ! tis_auto_exposure ! bayer2rgb ! ffmpegcolorspace ! queue ! xvimagesink '
0:00:00.006642121 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/usr/lib64/gstreamer-0.10/libgstvideo4linux2.so" loaded
0:00:00.006652224 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "v4l2src"
0:00:00.006834394 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseSrc@0x24900e0> adding pad 'src'
0:00:00.007387253 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/home/philippe/.gstreamer-0.10/plugins/libgsttiscolorize.so" loaded
0:00:00.007394426 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "tiscolorize"
0:00:00.007456151 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24980f0> adding pad 'sink'
0:00:00.007466118 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24980f0> adding pad 'src'
0:00:00.008522360 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/home/philippe/.gstreamer-0.10/plugins/libgsttiswhitebalance.so" loaded
0:00:00.008529138 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "tiswhitebalance"
0:00:00.008584996 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x249e0b0> adding pad 'sink'
0:00:00.008594707 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x249e0b0> adding pad 'src'
0:00:00.008700517 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/home/philippe/.gstreamer-0.10/plugins/libgsttis_auto_exposure.so" loaded
0:00:00.008708159 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "tis_auto_exposure"
0:00:00.008753274 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24a20a0> adding pad 'sink'
0:00:00.008761897 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24a20a0> adding pad 'src'
0:00:00.008863510 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/usr/lib64/gstreamer-0.10/libgstbayer.so" loaded
0:00:00.008869141 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "bayer2rgb"
0:00:00.008966688 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24983f0> adding pad 'sink'
0:00:00.008979114 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24983f0> adding pad 'src'
0:00:00.009090742 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/usr/lib64/gstreamer-0.10/libgstffmpegcolorspace.so" loaded
0:00:00.009096647 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "ffmpegcolorspace"
0:00:00.009345582 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24a2420> adding pad 'sink'
0:00:00.009355570 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24a2420> adding pad 'src'
0:00:00.009577758 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/usr/lib64/gstreamer-0.10/libgstcoreelements.so" loaded
0:00:00.009584118 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "queue"
0:00:00.009642331 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstQueue@0x24b8090> adding pad 'sink'
0:00:00.009657211 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstQueue@0x24b8090> adding pad 'src'
0:00:00.010429479 13849      0x234fa00 INFO      GST_PLUGIN_LOADING gstplugin.c:859:gst_plugin_load_file: plugin "/usr/lib64/gstreamer-0.10/libgstxvimagesink.so" loaded
0:00:00.010444557 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "xvimagesink"
0:00:00.010546496 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseSink@0x24c05f0> adding pad 'sink'
0:00:00.010570501 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "pipeline"
0:00:00.010604831 13849      0x234fa00 INFO            GST_PIPELINE ./grammar.y:567:gst_parse_perform_link: linking v4l2src0:(any) to tiscolorize0:(any) (0/0) with caps "video/x-raw-gray, width=(int)1280, height=(int)720, framerate=(fraction)30/1"
0:00:00.010624239 13849      0x234fa00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:376:gst_element_factory_create: creating element "capsfilter"
0:00:00.010663137 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24c22b0> adding pad 'sink'
0:00:00.010678813 13849      0x234fa00 INFO        GST_ELEMENT_PADS gstelement.c:728:gst_element_add_pad:<GstBaseTransform@0x24c22b0> adding pad 'src'
[... cut ...]
0:00:01.069826267 13849      0x234fa00 INFO                GST_INIT gst.c:1023:gst_deinit: deinitializing GStreamer
0:00:01.070756250 13849      0x234fa00 INFO                GST_INIT gst.c:1129:gst_deinit: deinitialized GStreamer

I guess it comes from the v4l2src which is advertising the following formats:
YU12 (emulated)
YV12 (emulated)
BGR3 (emulated)
RGB3 (emulated)

while tiscolorize is setup to have video/x-raw-gray.
But I've no idea on how to make v4l2src advertise the correct format.

Thanks for the help !

TIS-Stefan, 2015-11-02:
Hello

I have the suspect, you wrote the wrong firmware into the camera.
Please use
`sudo ./euvccam-fw -u ../../data/firmware/usb2/dmk72uc02_162_uvc.euvc`
`sudo ./euvccam-fw -m uvc`

Then disconnect and reconnect the camera. The pipeline should be
`gst-launch-0.10 v4l2src device=/dev/video1 ! video/x-raw-gray,width=1280,height=720,framerate=30/1 ! tisvideobufferfilter ! tiscolorize ! tiswhitebalance ! tis_auto_exposure ! bayer2rgb ! ffmpegcolorspace ! queue ! xvimagesink`

Please use the `tisvideobufferfilter` because the first frame can be incomplete, which will lead to a segmentation fault, if it is not dropped.

retornaz, 2015-11-02:
great, thanks now it works ! 

FYI, I used the tools/euvccam-fw/assistant tool to update the firmware. Maybe it needs a fix ? 

Anyway, I consider this solved now!

TIS-Stefan, 2015-11-02:
The currently available version of euvccam-fw works fine. If this did not work, then your camera would be unusable by now.
BTW: The tiswhitebalance was update today. You may do a
`git pull`
`cd ~/tiscamera/scr`
`make clean`
`make`
`make install`

closed 2015-11-02 by TIS-Stefan