TheImagingSource/tiscamera #54: mxc_v4l_open: Mxc Camera no sensor ipu0/csi0

reinha, 2015-12-05:
Hello,

I am trying to setup my DMK 22BUC03 Usb Camera on Ubuntu 14.04 (armhf) on my i.mx6 Freescale SabreLite Board. The kernel is 3.10.17. I followed all the steps in the wiki but the camera is only recognized correctly as USB device ( lsusb shows Imaging Source device is detected. ) I updated camera Firmware successfully, and /dev/video* shows up. But when trying with each /dev/video* to start any program (v42l-utils, fswebcam, gstreamer ...) which tries to access camera, the program aborts with things like "can't read / write from /dev/video0" or "resource temporarily unavailable" and i get always the following in my syslog.

mxc_v4l_open: Mxc Camera no sensor ipu0/csi0
mxc_v4l_open: Mxc Camera no sensor ipu0/csi1
mxc_v4l_open: Mxc Camera no sensor ipu1/csi1

What am I missing? Hopefully we can resolve this.

Regards

Reinhard

TIS-Stefan, 2015-12-08:
Hello Reinhard

I searched in the internet for "mxc_v4l_open: Mxc Camera no sensor ipu0/csi0" and found some hints to the USB controller. Thus, I would like you to try an active, poweredUSB hub between camera and your computer.

It might be a good idea to check the camera on another Linux computer to make sure, it is working fine.

reinha, 2015-12-08:
Hello,

This issue is resolved now. It turned out that uvcvideo was not activated in my kernel. Newbie mistake I guess

TIS-Stefan, 2015-12-08:
I must admit, I never would have found that on my own...

Stefan

closed 2015-12-08 by TIS-Stefan