TheImagingSource/tiscamera #57: run /tiscamera-master/examples/aravis/c/gstexample  error

ling459903347, 2015-12-21:
Hi:
Camera:DMK 23G445
Environment: ubuntu 14.04 64

(gstexample:28589): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed
(gstexample:28589): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed
(gstexample:28589): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed
(gstexample:28589): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed
(gstexample:28589): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed
(gstexample:28589): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Unable to retrieve camera

    thank you a lot !

TIS-Stefan, 2015-12-21:
Hello

I suppose, you adapted the part
/* our camera variables */
#define CAMERA "The Imaging Source Europe GmbH-46210199"
#define WIDTH  2592
#define HEIGHT 1944
#define PATTERN "rggb"
#define INIT_FPS 15

in the source code to your camera? e.g. serial number, resolution, frame rate? ( 1280x960 @ 20fps)?

ling459903347, 2015-12-22:
The problem has been solved, and thank you again.

FarahnazJ, 2016-06-21:
Hi 
as a similar question, I have a DMK 33 GP 1300 camera
and OS: UBUNTU 14.04

I have problem in running `gstexample.c`
this is my `arv-camera-test` results:

 Looking for the first available camera
vendor name           = The Imaging Source Europe GmbH
model name            = DMK 33GP1300
device id             = 19614095
image width           = 1280
image height          = 1024
horizontal binning    = 1
vertical binning      = 1
payload               = 1310720 bytes
exposure              = 33333 µs
gain                  = 9 dB
gv n_stream channels  = 0
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 9000 bytes
Frame rate = 46 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 31 Hz
Frame rate = 30 Hz
^CFrame rate = 30 Hz
Completed buffers = 197
Failures          = 0
Underruns         = 0

but I get the error:

gstaravis could not be created.

I have followed the instructions in  [Getting Aravis example code work for DMK23G274 #21] issue, fix the `gstexample.c` due to patch file satisfying my cam, and solving last error but i found a new error:

(gstexample:5754): GLib-GObject-WARNING **: value "-nan" of type 'gdouble' is invalid or out of range for property 'value' of type 'gdouble'
Now playing stream from: The Imaging Source Europe GmbH-19614095

(gstexample:5754): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'

(gstexample:5754): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'
Segmentation fault (core dumped)

thanks a lot :)

TIS-Stefan, 2016-06-21:
The error message "(gstexample:5754): GLib-GObject-WARNING **: value "-nan" of type 'gdouble' is invalid or out of range for property 'value' of type 'gdouble'" is caused by an error in the program: near line 411 is 
`double rangelen2 = log(rmax) - log(rmin);`
Unfornately log(0)  (rmin = 0) is undefined, thus the following usage of rangelen2 works on undefined values. 
Use instead:
```
double rangelen2 = rmax - rmin;
double exposure_value = 10.0;// (ABSVAL_SLIDER_TICKS) / rangelen2 * ( log(value) - log(rmin) );
```
and this error message is away.

>  (gstexample:5754): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'
> 
> (gstexample:5754): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'
> 

I must admit, I do not know, where this error comes from, it must be something inside aravis.

I would like to know, whether the following pipeline works over a timeperiod of 2 Minutes:
`gst-launch-0.10 aravissrc camera-name="The Imaging Source Europe GmbH-19614095" ! video/x-raw-bayer,width=1280,height=1024,framerate=30/1 ! fakesink
`
Is the camera after this time period still available? On my computer, it is lost. That may leads to the segmentation fault.

FarahnazJ, 2016-06-22:
thanks for your comment

I found there must be a fault inside aravis. when i run the your mentioned pipeline I got the error:

ERROR: pipeline could not be constructed: no element "aravissrc".

when i address the aravissrc, as:
`gst-launch-0.10 --gst-plugin-load=/home/javidi/aravis/gst-0.10/.libs/libgstaravis-0.4.so aravissrc camera-name="The Imaging Source Europe GmbH-19614095" num-buffers=1 ! video/x-raw-gray,width=1280,height=1024,framerate=15/1 ! fakesink`

It leads to the same segmentation fault:
>Setting pipeline to PAUSED ...

>(gst-launch-0.10:4299): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'

>(gst-launch-0.10:4299): GStreamer-CRITICAL **: range start is not smaller than end for `GstFractionRange'

>(gst-launch-0.10:4299): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed

>(gst-launch-0.10:4299): Aravis-CRITICAL **: arv_stream_push_buffer: assertion 'ARV_IS_STREAM (stream)' failed
>Pipeline is live and does not need PREROLL ...
>Setting pipeline to PLAYING ...
>New clock: GstSystemClock

>(gst-launch-0.10:4299): Aravis-CRITICAL **: arv_stream_timeout_pop_buffer: assertion 'ARV_IS_STREAM (stream)' failed
>ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data flow error.
Additional debug info:
>gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
>streaming task paused, reason error (-5)
>Execution ended after 438479 ns.
>Setting pipeline to PAUSED ...
>Setting pipeline to READY ...
>Setting pipeline to NULL ...
>Freeing pipeline ...

what is the solution? is there any problem with aravis?
Best Regards

FarahnazJ, 2016-06-22:
I should say that after running the pipeline, i get the _GstFractionRange_ message and then after 10 seconds the camera losts.

TIS-Stefan, 2016-06-22:
Hello

> I should say that after running the pipeline, i get the GstFractionRange message and then after 10 seconds the camera losts.

This is exact the same behavior, I have here. The GstFractionRange is not the problem. I suspect the camera's firmware. Using Wireshark for debugging, we see the camera stops answering.  The firmware engineer is working on this. 

The DFK 33GP006 works flawlessly. 

I am very sorry for the problem.

Stefan

FarahnazJ, 2016-06-22:
thanks anyway! :(
is there any other solution for capturing image from the video stream with this camera? does hardware trigger helps?

TIS-Stefan, 2016-06-22:
Hello

With Gstreamer currently not. The arv-viewer seem to work, that means using the aravis directly could do the job. However, we are working on this.

TIS-Arne, 2017-08-10:
Closed (outdated)

closed 2017-08-10 by TIS-Arne