TheImagingSource/tiscamera #58: Error while running gst_launch pipeline with DFK 22BUC03

swetham92, 2016-01-04:
Hello sir,

I am getting the problem/error message-

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 21270000 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

while trying to using the following gst-launch pipeline command with the odroid board (ubuntu version 15.04).

gst-launch-0.10 v4l2src ! video/x-raw-gray ! tisvideobufferfilter ! tiscolorize ! tis_auto_exposure ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! ximagesink

However the same pipeline worked with a desktop PC with Ubuntu version 12.04 installed in it. I had followed the steps outlined in the below link in both the cases. And the camera module I am using is Imaging source camera DFK 22BUC03.

Link-
https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-cameras-on-a-Raspberry-PI-or-on-other-Linux-Systems

Please give appropriate suggestions so that I can get the problem rectified at the earliest.

Thanks & regards

TIS-Stefan, 2016-01-04:
What happens, if you remove the tis_auto_exposure?

BTW: It is just fine writing this error report once.

Stefan

swetham92, 2016-01-04:
Error message still remains the same after removing tis_auto_exposure.

TIS-Stefan, 2016-01-04:
I installed Ubuntu 15.04 Kernel 3.19 into virtual machine, because I am running out of computers for each  distribution and Kernel versions..... 

However, I was not able to run the DFK 22BUC03 in that system, it was disconnected. Then I tried another USB 3.0 camera to make sure, the pipeline modules in exactly the same pipeline you use, work. They do. No problem. However, the resolution was not 744x480 @ 60 fps. It was much higher.

You use an Odroid system. I suggest to use a lower frame rate, for example:

gst-launch-0.10 v4l2src  ! video/x-raw-gray,width=744,height=480,framerate=15/1 ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! ximagesink

Stefan

swetham92, 2016-01-04:
The error still persists when I am running the above pipeline in odroid system (ubuntu 15.04) at the lower frame rate of 15fps. The camera that is available with me currently DFK 22BUC03, so I can go ahead and use this camera only for the time being.

Initially I tried interfacing the camera through  python-openCV code with commands like  
cap=cv2.VideoCapture(0)
ret,frame=cap.read()
cv2.imshow('frame',frame)
 But it was saving the images as 3 separate lower resolution images the same way as was observed with a monochrome camera discussed  in the link https://github.com/TheImagingSource/tiscamera/issues/20

I also tried displaying the image with an mplayer command such as 
mplayer tv:// -tv driver=v4l2:device=/dev/video0 
This was displaying the image from the camera in full resolution,however the format of displayed image was gray scale and not color/RGB. The post in https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras
suggests that the firmware for color and monochrome is same. Should anything be tweaked there so that the mplayer stores the color image?

So that is when I came accross the post https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-cameras-on-a-Raspberry-PI-or-on-other-Linux-Systems
that suggests the usage of gstreamer modules to get the color image with imaging source color cameras. I doubt if some extra modules has to be installed or some steps have to be debugged to eliminate the problem of displaying color images with gst_launch pipelines?
However the surprising thing is that the gst_launch pipelines worked perfectly fine in the ubuntu 12.04 version in my desktop without any errors and was a color image!!

Thanks & Regards,
Swetha

TIS-Stefan, 2016-01-04:
Open CV and mplayer do not handle the camera, because they expect an RGB 24 video format, which makes not really sense for monochrome cameras.

What really concerns me is "v4l2src0: Internal data flow error." That means, the problem already happens in the v4l2src and not in the following pipeline modules.

Do you receive an error, if you use
gst-launch-0.10 v4l2src ! video/x-raw-gray ! fakesink

If it runs successfully, you receive only 
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

Then hit Ctrl+C to end.

swetham92, 2016-01-04:
No errors received with gst-launch-0.10 v4l2src ! video/x-raw-gray ! fakesink. Its working fine like how you had mentioned in the comment

TIS-Stefan, 2016-01-05:
Hmm. 

So next step is narrowing down step by step, where the problem occurs. You should try different pipelines:

gst-launch-0.10 v4l2src ! video/x-raw-gray,width=744,height=480,framerate=15/1 ! tisvideobufferfilter ! ! tiscolorize !  bayer2rgb !  ffmpegcolorspace ! ximagesink

Unfortunately I can not do this for you, because I do not have an Odroid like you.

Also I suggest to search in the internet for the error message on Odroid systems,

I am very sorry for not having a better answer right now.

gauravbaluni, 2016-02-25:
@swetham92 @TIS-Stefan Could you help me on this issue https://github.com/TheImagingSource/tiscamera/issues/60

TIS-Stefan, 2016-02-25:
I did help

closed 2016-02-25 by TIS-Stefan