TheImagingSource/tiscamera #59: Error from camera-ip-conf rescue

cmarmo, 2016-02-22:
Hi,
I'm using a GigE camera DMK 23G445. The camera and the pc used to managing it are both connected to the same switch (the camera is on a PoE port). They are configured to acquire a specific IP address on the DHCP pool (corresponding to their MAC address).
The system worked perfectly during 2 days then the camera disappeared from our screens. We deconnect and reconnect a lot of times but nothing had changed.
I tried using the 'rescue' command (is this the right way to use it?)
`$ camera-ip-conf-cli rescue --mac <macnumber> ip=<ipaddress> subnet=<subnet> gateway=<gateway>`
and the following error was thrown
`terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
Aborted`

Did I something wrong?

Thanks for any help,
Chiara

TIS-Stefan, 2016-02-22:
Hello

can you please let me know the output of 
./camera-ip-conf-cli -l

I would like you to check, whether the camera's IP still matches to your network IPs address range.

Also: The green LED must light constantly, the yellow must blink. If this is not the case, then I would point to a power problem to the camera or the speed of your network changed.

cmarmo, 2016-02-22:
Hello,
here is the output
`$ camera-ip-conf-cli list
No cameras found.`
:(
I will ask about the leds...

TIS-Stefan, 2016-02-22:
Hello

No cameras found, not good. Check all cables, the LEDs also.

cmarmo, 2016-02-22:
Ok, there was a connection problem. The camera is there now.
But when is the `rescue` command supposed to be used and was the `Abort` due to the fact that no camera was detected?
Thanks for your help.

TIS-Stefan, 2016-02-22:
I must admit, I do not know the use of these commands, I did not wrote the program. 

Stefan

cmarmo, 2016-02-22:
OK, well, no problem, I will look into, if I'll find some time... :)

Thanks for your time.
Chiara

closed 2016-11-23 by TIS-Stefan