TheImagingSource/tiscamera #61: Gscam support for tiscolorize

lamogui, 2016-02-25:
We have a problem with color tiscamera. We try to use gscam with tiscolorize plugin but it fail or show nothing:

`v4l2src device=/dev/video1 ! tiscolorize ! tis_auto_exposure  ! ffmpegcolorspace`
Dont fail but no image is sent (nothing visible)

`v4l2src device=/dev/video1 ! tiscolorize ! tis_auto_exposure ! bayer2rgb  ! ffmpegcolorspace`
Send the error: 
`[creative_cam-2] process has died [pid 16444, exit code -11, cmd /home/nyx/dev/catkin_ws/devel/lib/gscam/gscam camera/image_raw:=creative_cam/image_raw __name:=creative_cam __log:=/home/nyx/.ros/log/fd851872-dbd9-11e5-8b89-e840f20aaf02/creative_cam-2.log].
log file: /home/nyx/.ros/log/fd851872-dbd9-11e5-8b89-e840f20aaf02/creative_cam-2*.log
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[creative_view_rect-5] process has died [pid 16447, exit code -6, cmd /opt/ros/jade/lib/image_view/image_view image:=/creative_cam/image_rect_color __name:=creative_view_rect __log:=/home/nyx/.ros/log/fd851872-dbd9-11e5-8b89-e840f20aaf02/creative_view_rect-5.log].
log file: /home/nyx/.ros/log/fd851872-dbd9-11e5-8b89-e840f20aaf02/creative_view_rect-5*.log

`

TIS-Stefan, 2016-02-26:
May be you receive incomplete images, that must be filtered by the "tisvideobufferfilter". So you please try:
`v4l2src device=/dev/video1 ! tisvideobufferfilter !  tis_auto_exposure ! ffmpegcolorspace`

This should provide bayer raw data. The next iteration is:

`v4l2src device=/dev/video1 ! tisvideobufferfilter ! tiscolorize ! tis_auto_exposure ! bayer2rgb ! ffmpegcolorspace`

The error message you quoted, does not say anything to me, so you may ask the ROS peoples about this. I must admit, I never worked with ROS, but may be here we have a good starting point. 

Which camera do you use?

lamogui, 2016-02-26:
You are rigth. The issue was from gscam/bayer2rgb association. Not from this code. My bad :-S here a link:
https://github.com/ros-drivers/gscam/issues/26

-----Message d'origine-----

TIS-Stefan, 2016-02-26:
I think it should be possible to force RGB24:
`v4l2src device=/dev/video1 ! tiscolorize ! tis_auto_exposure ! bayer2rgb ! ffmpegcolorspace ! video/x-raw-rgb, bpp=24`

lamogui, 2016-02-26:
thx for your answer :)
we already solve the problem but I think that could help other people

closed 2016-02-26 by TIS-Stefan