TheImagingSource/tiscamera #63: Frame grabs with gstreamer and color cam

nikoeder, 2016-04-07:
So ...
I've been trying to capture a single uncompressed frame from the DFK 22BUC03 TIS camera.

The pipeline I use for a live feed works just fine:
`gst-launch-0.10 v4l2src device=/dev/video1 ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! ximagesink`

But if I modify only the filesink location like so:
`gst-launch-0.10 v4l2src device=/dev/video1 ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! pngenc ! filesink location=test.png`

I end up with pictures like these:
[image]

I guess it's got something to do with the color space ... But I didn't fully understand how the recoloring process works. Following the "First Steps Guide" here I uploaded the "dmk22uc03_163_uvc.euvc" firmware, which is (if I understood it correctly) the monochrome equivalent to my camera.

Or is it the pngencoder? But I couldn't find any arguments to pass to pngenc except "snapshot" and the compression level ... And playing around with those doesn't make it better ... 

Or is it something about the timing ... Because the size of those oddly colored frames as seen in the picture vary slightly ...

I just find it weird, that in live streaming mode everything works just fine and as soon as I want to save a single frame it messes up the picture...

I would appreciate every single hint!
Thank you and
with kind regards,
Niko

TIS-Stefan, 2016-04-07:
Hello,

I do not really now the input, the pngenc module needs, but for JPEG I used

`bayer2rgb !  ffmpegcolorspace ! video/x-raw-yuv ! jpegenc quality=35  ! multifilesink location="frame%05d.jpg"`

You may try this and check the result

nikoeder, 2016-04-20:
Thank you, it worked. I need an uncompressed format, though.

But interestingly enough, the color errors are consistent during the first 10 or so frames and then there is 2 or 3 frames in which the frames get more and more saturated until they are normal colors.
Any ideas where that comes from?

Thank you!

TIS-Stefan, 2016-04-20:
The color error occur, because there are image data blocks not picked up from the USB controller. Thus the Bayer Mosaic pattern is  wrong at the position, where the data block is missing. 
I do not know, why the USB controller misses these data packages. It is either, they are not picked up from the camera or the USB controller was not able to save them in memory. You may use a lower frame rate, so the USB controller has a little bit more time.

nikoeder, 2016-04-20:
OK ... I solved it by just grabbing 10 frames and deleting the first 9 all in one script ... I know it's not very elegant, but it works now ...

Thank you for your help!

AndrHe, 2017-03-23:
Hi Niko,

can you send me this script you mentioned?

Thank you in advance

Andreas

nikoeder, 2017-03-23:
Hey Andreas!

I just dug them up. Here's the final script I ended up using. It's used to take four frame grabs from four different cameras connected via USB.
I had to change the file type from .sh to .txt to upload it here.

[frameGrab_public.txt](https://github.com/TheImagingSource/tiscamera/files/865194/frameGrab_public.txt)

I hope it helps!

AndrHe, 2017-03-23:
Danke

closed 2017-05-15 by TIS-Edgar