TheImagingSource/tiscamera #64: Problem with the ShowCamera example

Ishtar-UK, 2016-05-02:
Hello everyone

I am currently playing with the showcamera example but am experiencing problems. Here are the details of my setup:

- OS: Fedora 23
-  Kernel Version:  4.4.8-300
- GStreamer Version: 0.10.36
- GStreamer Plugins Base: 0.10.36
- GStreamer Plugins Good: 0.10.36
- GStreamer Plugins Bad: 0.10.36
- GStreamer Plugins Ugly: 0.10.36
- Camera: DMK21AU04 & DFM21AU04

Initially, when I run showcamera I get a Qt UI start up with a black window in the middle. I then select my camera and click start but no image appears. Instead, the black window disappears. According to the console debugging output it seems to go wrong at this point:

`Camera Change DMx 21AU04.AS 3900322
Pipeline build start
The pipeline element "tisvideobufferfilter", "bufferfilter" could not be created.`

Does anybody have any idea of what is wrong?

Many thanks
Amanda

For more information here is the full debugging output from my console:

`
[BUILD]$ ./showcamera
/dev/video1 The Imaging Source Europe GmbH DMx 21AU04.AS 3900322
Checking /dev/video1
Property id 0x980900 name Brightness min: 0, max: 63, default 0, type 1
Property id 0x980910 name Gamma min: 1, max: 500, default 100, type 1
Property id 0x980913 name Gain min: 260, max: 1023, default 260, type 1
Property id 0x9a0901 name Exposure, Auto min: 0, max: 3, default 3, type 3
Property id 0x9a0902 name Exposure (Absolute) min: 1, max: 36000000, default 333, type 1
End
/dev/video0 Generic HP Truevision HD 200901010001
Checking /dev/video0
Property id 0x980900 name Brightness min: -64, max: 64, default 0, type 1
Property id 0x980901 name Contrast min: 0, max: 100, default 50, type 1
Property id 0x980902 name Saturation min: 0, max: 100, default 64, type 1
Property id 0x980903 name Hue min: -180, max: 180, default 0, type 1
Property id 0x98090c name White Balance Temperature, Auto min: 0, max: 1, default 1, type 2
Property id 0x980910 name Gamma min: 100, max: 500, default 300, type 1
Property id 0x980913 name Gain min: 0, max: 100, default 0, type 1
Property id 0x980918 name Power Line Frequency min: 0, max: 2, default 2, type 3
Property id 0x98091a name White Balance Temperature min: 2800, max: 6500, default 4650, type 1
Property id 0x98091b name Sharpness min: 0, max: 100, default 50, type 1
Property id 0x98091c name Backlight Compensation min: 0, max: 1, default 0, type 1
Property id 0x9a0901 name Exposure, Auto min: 0, max: 3, default 3, type 3
Property id 0x9a0902 name Exposure (Absolute) min: 50, max: 10000, default 312, type 1
Property id 0x9a0903 name Exposure, Auto Priority min: 0, max: 1, default 0, type 2
End
Checking Generic camera  /dev/video0
	Format 0 :  YUYV 4:2:2  1 YUYV
		640x480
			30/1
		160x120
			30/1
		320x240
			30/1
		1280x720
			10/1
	Format 1 :  Motion-JPEG  1 MJPG
		640x480
			30/1
		160x120
			30/1
		320x240
			30/1
		1280x720
			30/1
Property id 0x980900 name Brightness min: -64, max: 64, default 0, type 1
Property id 0x980901 name Contrast min: 0, max: 100, default 50, type 1
Property id 0x980902 name Saturation min: 0, max: 100, default 64, type 1
Property id 0x980903 name Hue min: -180, max: 180, default 0, type 1
Property id 0x98090c name White Balance Temperature, Auto min: 0, max: 1, default 1, type 2
Property id 0x980910 name Gamma min: 100, max: 500, default 300, type 1
Property id 0x980913 name Gain min: 0, max: 100, default 0, type 1
Property id 0x980918 name Power Line Frequency min: 0, max: 2, default 2, type 3
Property id 0x98091a name White Balance Temperature min: 2800, max: 6500, default 4650, type 1
Property id 0x98091b name Sharpness min: 0, max: 100, default 50, type 1
Property id 0x98091c name Backlight Compensation min: 0, max: 1, default 0, type 1
Property id 0x9a0901 name Exposure, Auto min: 0, max: 3, default 3, type 3
Property id 0x9a0902 name Exposure (Absolute) min: 50, max: 10000, default 312, type 1
Property id 0x9a0903 name Exposure, Auto Priority min: 0, max: 1, default 0, type 2
End
Camera Change DMx 21AU04.AS 3900322
Pipeline build start
The pipeline element "tisvideobufferfilter", "bufferfilter" could not be created.

(showcamera:28487): GStreamer-CRITICAL **: passed '0' as denominator for `GstFraction'
Module "autoexposure" is null.
Module "bufferfilter" is null.

** (showcamera:28487): CRITICAL **: Unable to link source pipeline to autoexposure
Module "image_sink" is null.

(showcamera:28487): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Module "image_sink" is null.

(showcamera:28487): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(showcamera:28487): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed Playing

`

For more information, Here is the output of ldd ./showcamera

`[BUILD]$ ldd ./showcamera
	linux-vdso.so.1 (0x00007ffc387f9000)
	libQtCore.so.4 => /lib64/libQtCore.so.4 (0x00007f11e2c81000)
	libQtGui.so.4 => /lib64/libQtGui.so.4 (0x00007f11e1f47000)
	libgstinterfaces-0.10.so.0 => /usr/local/image-processor-dependencies/lib/libgstinterfaces-0.10.so.0 (0x00007f11e1d35000)
	libgstapp-0.10.so.0 => /usr/local/image-processor-dependencies/lib/libgstapp-0.10.so.0 (0x00007f11e1b28000)
	libgstbase-0.10.so.0 => /usr/local/image-processor-dependencies/lib/libgstbase-0.10.so.0 (0x00007f11e18ce000)
	libgstreamer-0.10.so.0 => /usr/local/image-processor-dependencies/lib/libgstreamer-0.10.so.0 (0x00007f11e15e0000)
	libgobject-2.0.so.0 => /lib64/libgobject-2.0.so.0 (0x00007f11e138e000)
	libgmodule-2.0.so.0 => /lib64/libgmodule-2.0.so.0 (0x00007f11e1189000)
	libgthread-2.0.so.0 => /lib64/libgthread-2.0.so.0 (0x00007f11e0f87000)
	libglib-2.0.so.0 => /lib64/libglib-2.0.so.0 (0x00007f11e0c4e000)
	libxml2.so.2 => /lib64/libxml2.so.2 (0x00007f11e08e4000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007f11e08c3000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f11e0541000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f11e023e000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f11e0027000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f11dfc66000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f11dfa48000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f11df832000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f11df62e000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f11df425000)
	/lib64/ld-linux-x86-64.so.2 (0x000055a9dabd3000)
	libpng16.so.16 => /lib64/libpng16.so.16 (0x00007f11df1f2000)
	libfreetype.so.6 => /lib64/libfreetype.so.6 (0x00007f11def48000)
	libSM.so.6 => /lib64/libSM.so.6 (0x00007f11ded3e000)
	libICE.so.6 => /lib64/libICE.so.6 (0x00007f11deb22000)
	libXi.so.6 => /lib64/libXi.so.6 (0x00007f11de912000)
	libXrender.so.1 => /lib64/libXrender.so.1 (0x00007f11de706000)
	libXrandr.so.2 => /lib64/libXrandr.so.2 (0x00007f11de4fb000)
	libXfixes.so.3 => /lib64/libXfixes.so.3 (0x00007f11de2f5000)
	libXcursor.so.1 => /lib64/libXcursor.so.1 (0x00007f11de0e9000)
	libXinerama.so.1 => /lib64/libXinerama.so.1 (0x00007f11ddee6000)
	libfontconfig.so.1 => /lib64/libfontconfig.so.1 (0x00007f11ddca2000)
	libXext.so.6 => /lib64/libXext.so.6 (0x00007f11dda8f000)
	libX11.so.6 => /lib64/libX11.so.6 (0x00007f11dd74f000)
	libffi.so.6 => /lib64/libffi.so.6 (0x00007f11dd546000)
	liblzma.so.5 => /lib64/liblzma.so.5 (0x00007f11dd320000)
	libselinux.so.1 => /lib64/libselinux.so.1 (0x00007f11dd0fd000)
	libresolv.so.2 => /lib64/libresolv.so.2 (0x00007f11dcee1000)
	libdw.so.1 => /lib64/libdw.so.1 (0x00007f11dcc97000)
	libcap.so.2 => /lib64/libcap.so.2 (0x00007f11dca92000)
	libbz2.so.1 => /lib64/libbz2.so.1 (0x00007f11dc881000)
	libuuid.so.1 => /lib64/libuuid.so.1 (0x00007f11dc67c000)
	libexpat.so.1 => /lib64/libexpat.so.1 (0x00007f11dc451000)
	libxcb.so.1 => /lib64/libxcb.so.1 (0x00007f11dc22f000)
	libpcre.so.1 => /lib64/libpcre.so.1 (0x00007f11dbfbf000)
	libelf.so.1 => /lib64/libelf.so.1 (0x00007f11dbda6000)
	libattr.so.1 => /lib64/libattr.so.1 (0x00007f11dbba0000)
	libXau.so.6 => /lib64/libXau.so.6 (0x00007f11db99b000)
`

Ishtar-UK, 2016-05-02:
**** UPDATE ****

Since posting this thread I have compiled the Imaging Source GStreamer modules. This has changed the nature of the error. The application still does not work but it now outputs the following debugging information:

Camera Change DMx 21AU04.AS 3900322
Pipeline build start
(showcamera:2948): GStreamer-CRITICAL **: passed '0' as denominator for `GstFraction'
** (showcamera:2948): CRITICAL **: Unable to link source pipeline to autoexposure
Playing

Many thanks
Amanda

TIS-Stefan, 2016-05-02:
Hello,

the showcamera sample is not suitable for your camera, since it makes auto exposure on its own. You can directly use programs like guvcview with this camera mode.

The error message indicates, that the frame rate seems not to be read out correctly.

Stefan

Ishtar-UK, 2016-05-02:
Hi Stefan

Yes, I can use gnucview (works perfectly) but my ultimate aim is to write my own image capture application. I've been trying to use the showcamea sample as a way to learn what I need to do. Do you think I would be able to modify showcamera to remove auto exposure or is it too well integrated with the code?

Regards
Amanda

TIS-Stefan, 2016-05-03:
Hello

Well, the showcamera has some automatics for building the Gstreamer pipeline, which is optimized for the CMOS cameras, that do not provide color images and automatics. Also it was my first Gstreamer project, so it may is somewhat hard to understand.
Much simpler to use is the "gst_parse_launch ("...", &error); " function. You can create a working pipeline using gst_launch-0,10 on the command line pass this pipeline to the function above. This is much shorter and easy to maintain.

Stefan

Ishtar-UK, 2016-05-03:
Thanks Stefan, I appreciate it.

Do you think it will be easier to use V4L2 directly? Is it even possible or will I need to use GStreamer. To be honest, I have never written a project which uses GStreamer before so it would be a learning curve, therefore I would prefer to access the DMK / DFK  directly with V4l2 . 

Amanda

TIS-Stefan, 2016-05-04:
Hi Amanda

for the DMK it does not really matter, whether you use GStreamer or V4L2 directly.

Stefan

Ishtar-UK, 2016-05-05:
Thanks for you help Stefan, I appreciate it.

Regards
Amanda

closed 2016-05-05 by Ishtar-UK