TheImagingSource/tiscamera #65: Long delay while streaming on Linux

AHW-Cambridge-UK, 2016-05-19:
ImagingSource DMK 72AUC02 camera on a Raspberry Pi.
Followed instructions on https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-cameras-on-a-Raspberry-PI-or-on-other-Linux-Systems

I'm using the V4L2 video capture example from https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e
I call start_capturing(), then a thread makes repeated calls to snapFrame() and memcpy the buffer into an OpenCV Mat.

That all works nicely, thank you.

I placed a clock with a second-hand in front of the camera, and I see the displayed image lag behind reality by about 2 seconds. This happens even with short exposure times of (say) 0.01s.

Any suggestions would be very welcome.
Thanks in advance,
Alasdair

TIS-Stefan, 2016-05-26:
Hello

The Raspberry PI is not a very fast computer, thus all operations on the 5 MByte size raw data and later 15 or 20 MByte size colored data for display will take some time. 
also the frame rate of the camera determines, how fast a frame is transferred from the camera to the computer. In case you run with 30 fps, the image data transfer will last 1000ms/30 fps = 33ms.

Two seconds is long, but I think, it is caused by the PI. The camera has no memory for images. As soon as a line of the sensor is exposed, the data will be send to the computer.

closed 2016-05-26 by TIS-Stefan

AHW-Cambridge-UK, 2016-05-26:
Thank you for your reply.
I'm (painfully) aware of the speed issue, but this isn't much of a problem for our application.
We are actually working with relatively at a frame period of 1/5s, so the time taken through the OpenCV processing isn't so significant.

A more serious problem for me seems to be some sort of pipeline effect, although you have explained that there is no storage within the camera itself.
If I change the exposure time, the next two images returned appear to be captured at the original exposure time. This creates some difficulty because I am attempting to make precise calculations based on image contents and exposure time.

I understand that if the camera is streaming live data, setting an exposure time during a frame may not have effect until the following frame.
However, it appears to need to flush at least the next two frames before an image at the correct exposure time turns up.

In the process of writing this, I'm looking back at the V4L2 video capture example from https://gist.github.com/TIS-Edgar/10f04501f49b6b3bf75e and I notice that functions init_mmap and init_userp appear to use 4 buffers in rotation, presumably to allow one to be accessed while another is being written to.
I must admit that I have incorporated this example code without fully understanding it.
Would it be possible for you to explain why 4 buffers rather than 2 are required, please?

Thanks again for your reply, and for any further help you can give me.
Alasdair

TIS-Stefan, 2016-05-26:
I wonder, whether a Raspberry PI is a sufficient computer for your task.

Changing the exposure needs a frame to become effective in the sensor.

> Would it be possible for you to explain why 4 buffers rather than 2 are required, please?
Unfortunately I do not know this. You may ask the authors about the v4l2 implementation. In Windows we use more than one buffer, because while one buffer is processed by the software, the camera driver can fill the next buffer.

AHW-Cambridge-UK, 2016-05-26:
Thank you again.

Build-cost is a major factor for our product, hence using Raspberry. Not ideal, but that's business and it's mostly working out OK.

I realise that the V4L2 example is a generic sample, and it isn't your responsibility.
Any further comments or explanation would be helpful, and I would welcome suggestions from anybody else on this group.
I will look at the effect of changing the number of buffers - that might be feasible at the slow framerate we use.

Alasdair

TIS-Stefan, 2016-05-26:
Hello Alasdair

You mean, you buy a VW Golf and expect power rating of a big Mercedes? (I know, my pictures are  weird sometimes.) But you must admit: If you want to save money, you can not expect powerful computers.

AHW-Cambridge-UK, 2016-07-07:
I've finally found my problem: unexpectedly slow GLib threads.

I knew that the processing overhead on a Pi was significant, but even though I was using maximum image size of 2592x1944, rotated landscape to portrait and false-coloured, that didn't add up to 4 seconds.

I noticed that v4ldevice.cpp cycles through 4 buffers in turn, so I modified that to make it configurable from my application: using 2 buffers reduced the lag from 2 seconds to 1 second.
Still a long time, and now having looked through the example code I could see that there was nowhere for the image data to hide.

Eventually I found the answer in my own program's threading, which uses thread pools to read from the camera while maintaining control of other hardware and keeping reasonable response to the user interface.
Specifics for anybody else following...
I called g_thread_pool_new to create a thread pool with one non-exclusive thread.
What I hadn't anticipated is that my repeated call to g_thread_pool_push to start each new read actually takes several hundred milliseconds.
This might be well-known behaviour, or might be Raspberry-specific - I didn't find anything documented.

I changed the program so g_thread_pool_new creates an exclusive thread, which is retained between calls to g_thread_pool_push.

Now the live image from my ImagingSource camera updates a rotated false-colour image at a framerate of 4 or 5fps, fast enough for the user to adjust on-screen focus & zoom controls comfortably.
So not a racing car, perhaps just my 2.0TDI Golf?

Conclusion for anybody interested: the Raspberry Pi + ImagingSource camera is a viable proposition to make good commercial products.

Thanks,
Alasdair