TheImagingSource/tiscamera #67: tcamsrc fails with bayer output

mattikra, 2016-06-04:
I am trying to get the gstreamer 1.0 plugins from the beta-release branch working with a DFK77BUC02 (x86-64, Ubuntu 14.04). Compile and install works, although I had to tweak GST_PLUGIN_PATH_1_0 and LD_LIBRARY_PATH to bring the plugins to life (there seems to be a minor flaw in the makefile). Grayscale works fine with the following pipeline:

gst-launch-1.0 tcamsrc ! video/x-raw ! videoconvert ! autovideosink

However, when I try to get color from the camera with this pipeline:

gst-launch-1.0 tcamsrc ! video/x-bayer ! bayer2rgb ! autovideosink

gstreamer fails:

Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:11893): GStreamer-CRITICAL **: gst_structure_set_value: assertion 'G_IS_VALUE (value)' failed
ERROR: from element /GstPipeline:pipeline0/GstTcam:tcam0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcam:tcam0:
streaming task paused, reason not-negotiated (-4)

This is the gdb backtrace of the first critical message:

#0  0x00007ffff75ccc13 in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff75ccd72 in g_log () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7b12fda in gst_caps_get_structure (caps=caps@entry=0x612050, index=index@entry=0) at gstcaps.c:814
#3  0x00007ffff3cf925c in gst_bayer2rgb_transform_caps (base=<optimized out>, direction=GST_PAD_SINK, caps=0x612050, 
    filter=<optimized out>) at gstbayer2rgb.c:291
#4  0x00007ffff5af8acd in gst_base_transform_transform_caps (trans=trans@entry=0x7ea0e0, direction=GST_PAD_SINK, 
    caps=caps@entry=0x612050, filter=filter@entry=0x0) at gstbasetransform.c:523
#5  0x00007ffff5af9e67 in gst_base_transform_query_caps (filter=0x612050, pad=0x7e42f0, trans=0x7ea0e0)
    at gstbasetransform.c:699
#6  gst_base_transform_default_query (trans=0x7ea0e0, direction=<optimized out>, query=0x8062d0) at gstbasetransform.c:1512
#7  0x00007ffff7b3c4ea in gst_pad_query (pad=pad@entry=0x7e42f0, query=query@entry=0x8062d0) at gstpad.c:3511
#8  0x00007ffff7b3cad0 in gst_pad_peer_query (pad=pad@entry=0x7e5010, query=query@entry=0x8062d0) at gstpad.c:3642
#9  0x00007ffff7b6dd54 in gst_pad_peer_query_caps (pad=pad@entry=0x7e5010, filter=filter@entry=0x612050) at gstutils.c:2810
#10 0x00007ffff5af9f64 in gst_base_transform_query_caps (filter=0x8061e0, pad=0x7e4de0, trans=0x804130)
    at gstbasetransform.c:715
#11 gst_base_transform_default_query (trans=0x804130, direction=<optimized out>, query=0x806230) at gstbasetransform.c:1512
#12 0x00007ffff7b3c4ea in gst_pad_query (pad=pad@entry=0x7e4de0, query=query@entry=0x806230) at gstpad.c:3511
#13 0x00007ffff7b3cad0 in gst_pad_peer_query (pad=pad@entry=0x7e40c0, query=query@entry=0x806230) at gstpad.c:3642
#14 0x00007ffff7b6dd54 in gst_pad_peer_query_caps (pad=0x7e40c0, filter=0x8061e0) at gstutils.c:2810
#15 0x00007ffff62413c9 in gst_tcam_negotiate () from /usr/local/TCAM_INSTALL_GST_1_0/libgsttcamsrc.so
#16 0x00007ffff5af5258 in gst_base_src_negotiate (basesrc=0x7e2150) at gstbasesrc.c:3154
#17 gst_base_src_loop (pad=0x7e40c0) at gstbasesrc.c:2629
#18 0x00007ffff7b64549 in gst_task_func (task=0x830050) at gsttask.c:316
#19 0x00007ffff75eb88c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff75eaf05 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff7367182 in start_thread (arg=0x7fffea5ca700) at pthread_create.c:312
#22 0x00007ffff709447d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

Any clue what's going on? Should bayer output be already working in the beta branch?

TIS-Edgar, 2016-06-06:
Hi,

Since gray works I am assuming, that you flashed the camera with a dmk firmware right?

The v4l2 backend currently only shows those formats that the camera says it has.
This means you will have to try something like this:
    gst-launch-1.0 tcamsrc !  video/x-raw,format=GRAY8 ! capssetter caps="video/x-bayer,format=bggr" replace=true ! video/x-bayer ! bayer2rgb ! autovideosink

Please be aware, that the beta branch is only only for preview purposes and is most definitely not production ready. 

Regards,

Edgar

TIS-Edgar, 2016-11-16:
Support for Gstreamer-1.0 has been officially released.

closed 2016-11-16 by TIS-Edgar