TheImagingSource/tiscamera #68: DFK / VIDIOC_S_CTRL Problem

Ishtar-UK, 2016-06-19:
Hello Everyone

I am in the process of writing a Linux application to control the Imaging Source USB cameras and testing with both a DMK21 (GREY) and an DFK colour camera. However, I have run into a problem.

I have written the following code to update the Gamma (Or indeed, any camera parameters I choose to set). This code snippet works perfectly with normal webcams (that is, the gamma is updated when cam_ctrl_gamma is changed) but does nothing with the DFK / DMK cameras. 

Does anybody know what I am doing wrong?

`struct` v4l2_control control;
memset(&control, 0, sizeof (control));

 control.id = V4L2_CID_GAMMA;
 control.value = cam_ctrl_gamma;

 if (ioctl(fd, VIDIOC_S_CTRL, &control) < 0)
 {
      cerr << "ERROR: " + (string) strerror(errno) << "\n";
      return;
 }`

Kind Regards
Amanda

TIS-Stefan, 2016-06-20:
Hello 

I would like you to try that with guvcview. On my computer, gamma works as expected. Please let me know your results. I suppose, you installed the rules.

Ishtar-UK, 2016-06-20:
Hi Stefan and thanks for your response. 

I have tested on guvcview and it has the same problem so it doesn't look like it's a problem with my code.  BTW: It's not only a problem with the Gamma—all user controls don't work. I have installed the UDEV rules. 

Any idea?

Kind Regards
Amanda

Ishtar-UK, 2016-06-20:
Sorry, forgot to add. I am using Fedora 23, Kernel version 4.5.7

Many thanks
Amanda

TIS-Stefan, 2016-06-20:
I have no real idea, since it works fine on my machine. But I installed the UVC extension Unit. https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units
Maybe this makes the difference?

Ishtar-UK, 2016-06-20:
Thanks Stefan

I just tired installing the UVC extension unit but got lots of errors back. Any ideas?

`uvcdynctrl -i tisEUVC.xml -d /dev/video1
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090E (V4L2_CID_USER_BASE + 14)
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090F (V4L2_CID_USER_BASE + 15)

Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml:48: parser error : Specification mandate value for attribute data-pjax-transient
    <meta name="selected-link" value="repo_source" data-pjax-transient>
                                                                      ^
tisEUVC.xml:87: parser error : Specification mandate value for attribute data-pjax-transient
ource/tiscamera/blob/master/data/uvc-extensions/tisEUVC.xml" data-pjax-transient
                                                                               ^
tisEUVC.xml:88: parser error : Opening and ending tag mismatch: link line 87 and head
  </head>
         ^
tisEUVC.xml:104: parser error : Opening and ending tag mismatch: svg line 104 and path
.93-.01 2.2 0 .21.15.46.55.38A8.013 8.013 0 0 0 16 8c0-4.42-3.58-8-8-8z"></path>
                                                                               ^
tisEUVC.xml:104: parser error : Opening and ending tag mismatch: a line 103 and svg
01 2.2 0 .21.15.46.55.38A8.013 8.013 0 0 0 16 8c0-4.42-3.58-8-8-8z"></path></svg
                                                                               ^
tisEUVC.xml:105: parser error : Opening and ending tag mismatch: div line 102 and a
    </a>
        ^
tisEUVC.xml:147: parser error : Opening and ending tag mismatch: input line 137 and label
    </label>
            ^
tisEUVC.xml:148: parser error : Opening and ending tag mismatch: label line 135 and form
</form></div>
       ^
tisEUVC.xml:148: parser error : Opening and ending tag mismatch: form line 134 and div
</form></div>
             ^
tisEUVC.xml:150: parser error : Opening and ending tag mismatch: div line 133 and nav
        </nav>
              ^
tisEUVC.xml:151: parser error : Opening and ending tag mismatch: nav line 128 and div
    </div>
          ^
tisEUVC.xml:164: parser error : Specification mandate value for attribute itemscope
        <div itemscope itemtype="http://schema.org/SoftwareSourceCode">
                       ^
tisEUVC.xml:164: parser error : attributes construct error
        <div itemscope itemtype="http://schema.org/SoftwareSourceCode">
                       ^
tisEUVC.xml:164: parser error : Couldn't find end of Start Tag div line 164
        <div itemscope itemtype="http://schema.org/SoftwareSourceCode">
                       ^
tisEUVC.xml:165: parser error : Specification mandate value for attribute data-pjax-container
    <div id="js-repo-pjax-container" data-pjax-container>
                                                        ^
tisEUVC.xml:228: parser error : Specification mandate value for attribute itemscope
     itemtype="http://schema.org/BreadcrumbList"
     ^
tisEUVC.xml:228: parser error : attributes construct error
     itemtype="http://schema.org/BreadcrumbList"
     ^
tisEUVC.xml:228: parser error : Couldn't find end of Start Tag nav line 226
     itemtype="http://schema.org/BreadcrumbList"
     ^
tisEUVC.xml:232: parser error : Specification mandate value for attribute itemscope
  <span itemscope itemtype="http://schema.org/ListItem" itemprop="itemListElemen
                  ^
tisEUVC.xml:232: parser error : attributes construct error
  <span itemscope itemtype="http://schema.org/ListItem" itemprop="itemListElemen
                  ^
tisEUVC.xml:232: parser error : Couldn't find end of Start Tag span line 232
[... 111 lines cut ...]
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 41
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 40
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 39
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 35
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 34
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 33
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 32
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 31
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 30
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 29
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 28
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 27
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 26
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 25
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 24
tisEUVC.xml:2055: parser error : Premature end of data in tag link line 23
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 19
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 18
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 17
tisEUVC.xml:2055: parser error : Premature end of data in tag meta line 4
tisEUVC.xml:2055: parser error : Premature end of data in tag head line 3
tisEUVC.xml:2055: parser error : Premature end of data in tag html line 2
uvcdynctrl: /builddir/build/BUILD/libwebcam-src-0.2.2/libwebcam/dynctrl.c:956: parse_dynctrl_file: Assertion `parser->valid' failed.
Aborted
`

Ishtar-UK, 2016-06-20:
Hi Stefan

The first time I tried installing the extension units I lots of errors. I have downloaded the XML again and this time I got less errors but still not running correctly. There seems to be lots of no such file or directory errors.Any idea what I am missing?

sudo uvcdynctrl -i tisEUVC.xml -d /dev/video1
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090E (V4L2_CID_USER_BASE + 14)
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090F (V4L2_CID_USER_BASE + 15)
Importing dynamic controls from file tisEUVC.xml.
tisEUVC.xml: error: video1: unable to map 'ExtIO' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainR' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainG' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'GainB' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Trigger Global Reset Shutter' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Software Trigger' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Binning' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Skipping' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'X Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
tisEUVC.xml: error: video1: unable to map 'Y Offset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

TIS-Stefan, 2016-06-20:
For using the  tisEUVC.xml file you can use the scripts from
https://github.com/TheImagingSource/tiscamera/tree/beta-release/tools/uvcctls

Ishtar-UK, 2016-06-20:
Hello Stefan

I have tested guvcview again and none of the sliders have any effect on the DFK output. Here is the output of v4l2-ctl --all

Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : DFx 21AU04.AS
	Bus info      : usb-0000:00:14.0-1
	Driver version: 4.5.7
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : Default
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=0 max=63 step=1 default=0 value=48
                     saturation (int)    : min=0 max=255 step=1 default=128 value=128
                            hue (int)    : min=0 max=359 step=1 default=180 value=180
   white_balance_component_auto (bool)   : default=1 value=1
    white_balance_red_component (int)    : min=0 max=95 step=1 default=32 value=32 flags=inactive
   white_balance_blue_component (int)    : min=0 max=95 step=1 default=32 value=33 flags=inactive
                          gamma (int)    : min=1 max=500 step=1 default=100 value=187
                           gain (int)    : min=260 max=1023 step=1 default=260 value=260
                  exposure_auto (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute (int)    : min=1 max=36000000 step=1 default=333 value=333 flags=inactive

TIS-Stefan, 2016-06-20:
Hello

I must admit, I have no more ideas. You do not gent an error, the properties look as expected. It works fine on my computer, thus I have no idea where to look for.  Did you try the other video formats, such as Y800?

Ishtar-UK, 2016-06-20:
Hi Stefan

It works perfectly on Y800 (aside from my display being the work colour because I've not yet implemented the converter). So, it appears to be a problem with YUV. I'm lost on this one too...

TIS-Stefan, 2016-06-21:
Hi Ishtar

Y800 should not give a strange color. It gives a b/w image with a checkerboard pattern. The pattern is from the bayer mosaic raw data.

In my guvcview all properties, including gamma, have the expected effect, also if I select YUYV, 640x480 at 30 fps.

I am lost here, I do not know, why it does not work on your computer. If it is possible, I would like you to test this on another computer, e.g. Windows or Ubuntu.

Ishtar-UK, 2016-06-21:
Thanks Stefan

The reason I'm getting a greenish image is because I am using an SDL2 Texture specifically set for YUY2 and have no conversion in place yet. I will try this on another Linux system. 

Thanks for your help
Regards
Amanda

closed 2016-11-23 by TIS-Stefan