TheImagingSource/tiscamera #71: Errors with gstexample.py on Ubuntu 14.04

sven-bock, 2016-08-17:
Setup: DFK Z12G445 - GigE Color camera.

I installed aravis, all dependencies and everything according to the wiki and fixed the first bugs.
Now I am a little bit stuck.

I wanted to use the gstexample.py. The windows opens and the video stream seems to be started, but I dont see any image. If I touch one of the parameters I get:
**`(gstexample.py:6395): GStreamer-CRITICAL **: range start is not smaller than end for 'GstIntRange'`**

And on shutdown I get:
```
(gstexample.py:6395): GStreamer-CRITICAL **: 
Trying to dispose element colorspace, but it is in PAUSED instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

```
I assume that my configuration on the top of the script is not good, but I didnt find a proper doc so far.
My current config looks like (The serial number corresponds to my camera):
```
class Camera:
    """Camera representation containing everything we need"""
    name = "The Imaging Source Europe GmbH-32610040"  # name used by aravis
    width = 1280  # width of our image
    height = 960  # height of our image
    pattern = "rggb"  # bayer pattern; only needed for color cameras
    bpp = 32
    depth = 24
    init_fps = 15     # fps we want to start with
```
What do I need to set for the range? Do you have any working python sample with a GigE camera?
I am grateful for any input.

P.S.: The only working example where I could see a grayscale image was the arvviewer from aravis.

TIS-Stefan, 2016-08-18:
Hello Sven

sorry, for the delay. I can not get the Aravis import running on my computer currently. However, I suppose the "bpp" and "depth" are wrong. Both must be 8, since the camera provides raw data only.

I hope this is not a wrong information. Please try this.

Stefan

sven-bock, 2016-08-18:
Hello Stefan,
thank you for your reply. I played around with bpp and depth as well as the pattern but could not get it to work. 
I tried the project from here: https://github.com/CaeruleusAqua/camera_aravis/
With this code I am able to get the Mono16 image out of the camera, but I still struggle to get a colored image. Can you tell me if there is any other PixelFormat for color on the camera than "bayer_gb8"? Unfortunately this format is not nativly supported by ROS. 

With the command below I got the XML file from the camera
`arv-tool-0.4 -n "The Imaging Source Europe GmbH-32610040" genicam`
In the XML file I found this section:
```
<Enumeration Name="PixelFormat" NameSpace="Standard">
    <ToolTip>PixelFormat</ToolTip>
    <EnumEntry Name="Mono8">
      <DisplayName>Mono 8</DisplayName>
      <Value>0x01080001</Value>
    </EnumEntry>
    <EnumEntry Name="Mono16">
      <DisplayName>Mono16</DisplayName>
      <Value>0x01100007</Value>
    </EnumEntry>
    <EnumEntry Name="BayerGB8">
      <DisplayName>Bayer GB 8</DisplayName>
      <Value>0x0108000A</Value>
    </EnumEntry>
```
Am I correct that there is not other image format? If, yes, does the IC capture software on windows convert the stream to RGB32?

TIS-Stefan, 2016-08-18:
Hi Sven

you can try the arv-viewer. It provides a colored image. 

Unfortunately I do not know, how to get the line
from gi.repository import Aravis
working, since I do not know, how to install this Aravis import for Python2.7.  If I knew that, I would create the correct sample for you.
In C++ with a Gstreamer pipeline, I use the bayer2rgb in order to create a colored image from the raw data sent by the camera. I am not good in Python, I am better in C++. For that I have a sample code.

I found a pipeline in my samples:
`gst-launch-0.10 aravissrc name='The Imaging Source Europe GmbH-34510279"' ! video/x-raw-bayer,width=1280,height=960,framerate=15/1 ! queue !  bayer2rgb ! ffmpegcolorspace ! ximagesink sync=false
`

Stefan

sven-bock, 2016-08-18:
I think I can help you with the aravis import. I wrote a walkthrough for this. If you have already installed everything, you just need the exports below. I have also added the aravis libs to my PYTHONPATH but I think they are not required.

Install Aravis:
```
https://github.com/AravisProject/aravis
cd aravis
./autogen.sh --enable-introspection=yes --enable-gst-plugin --enable-gst-plugin-0.10 --enable-viewer
make
sudo make install
```
Install TIS camera:
git clone https://github.com/theimagingsource/tiscamera
cd tiscamera
```
./bootstrap.sh
./configure --enable-aravis
# if no errors continue
make
make install
```
Now the imporant environment setup (put this into your ~/.bashrc):
export ARV_HOME=YOURPATHTOARAVIS/aravis
# adjust gobject introspection search path to find the file Aravis-0.4.gir
export GI_TYPELIB_PATH=$ARV_HOME/src
# adjust runtime path for shared libraries to find libaravis-0.4.so
export LD_LIBRARY_PATH=$ARV_HOME/src/.libs:$LD_LIBRARY_PATH
export GST_PLUGIN_PATH=$ARV_HOME/gst-0.10/.libs:$GST_PLUGIN_PATH

sven-bock, 2016-08-18:
Your pipeline recommendation actually works. The interessting part is that I get the same error as for your python sample file.
`(gst-launch-0.10:5318): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'
`
Apparently this does not influence the result.
Now I need to understand this line =)

TIS-Stefan, 2016-08-18:
Hi Sven

the Aravis import works. Program starts, I get the same errors, you describe. But I do not see a video. The Python script uses the xvimagesink, while the commandline uses the ximagesink. Which is a significant change on my computer, since Ubuntu failed to install an NVidia driver. However, no image right now from the Python script, but I am investigating.

Stefan

sven-bock, 2016-08-18:
Yes, I have the same problem within the python script.

-- 

*Cordialement / Best Regards / Mit freundlichen Grüßen / Sven Bock
M.Sc.Research Engineer*
*Tel: +33 556 393 705*
*Tel: +43 316 39 77 76*

*Génération Robots <http://www.generationrobots.com/> | HumaRobotics
<http://www.humarobotics.com/>*
*DISCOVER ADEPT...*
<http://www.generationrobots.com/en/348-adept-mobile-robots>
*The world's most popular mobile robots **for education and research*
<http://www.generationrobots.com/en/348-adept-mobile-robots>

TIS-Stefan, 2016-08-18:
Hi Sven

I have no idea, why the sample does not run as expected. I must ask my Linux expert tomorrow.

Stefan

sven-bock, 2016-08-18:
That would be very nice.
Thank you Stefan!

-- 

*Cordialement / Best Regards / Mit freundlichen Grüßen / Sven Bock
M.Sc.Research Engineer*
*Tel: +33 556 393 705*
*Tel: +43 316 39 77 76*

*Génération Robots <http://www.generationrobots.com/> | HumaRobotics
<http://www.humarobotics.com/>*
*DISCOVER ADEPT...*
<http://www.generationrobots.com/en/348-adept-mobile-robots>
*The world's most popular mobile robots **for education and research*
<http://www.generationrobots.com/en/348-adept-mobile-robots>

Rouniy, 2017-06-15:
>> Hi Sven
>> I have no idea, why the sample does not run as expected. I must ask my Linux expert tomorrow.
>> Stefan
Did you have time to do that ?
Looks like it still doesn't works in some cases.

closed 2018-04-12 by TIS-Stefan