TheImagingSource/tiscamera #72: showcamera example does not work on Ubuntu 16.04

bobismijnnaam, 2016-09-19:
Hi everyone,

I'm trying to run the showcamera program with a DFK 23UX174 on Ubuntu 16.04, but I cannot get it to work. The showcamera program lists the camera as an option, but when I press start it says the following:

```
Camera Change DFK 23UX174 45510106
Pipeline build start
The pipeline element "bayer2rgb", "bayer" could not be created.

(showcamera:14497): GStreamer-CRITICAL **: passed '0' as denominator for `GstFraction'

** (showcamera:14497): CRITICAL **: Unable to link source pipeline to autoexposure
Module "image_sink" is null.

(showcamera:14497): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Module "image_sink" is null.

(showcamera:14497): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(showcamera:14497): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Playing
Dialog Destructor calleectord
End of Program
```

The screen just stays the same, as in the image. 

[image]

It also lists very little properties:

[image]

```
/dev/video1 The Imaging Source Europe GmbH DFK 23UX174 45510106
Checking /dev/video1
Property id 0x980900 name Brightness min: 0, max: 511, default 240, type 1
Property id 0x980913 name Gain min: 0, max: 480, default 0, type 1
Property id 0x9a0902 name Exposure (Absolute) min: 1, max: 300000, default 333, type 1
End
```

Contrary to my built in usb webcam, which has very bad quality but has 4 times the properties. I followed the "Getting Started with USB 2.0 cameras on a Raspberry PI or on other Linux Systems" to the letter, except for one difference. During installation of the dependencies, apt-get gave the following error:

```
Package gstreamer0.10-plugins-bad is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  gstreamer0.10-plugins-good:i386 gstreamer0.10-plugins-base:i386 gstreamer0.10-plugins-good
  gstreamer0.10-plugins-base

E: Package 'gstreamer0.10-plugins-bad' has no installation candidate
```

I did what apt-get told me, removed the gstreamer-bad dependency, and put the gstreamer-good variants in the call. It then stopped complaining and installed all the listed dependencies. Building also gave no errors, both in the src directory and in the showcamera directory. The bootstrap, configure, make, make install, cmake .., make, all went flawlessly.

Is there any way to get the camera to work correctly on Ubuntu 16.04? Or did I miss a specific instruction?

bobismijnnaam, 2016-09-22:
No response? We are quite desperate :(

TIS-Edgar, 2016-09-22:
Hi,

Sorry for the late reply. From what I can tell you did not miss an instruction.

First lets make sure the camera works correctly.
A standard application like qv4l2 should display the camera(you will have to open /dev/video1).

The problem you are seeing is that showcamera tries to display a color image which requires debayering which requires the bayer2rgb module which is in the gstreamer0.10-bad package.

But you probably already knew that...

There are a few possible solutions here:

1. Install the package manually from the previous release. Get the package from the bottom of the page: http://packages.ubuntu.com/wily/gstreamer0.10-plugins-bad
Install with:
    sudo dpkg -i <package>
    sudo apt-get -f install #installs dependencies
2. build from source
Get the tarball from the gstreamer repository and compile from source.

I will look into a more elegant way of solving this....

bobismijnnaam, 2016-09-22:
Thanks for your reply! I will investigate!

bobismijnnaam, 2016-09-22:
Okay, I think I have the gst-launch invocations working now with a nice and colorful camera image. Had to install three packages:

https://launchpad.net/ubuntu/wily/amd64/libgstreamer-plugins-bad0.10-0/0.10.23-8.1ubuntu3
https://answers.launchpad.net/ubuntu/utopic/amd64/libsoundtouch0/1.8.0-1
http://packages.ubuntu.com/wily/libvpx2

Then `sudo apt-get install --fix-missing` and the gst-launch invocations works. I also have a lot more parameters in the `showcamera` executable. The showcamera does not show any camera however. The error is the following:

```
...3 pages of parameter settings omitted...
Pipeline build start

(showcamera:24905): GStreamer-CRITICAL **: passed '0' as denominator for `GstFraction'

** (showcamera:24905): CRITICAL **: Unable to link source pipeline to autoexposure
Playing
End thread 

(showcamera:24905): GLib-CRITICAL **: g_source_remove: assertion 'tag > 0' failed
Dialog Destructor calleectord
End of Program
```

Any ideas?

TIS-Edgar, 2016-09-22:
The problem is that the indexing of the video formats is faulty.
For some reason the indexing is not done via fourcc as it should be but through the description text. That has changed in a recent kernel version, thus breaking it.

I have pushed a patch that should fix your issue. 
Pull the change and rebuild. Let me know if this issue then still exists...

bobismijnnaam, 2016-09-22:
If I open a camera image with gst-launch, and then fiddle with the parameters in showcamera, some work, some don't. Basic stuff like gain and exposure works. Turning the white balance on and off and red/green/etc. doesn't. If I remove tis-whitebalance filter from the gst-launch, turning the white balance on and off in the showcamera dialog does not make a difference either.

I will pull and test!

bobismijnnaam, 2016-09-22:
Yes! Showcamera works properly now! I'm not sure how white balance works (when you disable it the red/green/blue sliders work once you click on them?) but I think that's because of my lacking knowledge of optics, haha. Thanks for the quick help!

TIS-Edgar, 2016-09-22:
No problem. The whitebalance module has two 'disable' settings the first disables the automatic correction allowing you to manually set the correction values that the whitebalance module applies. The second completely disables the module giving the image the distinct green look of a bayer image.

If you encounter any further problems, please let me know.

closed 2016-11-16 by TIS-Edgar