TheImagingSource/tiscamera #76: auto-exposure up to 30 sec and tcamwhitebalance/bayer2rgb

fehlfarbe, 2017-01-16:
Hi,

I tried the gstreamer-0.10 pipeline with auto exposure up to 30 seconds some months ago on my Linux system and it worked without problems.
But now I have an ARM system with Ubuntu 16.04 where I can't install gstreamer-0.10-plugins-bad.

So I tried gstreamer-1.0 with DFK 72AUC02 in my almost dark room:

`gst-launch-1.0 tcamsrc ! video/x-raw,width=2592,height=1944,framerate=6/1 ! tcamautoexposure auto-exposure=true exposure-max=300000 ! jpegenc ! avimux ! filesink location=video1.avi`

The exposure is under 1 sec and the image very dark (only some bright spots).

When I'm adding whitebalance and bayer2rgb to create color images I also get errors:

`gst-launch-1.0 tcamsrc ! video/x-raw,width=2592,height=1944,framerate=6/1 ! tcamwhitebalance ! tcamautoexposure auto-exposure=true exposure-max=300000 ! bayer2rgb ! jpegenc ! avimux ! filesink location=video1.avi`

`WARNING: erroneous pipeline: could not link tcamsrc0 to tcamwhitebalance0`

same error, when I change video/x-raw to video/x-bayer

What's wrong with my pipelines?

TIS-Stefan, 2017-01-17:
Hello

You should use
`gst-launch-1.0 tcamsrc ! video/x-bayer,width=2592,height=1944,framerate=6/1 ! tcamwhitebalance ! tcamautoexposure auto-exposure=true exposure-max=300000 ! bayer2rgb ! jpegenc ! avimux ! filesink location=video1.avi`

Stefan

fehlfarbe, 2017-01-17:
Hi Stefan,

I tried this but I always get the error message:
`WARNING: erroneous pipeline: could not link tcamsrc0 to tcamwhitebalance0`

TIS-Stefan, 2017-01-17:
Hi

Strange. I use the pipeline

`sprintf(pipeline, "tcamsrc serial=\"%s\" name=tcambin ! video/x-bayer,width=%d,height=%d,framerate=%d/%d ! tcamautoexposure ! tcamwhitebalance ! bayer2rgb ! videoconvert ! xvimagesink", serial, width,  height,  fpsnominator,  fpsdenomitor ); `

No problem here. Can you try that similar?

Stefan

fehlfarbe, 2017-01-17:
Hi Stefan,

I tried on my ARM system:
`gst-launch-1.0 -vv tcamsrc serial=xxxxxxx name=tcambin ! video/x-bayer,width=2592,height=1944,framerate=6/1 ! tcamautoexposure ! tcamwhitebalance ! bayer2rgb ! videoconvert ! xvimagesink`
and get
`WARNING: erroneous pipeline: could not link tcambin to tcamautoexposure0`

I tried the same pipeline on my Linux Mint 17.1 Notebook (Ubuntu 14.04 64 bit), Udoo Neo (ARM, clean install) Ubuntu 14.04 and Raspberry Pi (Raspbian Jessie, clean install) and on these systems I get:
`terminate called after throwing an instance of 'std::bad_function_call'
  what():  bad_function_call
Aborted`

Replacing video/x-bayer with video/x-raw moves the error to autoexposure->whitebalance:
`WARNING: erroneous pipeline: could not link tcamautoexposure0 to tcamwhitebalance0`

I installed all packages from README.

The only working way to get image data is:
`gst-launch-1.0 tcamsrc name=tcambin ! video/x-raw,width=2592,height=1944,framerate=6/1 ! tcamautoexposure ! jpegenc ! avimux ! filesink location=video1.avi`
on my Ubuntu 16.04 ARM system (headless, so I have to save the image data). But only with bayer pattern and without auto-exposure up to 30s.

TIS-Stefan, 2017-01-18:
Hello

one point is: the framerate 6/1 does not exist. Try 7/1 or 5/1 instead.

Make sure, you use the correct firmware. If you are in the tiscamera directory, perform
```
cd build/tools/firmwareupdate
sudo ./firmware-update -ud <insert serial number here> -f ../../../data/firmware/usb2/dfk72uc02_3012.euvc 
```
After doing so, the pipeline is:
`gst-launch-1.0 -vv tcamsrc serial=<insert serial number here>! video/x-bayer,width=2592,height=1944,framerate=5/1 ! tcamautoexposure auto-exposure=true exposure-max=300000 ! tcamwhitebalance ! bayer2rgb ! videoconvert ! ximagesink`

The tcambin generates color images, if you pass "video/x-raw, format=RGBx" instead of "video/x-bayer". But you are right, you can not configure the auto exposure properties from command line. This can be done in a program or script, that queries a pointer to the camera properties from the pipeline.

fehlfarbe, 2017-01-19:
Thanks for the pointer! I updated the Firmware - same error but when I change the framerate to 5/1 the pipeline is working.
But auto-exposure goes only up to 200ms or so. I started the pipeline with debug output and saw that the exposure time goes only up to 200ms. In Source code I found the line: https://github.com/TheImagingSource/tiscamera/blob/master/src/gstreamer-1.0/gsttcamautoexposure.cpp#L1066 where the maximum exposure time is limited to get a constant framerate. I deleted the line and recompiled. Now the output says the exposure time goes up to 30s:

`0:06:43.370952627  5757  0x1cb76c0 INFO        tcamautoexposure gsttcamautoexposure.cpp:1314:correct_brightness: offset = 119.000000, gain = 0.000000, exposure = 30000000.000000`

But the image is way too dark and without hotpixels so I think there is another function that limits the exposure time to get a constant framerate. Same problem without the framerate parameter. So is there a way to get a variable framerate based on current exposure time? I think the gstreamer0.10 pipeline worked this way even with a given framerate.
Setting the absolute exposure time with `v4l2-ctl -c exposure_absolute=100000` while grabbing images with OpenCV works but it disables the autoexposure. If gstreamer autoexposure won't work, I'll try to make my own exposure algorithm with OpenCV and v4l2-ctl.

And there seems to be a bug with the gain property. When I start the pipeline with debug and grep all messages from gsttcamautoexposure.cpp there is an error at every frame:

`0:06:39.724436633  5757  0x1cb76c0 INFO        tcamautoexposure gsttcamautoexposure.cpp:1285:retrieve_image_region: Region is from 0 0 to 2592 1944
0:06:39.724485882  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1300:correct_brightness: Calculating brightness
0:06:39.725576437  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1219:retrieve_current_values: Current exposure is 0
0:06:39.725748557  5757  0x1cb76c0 ERROR       tcamautoexposure gsttcamautoexposure.cpp:1227:retrieve_current_values: Tcam did not return gain
0:06:39.725849596  5757  0x1cb76c0 INFO        tcamautoexposure gsttcamautoexposure.cpp:1314:correct_brightness: offset = 119.000000, gain = 0.000000, exposure = 30000000.000000
0:06:39.725941677  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1331:correct_brightness: gain
0:06:39.725989384  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1192:modify_exposure: Returning (e_ref - new_exposure = diff) 35266437.188508 - 30000000.000000 = 5266437.188508
0:06:39.726055299  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1151:modify_gain: fabs(self->gain.value - new_gain) > K_SMALL == 0.000000 - 30.000000 =30.000000 (g_ref 526643.718851 diff 5266437.188508)
0:06:39.726125589  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1155:modify_gain: Comparing percentage_new 100.000000 > percentage 0.300000
0:06:39.726176504  5757  0x1cb76c0 INFO        tcamautoexposure gsttcamautoexposure.cpp:1106:set_gain: Setting gain to 0.300000
0:06:39.726280543  5757  0x1cb76c0 DEBUG       tcamautoexposure gsttcamautoexposure.cpp:1171:modify_gain: NEW GAIN: g_ref 526643.718851 - new_gain 30.000000 / K_GAIN 3.000000 = 175537.906284`

Same when I list the v4l2 parameters:

`pi@nanopiair:~$ v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : DFK 72AUC02
	Bus info      : usb-sunxi-ehci-1
	Driver version: 1.1.1
	Capabilities  : 0x04000001
		Video Capture
		Streaming
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 2592/1944
	Pixel Format      : 'GREY'
	Field             : None
	Bytes per Line    : 2592
	Size Image        : 5038848
	Colorspace        : Default
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 2592, Height 1944
	Default     : Left 0, Top 0, Width 2592, Height 1944
	Pixel Aspect: 1/1
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 7.500 (15/2)
	Read buffers     : 0
error 22 getting ctrl Brightness
error 22 getting ctrl Gain
error 22 getting ctrl Trigger Global Reset Shutter
              exposure_absolute (int)    : min=1 max=300000 step=1 default=127 value=127
                 focus_absolute (int)    : min=0 max=1000 step=1 default=0 value=0
                        privacy (bool)   : default=0 value=0
                        trigger (bool)   : default=0 value=0
               software_trigger (button) :
                          gainr (int)    : min=0 max=63 step=1 default=1 value=36
                          gaing (int)    : min=0 max=63 step=1 default=1 value=36
                          gainb (int)    : min=0 max=63 step=1 default=1 value=36
                        binning (int)    : min=1 max=4 step=1 default=1 value=1
                       x_offset (int)    : min=0 max=1794 step=2 default=0 value=0
                       y_offset (int)    : min=0 max=1464 step=2 default=0 value=0`

TIS-Stefan, 2017-01-20:
Hello

I also found out, the tcamautoexpsoure limits the exposure time in an unwanted way. Since you found the line in the code, where this happens, you may change the line, compile and see, what happens.

Regarding the other problem we have to look for. Since my Linux programmer is very busy, you may look into the source on your own?

Stefan

fehlfarbe, 2017-01-31:
Hi Stefan,

I changed some lines in `gsttcamautoexposure.cpp` to set an static exposure time of several seconds but it didn't work. Then I tried to change the framerate to 0.1 Hz in `gsttcamsrc.cpp` but this didn't work either because the software selects a valid framerate from the videoformatdescription list. So I added a value of 0.1 in `https://github.com/TheImagingSource/tiscamera/blob/master/src/gstreamer-1.0/gsttcamsrc.cpp#L614` which causes camera error and stops the pipeline.

After this I uninstalled my changes and tried to clone and reinstall the clean repo. Now I get the same error like my other (clean installed) devices when I start a gstreamer pipeline or the tcam-ctrl binary:
```
pi@orangepizero:~$ tcam-ctrl -l
terminate called after throwing an instance of 'std::bad_function_call'
  what():  bad_function_call
Aborted
```
(Armbian Ubuntu 16.04; Linux orangepizero 3.4.113-sun8i #50 SMP PREEMPT Mon Nov 14 08:41:55 CET 2016 armv7l armv7l armv7l GNU/Linux)

TIS-Stefan, 2017-02-01:
Hi

If you try to set values, that are not supported by the camera, then you must expect, that you wont get usable results. The video format description list is read from the camera to make sure, we have no invalid formats and frame rates.

The bad function call should be fixed with latest files from the git repository, it was checked in a few hours after you cloned the repository. So a "git pull", "cd build", "make -j" and "sudo make install" should do the job 
(I just tested that on a fresh installed Ubunut 16.04)

Stefan

closed 2017-02-01 by TIS-Stefan