TheImagingSource/tiscamera #79: Code to interact with UVC camera extension unit controls?

ronbarak, 2017-02-23:
I have an UVC camera, and I would like to manipulate its _extended controls_.
D'you knows of code snippets* demonstrating how to interact with UVC extension unit controls - using OpenCV?

---
* Preferably in Python.

TIS-Stefan, 2017-02-23:
Which camera do you have?

closed 2017-05-15 by TIS-Edgar

chaitanyantr, 2019-08-14:
i have see3cam UVC Compliant camera....I want to disable the AFrectangle in it .

The response i get from company is

 "**Dear Chaitanya,

Thanks for sharing the details

Regarding your query, autofocus rectangle control you can't access through v4l2-ctl, access these controls through HID control only "**

TIS-Stefan, 2019-08-15:
Dear Chaitanya,

Thank you for letting us know. For samples how to program with your camera, I suggest, as you already did, asking the camera manufacturer, which is not The Imaging Source. Also you may find v4l2 samples in the internet, that show, how to enumerate, get and set camera properties. 

I am very sorry, but we can not help with your camera of another company.

Stefan