TheImagingSource/tiscamera #81: Problem executing the pipline

FarzanZaheer, 2017-03-06:
I want to use my DFK 72AUC02 camera with raspberry pi 3. I have followed the instructions, but when I execute the pipline the following error shows up.

pi@MusIPi:~ $ gst-launch-0.10 v4l2src ! video/x-raw-gray,width=640,height=480,framerate=15/1 ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! ximagesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstXImageSink:ximagesink0: Could not initialise X output
Additional debug info:
ximagesink.c(1284): gst_ximagesink_xcontext_get (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0:
Could not open display
Setting pipeline to NULL ...
Freeing pipeline ...

My main goal is to connect the DFK 72AUC02 camera with raspberry pi 3, and access it in Matlab (with raspberry pi hardware support package already installed). DFK 72AUC02 camera is detected in Matlab but I can't take a snapshot.

TIS-Stefan, 2017-03-06:
Hello

"Could not open display" says all. Do you run a window manager (graphical UI) on your PI? If not, you may use the "aasink" instead of "ximagesink".

FarzanZaheer, 2017-03-07:
Hello
I think following the wiki insructions I have installed gstreamer 0.10. Nevertheless, I replaced the "ximagesink" with the "aasink", the gray scale stream shows up.
I am doing all this to install the required drivers or firmware on my raspberry, so I can connect my DFK 72AUC02 camera to my raspberry pi 3 and use it in Matlab (with raspberry hard ware support package installed). 
I can create a raspi object in Matlab with
`mypi = raspi();`
then I can also create the webcam object with
`wcam = webcam(mypi);`
The webcam object even shows the name of DFK 72AUC02 camera (DFK 72AUC02 (usb-3f980000.usb-1.2):), but when I try to take a snapshot with
`img = snapshot(wcam);`
The following error shows up in Matlab:

> Error using raspi/recvResponse (line 649)
> Error receiving data from the remote server.
> Additional Information: timeout.
> 
> Error in raspi.internal.webcam/snapshot (line 132)
>             data = typecast(obj.Hw.recvResponse(),'uint8');

TIS-Stefan, 2017-03-07:
Hmm. Sounds for a Matlab issue. I do not know, what the Matlab code does.  Thus I would ask the Matlab support.

FarzanZaheer, 2017-03-07:
There is no seperate window. Using the "aasink" the gray scale stream shows up in the same shell script. I am totally fine with it, if it means that the DFK 72AUC02 camera is working perfectly on the raspberry pi because all I want to do is to access the camera in Matlab.
The Matlab statement "wcam = webcam(mypi)" creates an object containing parameters name, resolution, and available resolutions of the camera conected to raspberry pi. It wil be great if some one from Matlab support could assist.

TIS-Stefan, 2017-03-07:
I simply conclude, you have no X installed.  Without that graphical user interface ximagesink can not display anything. Therefore the funny aasink works

I do not think, Matlab support looks in here. You must contact them directly.

FarzanZaheer, 2017-04-05:
Hi, 
I am stuck at "Could not open display". Can you suggest any UI with which I can view my camera video stream. I am accessing my raspberry pi from putty. Do I need to connect my raspberry pi to a monitor to view the stream? Sorry, my questions might sound stupid I am new to this.

TIS-Stefan, 2017-04-05:
As I already wrote: Use "aasink" instead of "ximagesink":

gst-launch-0.10 v4l2src ! video/x-raw-gray,width=640,height=480,framerate=15/1 ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! aasink

Then you will get ascii graphics in putty.

"Could not open display": Of  course, there is no graphical environment, where that could have been opened.

FarzanZaheer, 2017-06-12:
Hi,
I am using the following pipeline on Raspberry Pi with the camera DFK72AUCO2
`gst-launch-0.10 v4l2src ! video/x-raw-gray,width=640,height=480,framerate=15/1 ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! queue ! tiswhitebalance ! queue ! bayer2rgb ! queue ! ffmpegcolorspace ! ximagesink`

Every thing works perfect but when I change the resolution to 1280x960 or a lower resolution like 320x240, I get the follwoing error
```
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Could not negotiate format
Additional debug info:
gstbasesrc.c(2830): gst_base_src_start (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Check your filtered caps, if any
Setting pipeline to NULL ...
Freeing pipeline ...
```
Using 2592x1944 resolution is not giving any error but I know it is too much for Raspberry Pi 3 to handle. I want to know if there is a way I can use the resolution  1280x960? Even changing the framerate while using the 640x480 is producing format errors. Can you please help me in this regard or provide me a pipeline in which I can use 1280x960 resolution?

FarzanZaheer, 2017-07-03:
Hi, 
Can you provide me an example pipeline for capturing an image with DFK 72AUCO2?

StefanHope, 2019-04-04:
Hello, I have changed a camera which is DMK 23GP031. Got those error!
------------------------------------------------------------------------------
pi@raspberrypi:~ $ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)400000/57327, width=(int)2592, height=(int)1944, format=(string)GRAY16_LE
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Device lost
Additional debug info:
/home/pi/Desktop/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp(1045): gst_tcam_src_device_lost_callback (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source
Execution ended after 0:00:03.493881237
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
------------------------------------------------------------------------

TIS-Arne, 2019-04-04:
Created a new issue request. Closing this one because it is way old.

closed 2019-04-04 by TIS-Arne

Kunj6354, 2023-02-23:
[image]
I am using a banana pi and HDF5640 camera. I changed video0/1/2 but only video1 is showing the correct value. so the video port is 1.

TIS-Stefan, 2023-02-24:
@Kunj6354  I am afraid, you are in the wrong issue.

Kunj6354, 2023-02-24:
Anything you know about it?

TIS-Stefan, 2023-02-24:
No. If there is one camera only, then only one of the /dev/vid* should work.