TheImagingSource/tiscamera #83: Error connecting to DFKZ12G445

David-Estevez, 2017-03-14:
Hi! I've followed all the steps in this repository's wiki, but I have been unable to use the DFKZ12G445 GigE camera on my Linux Mint 17.3 (virtually equivalent to a Ubuntu Trusty distro) PC.

The same camera works well on Windows, I can receive images and modify parameters of the camera, although IC Capture sometimes freezes completely and becomes unresponsive until a reboot is performed.

On Linux, I am able to detect the camera:
```bash
user@pc ~ $ camera-ip-conf-cli -l

       Model -    Serial -  User Def. Name -      Current IP - Current Netmask - Current Gateway
 DFK Z12G445 -  03610021 -                 -   169.254.81.79 -     255.255.0.0 -         0.0.0.0
```

But then when I try to use any arv-tool, an error is obtained:

```bash
user@pc ~ $ arv-tool-0.4 --name="The Imaging Source Europe GmbH-03610021" features
The Imaging Source Europe GmbH-03610021
Device 'The Imaging Source Europe GmbH-03610021' not found
```

When I try to obtain a stream from the camera, another error happers:

```bash
user@pc ~ $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! video/x-raw-gray,width=1920,height=960,framerate=30/1 ! fakesink silent=false
ERROR: Could not load classifier cascade /usr/share/opencv/haarcascades/haarcascade_frontalface_alt2.xml

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to PAUSED ...

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_stream_timeout_pop_buffer: assertion 'ARV_IS_STREAM (stream)' failed
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming task paused, reason error (-5)
Execution ended after 323856 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

(gst-launch-0.10:3687): Aravis-CRITICAL **: arv_camera_stop_acquisition: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to NULL ...
Freeing pipeline ...
```

I'm new to aravis, gstreamer and GigE cameras, so currently I'm totally clueless. :sweat_smile:  Any help with this would be much appreciated.

TIS-Stefan, 2017-03-14:
Hello

the used video format width and height are wrong. The camera has 1280 px width and 960px height. You use 1920 px width. I suppose, this is the reason for no stream. 

Stefan

David-Estevez, 2017-03-14:
I've tried the same command with the correct video format  (1280x960px) and it is still not working:

```bash
user@pc ~ $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! video/x-raw-gray,width=1280,height=960,framerate=30/1 ! fakesink silent=false

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to PAUSED ...

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_stream_timeout_pop_buffer: assertion 'ARV_IS_STREAM (stream)' failed
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming task paused, reason error (-5)
Execution ended after 193863 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

(gst-launch-0.10:7392): Aravis-CRITICAL **: arv_camera_stop_acquisition: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to NULL ...
Freeing pipeline ...
```

TIS-Stefan, 2017-03-14:
Hmm. Does the arv-viewer work with the camera?

Stefan

David-Estevez, 2017-03-14:
When I run the arv-viewer the camera appears in the combobox, but when I select it the image is not correctly received (see screenshot):

[image]

About the tisvideobufferfilter, it seems that I don't have that one installed: 

```bash
user@pc ~ $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! tisvideobufferfilter ! video/x-raw-gray,width=1280,height=960,framerate=30/1 ! fakesink silent=false

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:9052): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
WARNING: erroneous pipeline: no element "tisvideobufferfilter"
```

Did I miss something when installing the camera toolchain?  Do I have to install it manually?

TIS-Stefan, 2017-03-14:
Maybe it is Jumbo Frames.....
sudo ipconfig eth0 mtu 9000

And sorry for the delayed answer.

Stefan

David-Estevez, 2017-03-15:
Hi Stefan, sorry for taking this long to reply, I had a pretty busy morning.

I have tried arv-viewer with the command you said to enable Jumbo Frames and there weren't any changes. The camera was selectable, but selecting it did not cause the images to appear. In addition, trying to change any parameter such as the framerate (setting it to 30, for example), causes the terminal to display the following error messages:
```bash
(arv-viewer:6321): Aravis-CRITICAL **: arv_camera_set_frame_rate: assertion 'ARV_IS_CAMERA (camera)' failed

(arv-viewer:6321): Aravis-CRITICAL **: arv_camera_get_frame_rate: assertion 'ARV_IS_CAMERA (camera)' failed

(arv-viewer:6321): Aravis-CRITICAL **: arv_camera_set_frame_rate: assertion 'ARV_IS_CAMERA (camera)' failed

(arv-viewer:6321): Aravis-CRITICAL **: arv_camera_get_frame_rate: assertion 'ARV_IS_CAMERA (camera)' failed
```

I also tried compiling and installing the gstreamer-0.10 branch, and even though the tisvideobufferfilter not found error has disappeared, the rest of the errors are still appearing:

```bash
user@pc ~/opt/tiscamera/src $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! tisvideobufferfilter ! video/x-raw-gray,width=1280,height=960,framerate=30/1 ! fakesink silent=false

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to PAUSED ...

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_gain: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_is_gain_available: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_exposure_time: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_exposure_time_auto: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_region: assertion 'ARV_IS_CAMERA (camera)' failed

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_get_binning: assertion 'ARV_IS_CAMERA (camera)' failed
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_stream_timeout_pop_buffer: assertion 'ARV_IS_STREAM (stream)' failed
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming task paused, reason error (-5)
Execution ended after 348079 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...

(gst-launch-0.10:6024): Aravis-CRITICAL **: arv_camera_stop_acquisition: assertion 'ARV_IS_CAMERA (camera)' failed
Setting pipeline to NULL ...
Freeing pipeline ...
```

TIS-Stefan, 2017-03-15:
Hmm.  From the error message "arv_camera_set_frame_rate: assertion 'ARV_IS_CAMERA (camera)' failed" I would think, the camera disconnected. 
Does the green LED lights on the camera? If not, you have not gige connection and wont get images. I am just guessing, since I am running out of ideas.

Which computer model do you use? 
Which Kernel version do you use?
Is the camera connected to eth0 or do you have more network controllers?
Is there a firewall, that may blocks packages?

David-Estevez, 2017-03-16:
The green LED is on in both connectors.

About the remaining questions:

* PC: Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz, 16GB RAM, NVIDIA GeForce GTX 960
* Network adapter: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller  (Realtek Semiconductor Co., Ltd.). 
  * Capabilities: bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
   * Configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=2.3LK-NAPI duplex=full firmware=rtl8168g-2_0.0.1 02/06/13 ip=163.117.150.250 latency=0 link=yes multicast=yes port=MII speed=100Mbit/s
    * resources: irq:27 ioport:d000(size=256) memory:f7100000-f7100fff memory:f2100000-f2103fff
* Kernel version:  3.19.0-32-generic
* eth0, no other network controller available.
* Firewall status checked, it is disabled.

TIS-Stefan, 2017-03-16:
Do you have an idea, what "speed=100Mbit/s" means? Is it possible, your network controller is limited to 100mBit? Please try

dmesg | grep -i duplex

If this shows something like 1000 Mbps, then the controller is fine. If not, then you may found the reason for the problem.

David-Estevez, 2017-03-16:
I guess it was the speed of the current connection (I am currently swapping between the camera and the internet connection, as I only have one network adapter on the PC). Every output I'm sending you now is with the camera connected.

The command you suggested didn't print any output, but this was the output of the `sudo ethtool eth0` command: 

```
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00000033 (51)
			       drv probe ifdown ifup
	Link detected: yes
```

If also attaching the Network Adapter info with the camera connected, in case it is useful:
```
network
        description: Ethernet interface
        product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
        vendor: Realtek Semiconductor Co., Ltd.
        physical id: 0
        bus info: pci@0000:03:00.0
        logical name: eth0
        version: 0c
        serial: 2c:56:dc:3b:7b:63
        size: 1Gbit/s
        capacity: 1Gbit/s
        width: 64 bits
        clock: 33MHz
        capabilities: bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
        configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=2.3LK-NAPI duplex=full firmware=rtl8168g-2_0.0.1 02/06/13 ip=163.117.150.250 latency=0 link=yes multicast=yes port=MII speed=1Gbit/s
        resources: irq:27 ioport:d000(size=256) memory:f7100000-f7100fff memory:f2100000-f2103fff
```

TIS-Stefan, 2017-03-17:
And do you see images from the camera now?

Stefan
-

David-Estevez, 2017-03-17:
I think I didn't explain myself very well. I was referring to your question about what "speed=100Mbit/s" meant, I didn't do any other changes. I just run those commands with the camera attached to provide you with more info about the adapter.

So no, still no images :disappointed:

TIS-Edgar, 2017-03-17:
Are you in the same subnet as the camera? In your initial post the camera has 169.254.81.79 which is link local. That can cause issues with stream but not detection.

David-Estevez, 2017-03-17:
Bingo! That was the problem with the connection, my bad.  :sweat: 

After configuring the IPs to be in the same subnet, I can connect with the camera. I have tested the connection with arv-viewer, and I have successfully obtained an image. I can also modify with arv-tool the parameters of the camera such as the Gain or the Zoom.

When running with gst-lauch, I still have some errors, but I think they are related to the sink rather than the source:
1. With gst-launch and a fakesink (I terminated it with cntrl+c after a while):
``` 
user@pc ~ $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! tisvideobufferfilter ! video/x-raw-gray,width=1280,height=960,framerate=30/1 ! fakesink silent=false
Setting pipeline to PAUSED ...

(gst-launch-0.10:6994): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'

(gst-launch-0.10:6994): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'

(gst-launch-0.10:6994): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Terminated
```

2. With gst-launch and ximagesink
```
user@pc ~ $ gst-launch-0.10 --gst-plugin-load=/usr/local/lib/gstreamer-0.10/libgstaravis-0.4.so aravissrc camera-name='The Imaging Source Europe GmbH-03610021' num-buffers=1 ! tisvideobufferfilter ! video/x-raw-gray,width=1280,height=960,framerate=30/1 ! ffmpegcolorspace ! ximagesink
Setting pipeline to PAUSED ...

(gst-launch-0.10:7063): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'

(gst-launch-0.10:7063): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'

(gst-launch-0.10:7063): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV accessing address 0x7f2db9307010
Spinning.  Please run 'gdb gst-launch 7063' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

Btw, thank you to both of you for your support, if we meet sometime I'll have to invite you to a beer (or coffee)! :blush:

closed 2017-08-10 by TIS-Arne