TheImagingSource/tiscamera #84: Compile error: can't resolve libraries to shared libraries: tcamprop

rleitner, 2017-03-15:
Hi,
I'm facing a problem compiling the current master branch with Yocto Project Jethro 2.0.3 for an ARM platform. During CMake I'm getting the following error message:
```
[ 88%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamsrc.dir/gsttcamsrc.cpp.o
cd /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/gstreamer-1.0 && /build/tmp/sysroots/x86_64-linux/usr/bin/arm-my-distro-linux-gnueabi/arm-my-distro-linux-gnueabi-g++   -DHAVE_USB -Dgsttcamsrc_EXPORTS -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9  --sysroot=/build/tmp/sysroots/my-machine  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -std=c++11 -pthread  -w -Wall -Wextra -pedantic -Wundef -Wshadow -fPIC -Wl,--gc-sections -ffunction-sections -fdata-sections -fpermissive -fPIC -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/gstreamer-1.0 -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gstreamer-1.0 -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build -I/build/tmp/sysroots/my-machine/usr/include/gstreamer-1.0 -I/build/tmp/sysroots/my-machine/usr/include/glib-2.0 -I/build/tmp/sysroots/my-machine/usr/lib/glib-2.0/include -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gobject -I/build/tmp/sysroots/my-machine/usr/include/gobject-introspection-1.0 -I/build/tmp/sysroots/my-machine/usr/lib/libffi-3.2.1/include    -o CMakeFiles/gsttcamsrc.dir/gsttcamsrc.cpp.o -c /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gstreamer-1.0/gsttcamsrc.cpp
g-ir-scanner: compile: arm-my-distro-linux-gnueabi-gcc -march=armv7-a -marm -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 --sysroot=/build/tmp/sysroots/my-machine -Wno-deprecated-declarations -pthread -I/build/tmp/sysroots/my-machine/usr/include/glib-2.0 -I/build/tmp/sysroots/my-machine/usr/lib/glib-2.0/include -O2 -pipe -g -feliminate-unused-debug-types -I/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src -I/build/tmp/sysroots/my-machine/usr/include/glib-2.0 -I/build/tmp/sysroots/my-machine/usr/lib/glib-2.0/include -c -o /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gobject/tmp-introspectGIezfr/Tcam-0.1.o /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gobject/tmp-introspectGIezfr/Tcam-0.1.c
g-ir-scanner: link: arm-my-distro-linux-gnueabi-gcc -march=armv7-a -marm -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 --sysroot=/build/tmp/sysroots/my-machine -o /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gobject/tmp-introspectGIezfr/Tcam-0.1 -O2 -pipe -g -feliminate-unused-debug-types -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/gobject/tmp-introspectGIezfr/Tcam-0.1.o -L. -Wl,-rpath=. -Wl,--no-as-needed -ltcamprop -L/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/gobject -Wl,-rpath=/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/gobject -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
ERROR: can't resolve libraries to shared libraries: tcamprop
```

The included `build/src/gobject` directory contains the "missing" library:
```
$ ls -l /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/gobject
drwxr-xr-x 4 leri leri  4096 Mar 15 09:37 CMakeFiles
-rw-r--r-- 1 leri leri  4889 Mar 15 09:37 cmake_install.cmake
lrwxrwxrwx 1 leri leri    18 Mar 15 09:37 libtcamprop.so -> libtcamprop.so.0.1
-rwxr-xr-x 1 leri leri 22980 Mar 15 09:37 libtcamprop.so.0.1
-rw-r--r-- 1 leri leri 11023 Mar 15 09:37 Makefile
```

Any help would be greatly appreciated!

Thank you & kind regards,
Richard Leitner

### Furthermore here are additional resources on my problem:
#### full compile log from bitbake:
```
DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Executing shell function do_compile
NOTE: make -j 8
/build/tmp/sysroots/x86_64-linux/usr/bin/cmake -H/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git -B/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build --check-build-system CMakeFiles/Makefile.cmake 0
/build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_progress_start /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/CMakeFiles /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
make -f src/CMakeFiles/tcam.dir/build.make src/CMakeFiles/tcam.dir/depend
make -f src/CMakeFiles/tcam-dfk73.dir/build.make src/CMakeFiles/tcam-dfk73.dir/depend
make -f src/algorithms/CMakeFiles/tcam-algorithms.dir/build.make src/algorithms/CMakeFiles/tcam-algorithms.dir/depend
make[2]: Entering directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
cd /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build && /build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_depends "Unix Makefiles" /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/CMakeFiles/tcam-dfk73.dir/DependInfo.cmake --color=
make[2]: Entering directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
cd /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build && /build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_depends "Unix Makefiles" /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src/algorithms /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/algorithms /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/algorithms/CMakeFiles/tcam-algorithms.dir/DependInfo.cmake --color=
make[2]: Entering directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
cd /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build && /build/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_depends "Unix Makefiles" /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/git/src /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src /build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build/src/CMakeFiles/tcam.dir/DependInfo.cmake --color=
Scanning dependencies of target tcam-algorithms
make[2]: Leaving directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
make -f src/algorithms/CMakeFiles/tcam-algorithms.dir/build.make src/algorithms/CMakeFiles/tcam-algorithms.dir/build
make[2]: Entering directory '/build/tmp/work/cortexa9hf-vfp-neon-my-distro-linux-gnueabi/gst-plugins-tis-1.0/git-r0/build'
[... cut ...]
HOMEPAGE = "http://www.theimagingsource.com/"
LICENSE = "Apache-2.0"
DEPENDS = "gstreamer1.0 gobject-introspection-stub libtinyxml libgirepository"
PR = "r0"

inherit pkgconfig cmake
SRC_URI = "\
	git://git@github.com/TheImagingSource/tiscamera.git;protocol=ssh;branch=master;name=git \
	"
SRCREV_git = "44f02d1b2b3937abd07f6911a1b1b36a0ad3f8c9"
LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"
EXTRA_OECMAKE = "-DBUILD_ARAVIS=OFF -DBUILD_GST_1_0=ON -DBUILD_TOOLS=OFF -DBUILD_V4L2=ON"
S = "${WORKDIR}/git"
```

TIS-Edgar, 2017-03-15:
Hi,

That is a rather weird build error. From what i can tell, the introspection descriptions are not correctly generated. If you compare the folder content you have to a successful build you will notice the two additional files.

    CMakeFiles
    cmake_install.cmake
    libtcamprop.so
    libtcamprop.so.0.1
    Makefile
    Tcam-0.1.gir
    Tcam-0.1.typelib

If you look at you output at ```[ 79%] Generating Tcam-0.1.gir``` you will see, that it attempts to generate the .gir file. The .typelib is not even mentioned.

The output for the gobject folder should look like:
```
make[2]: Entering directory '/home/edt/tmp/tiscamera/build'
[ 69%] Building C object src/gobject/CMakeFiles/tcamprop.dir/tcamprop.c.o
cd /home/edt/tmp/tiscamera/build/src/gobject && /usr/bin/cc  -DHAVE_USB -Dtcamprop_EXPORTS -I/home/edt/tmp/tiscamera/build/src/gobject -I/home/edt/tmp/tiscamera/src/gobject -I/home/edt/tmp/tiscamera/build -I/home/edt/tmp/tiscamera/src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include  -pthread -fPIC   -o CMakeFiles/tcamprop.dir/tcamprop.c.o   -c /home/edt/tmp/tiscamera/src/gobject/tcamprop.c
[ 71%] Linking C shared library libtcamprop.so
cd /home/edt/tmp/tiscamera/build/src/gobject && /usr/bin/cmake -E cmake_link_script CMakeFiles/tcamprop.dir/link.txt --verbose=1
/usr/bin/cc  -fPIC  -pthread  -shared -Wl,-soname,libtcamprop.so.0.1 -o libtcamprop.so.0.1 CMakeFiles/tcamprop.dir/tcamprop.c.o -Wl,-rpath,/home/edt/tmp/tiscamera/build/src: ../libtcam.so.0.8.0 -lglib-2.0 -ltinyxml 
cd /home/edt/tmp/tiscamera/build/src/gobject && /usr/bin/cmake -E cmake_symlink_library libtcamprop.so.0.1 libtcamprop.so.0.1 libtcamprop.so
make[2]: Leaving directory '/home/edt/tmp/tiscamera/build'
[ 71%] Built target tcamprop
make -f src/gobject/CMakeFiles/create_gobject.dir/build.make src/gobject/CMakeFiles/create_gobject.dir/depend
make[2]: Entering directory '/home/edt/tmp/tiscamera/build'
cd /home/edt/tmp/tiscamera/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/edt/tmp/tiscamera /home/edt/tmp/tiscamera/src/gobject /home/edt/tmp/tiscamera/build /home/edt/tmp/tiscamera/build/src/gobject /home/edt/tmp/tiscamera/build/src/gobject/CMakeFiles/create_gobject.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/edt/tmp/tiscamera/build'
make -f src/gobject/CMakeFiles/create_gobject.dir/build.make src/gobject/CMakeFiles/create_gobject.dir/build
make[2]: Entering directory '/home/edt/tmp/tiscamera/build'
[ 72%] Generating Tcam-0.1.gir
cd /home/edt/tmp/tiscamera/src/gobject && /usr/bin/g-ir-scanner tcamprop.c tcamprop.h --namespace=Tcam --nsversion=0.1 --warn-all --verbose --include=GObject-2.0 -I/home/edt/tmp/tiscamera/src --pkg=gobject-2.0 --library=tcamprop -L/home/edt/tmp/tiscamera/build/src/gobject --output="/home/edt/tmp/tiscamera/build/src/gobject/Tcam-0.1.gir"
/usr/bin/libtool: line 2459: printf: write error: Bad file descriptor
g-ir-scanner: link: libtool --mode=link --tag=CC x86_64-linux-gnu-gcc -pthread -o /home/edt/tmp/tiscamera/src/gobject/tmp-introspecteatb9h_2/Tcam-0.1 -export-dynamic tmp-introspecteatb9h_2/home/edt/tmp/tiscamera/src/gobject/tmp-introspecteatb9h_2/Tcam-0.1.o -L. -ltcamprop -L/home/edt/tmp/tiscamera/build/src/gobject -rpath /home/edt/tmp/tiscamera/build/src/gobject -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
libtool: link: x86_64-linux-gnu-gcc -pthread -o /home/edt/tmp/tiscamera/src/gobject/tmp-introspecteatb9h_2/Tcam-0.1 tmp-introspecteatb9h_2/home/edt/tmp/tiscamera/src/gobject/tmp-introspecteatb9h_2/Tcam-0.1.o -Wl,--export-dynamic -pthread -Wl,--export-dynamic  -L. -ltcamprop -L/home/edt/tmp/tiscamera/build/src/gobject -lgio-2.0 -lgmodule-2.0 -lgobject-2.0 -lglib-2.0 -pthread -Wl,-rpath -Wl,/home/edt/tmp/tiscamera/build/src/gobject
/usr/bin/libtool: line 2459: printf: write error: Bad file descriptor
[ 74%] Generating Tcam-0.1.typelib
cd /home/edt/tmp/tiscamera/build/src/gobject && /usr/bin/g-ir-compiler --includedir=/home/edt/tmp/tiscamera/build/src/gobject /home/edt/tmp/tiscamera/build/src/gobject/Tcam-0.1.gir -o /home/edt/tmp/tiscamera/build/src/gobject/Tcam-0.1.typelib
make[2]: Leaving directory '/home/edt/tmp/tiscamera/build'
[ 74%] Built target create_gobject
make -f src/algorithms/CMakeFiles/tcam-algorithms.dir/build.make src/algorithms/CMakeFiles/tcam-algorithms.dir/depend
make[2]: Entering directory '/home/edt/tmp/tiscamera/build'
cd /home/edt/tmp/tiscamera/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/edt/tmp/tiscamera /home/edt/tmp/tiscamera/src/algorithms /home/edt/tmp/tiscamera/build /home/edt/tmp/tiscamera/build/src/algorithms /home/edt/tmp/tiscamera/build/src/algorithms/CMakeFiles/tcam-algorithms.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/edt/tmp/tiscamera/build'
make -f src/algorithms/CMakeFiles/tcam-algorithms.dir/build.make src/algorithms/CMakeFiles/tcam-algorithms.dir/build
make[2]: Entering directory '/home/edt/tmp/tiscamera/build'
```

The error you are seeing with libtcamprop no found is very likely due to the fact that the introspection descriptions are implicitly required. Thus the build fails at that point and not sooner. I will look into checking the result of custom commands in cmake to provide better feedback.

Since i have no experience with bitbake or Yocto:
Have previous builds succeeded? If yes what changed?
Do all introspecton tools exists (/usr/bin/g-ir-scanner /usr/bin/r-ir-compiler)

rleitner, 2017-03-15:
Hi,
thanks for your feedback!
Regarding your questions:

**Have previous builds succeeded?**
Only the gstreamer0.10 version. This is my first try to compile the 1.0 version.

**Do all introspecton tools exists?**
```
$ ls -la /usr/bin/g-*
-rwxr-xr-x 1 root root   1874 Oct 22  2014 /usr/bin/g-ir-annotation-tool
-rwxr-xr-x 1 root root 172000 Oct 22  2014 /usr/bin/g-ir-compiler
-rwxr-xr-x 1 root root   1846 Oct 22  2014 /usr/bin/g-ir-doc-tool
-rwxr-xr-x 1 root root  35968 Oct 22  2014 /usr/bin/g-ir-generate
-rwxr-xr-x 1 root root   1857 Oct 22  2014 /usr/bin/g-ir-scanner
```

rleitner, 2017-03-27:
Any updates on this issue?
If not, what would be the appropriate/tested way to cross compile the gst1.0 version of this plugin for ARM?

TIS-Stefan, 2017-03-27:
Hallo Herr Leitner

Unfortunately no progress, since our Yocto experiences are very low. We made successful tests on Raspi Only. Some other tests we did on Odroid. Both of them use Ubuntu based systems.

Stefan

rleitner, 2017-06-26:
**Good news:** I finally managed to compile the gstreamer 1.0 modules via OpenEmbedded/YoctoProject using Poky v2.3 "pyro". *Hoooray!*

For the sake of completeness here's the recipe I used:

```
DESCRIPTION = "The Imaging Source Gstreamer 1.0 Plugins"
HOMEPAGE = "http://www.theimagingsource.com/"
LICENSE = "Apache-2.0"
DEPENDS = "gstreamer1.0 libtinyxml eudev libusb pkgconfig"
PR = "r0"

inherit pkgconfig cmake gobject-introspection

SRC_URI = "git://git@github.com/TheImagingSource/tiscamera.git;protocol=ssh;branch=master;name=git"

SRCREV_git = "0cc6781f6c7b2605a50ccdd6bb02a4415229697b"

LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"

# tiscamera build configuration
EXTRA_OECMAKE = "\
	-DBUILD_ARAVIS=OFF \
	-DBUILD_GST_1_0=ON \
	-DBUILD_TOOLS=OFF \
	-DBUILD_V4L2=ON \
	"

# workaround for invalid/not detected resources
EXTRA_OECMAKE += "\
	-DGSTREAMER_VIDEO_INCLUDE_DIRS=${WORKDIR}/recipe-sysroot/usr/include \
	-DGSTREAMER_VIDEO_LIBRARIES=${WORKDIR}/recipe-sysroot/usr/lib \
	-DINTROSPECTION_SCANNER=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-scanner-wrapper \
	-DINTROSPECTION_COMPILER=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-compiler-wrapper \
	-DINTROSPECTION_GENERATE=${WORKDIR}/recipe-sysroot/usr/bin/g-ir-generate \
	"

S = "${WORKDIR}/git"

FILES_${PN} = "\
	/usr/lib/tcam-0/libtcam-v4l2.so.0.8.0 \
	/usr/lib/libt* \
	/usr/lib/gstreamer-1.0/libgst* \
	/usr/lib/udev/rules.d/ \
	/usr/lib/girepository-1.0/ \
	/usr/share/uvcdynctrl/ \
	/usr/share/gir-1.0/ \
	"
```

closed 2017-06-26 by rleitner