TheImagingSource/tiscamera #86: EEPROM too small -> no update possible?

AndrHe, 2017-03-22:
Hi,

I tried to use my DFK 72AUC02 with an UBUNTU distribution, but the update 
prompted that there is an issue considering the EEPROM, which is too low to get the firmware (16..kB vs. 19...).

How can I manage that?

Thanks

TIS-Stefan, 2017-03-22:
Unfortunately you did not write, which firmware file you want to write in the camera. Thus I recommend to use dmk72uc02_129.euvc.

Stefan

AndrHe, 2017-03-22:
Hi Stefan,

I tried the dfk72uc02_3012.euvc, which is mentioned in the WIKI.

AndrHe, 2017-03-22:
The update with dfk72uc02_129.euvc worked (the Windows version?), but the dmesg doesn't show 
uvcvideo: Found UVC 1.00 device

AndrHe, 2017-03-22:
I tried your recommendation, but the dmesg shows 

[27658.306155] usb 1-3: new high-speed USB device number 5 using ehci-pci
[27658.707499] usb 1-3: New USB device found, idVendor=199e, idProduct=8207
[27658.707512] usb 1-3: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[27658.707519] usb 1-3: Product: DFK 72AUC02
[27658.707526] usb 1-3: Manufacturer: The Imaging Source Europe GmbH
[27658.707531] usb 1-3: SerialNumber: 06110418

but not uvcvideo: Found UVC 1.00 device 
as mentioned by TIS-Edgar

 TIS-Edgar commented on 7 Sep 2015

Hi,

Let's start with getting your devices working properly.

What does the command 'dmesg' say when you connect a camera?

Ideally it would be something like:

[ 3634.269253] usb 3-14: new high-speed USB device number 5 using xhci_hcd
[ 3635.996841] usb 3-14: config index 0 descriptor too short (expected 392, got 308)
[ 3635.997125] usb 3-14: New USB device found, idVendor=199e, idProduct=8302
[ 3635.997127] usb 3-14: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[ 3635.997128] usb 3-14: Product: DMK 22BUC03
[ 3635.997129] usb 3-14: Manufacturer: The Imaging Source Europe GmbH
[ 3635.997130] usb 3-14: SerialNumber: 21110463
[ 3635.997497] uvcvideo: Found UVC 1.00 device DMK 22BUC03 (199e:8302)
[ 3636.932874] input: DMK 22BUC03 as /devices/pci0000:00/0000:00:14.0/usb3/3-14/3-14:1.0/input/input18

TIS-Stefan, 2017-03-22:
My fault sorry. Use "dmk72uc02_162_uvc.euvc" That is the file for your camera. It makes a mono, but UVC compatible camera. Debayering should be done in software.

Stefan

AndrHe, 2017-03-22:
Thank you

that worked for me

Andreas

closed 2017-03-23 by AndrHe