TheImagingSource/tiscamera #87: Camera not detected (MD232371)

tim1mw, 2017-03-23:
I've successfully installed the tiscamera software and have the gige-daemon running, but my camera is never detected. If I use wireshark, I can see that the gige-daemon is sending out GVCP broadcasts, I can also see UDP packets coming from the camera, but calling

camera-ip-conf-cli list  or
tcam-ctrl  -l

Does not find any devices. Am I missing something? The discovery packet looks like this:
7	7.568323283	192.168.1.3	255.255.255.255	GVCP	50	> DISCOVERY_CMD 
While I'm seeing this from the camera in wireshark:
1	0.000000000	192.168.1.47	234.56.78.9	UDP	92	23456 → 4567  Len=50

I have a Fredi wifi camera, which identifies itself as "MD232371" in the windows software client.

Any help gratefully received!

TIS-Stefan, 2017-03-24:
Hello

What are the ip and submask settings of your network controller?

Stefn

tim1mw, 2017-03-24:
Ok, thanks. I don't think I stand much chance of getting a sensible reply from the manufacturer. 

Is this software intended to be a "standard" library for all devices which support the GVCP protocol, or is it hard-wired for just your own devices? I had assumed the former.

I can understand that you have no incentive to support other manufacturers devices, that's perfectly fair, but if this is simply a case of adding device ID's to the sourse code so that my device is recognised as a compatible camera, where would I look to make this modification?

tim1mw, 2017-03-24:
Second thought, when you say the subnet mask is weird, are you referring to the packet sent to 234.56.78.9? To be honest I have no idea what's causing this, that IP isn't in any of the settings visible in the windows software. Wireshark describes this packet as IPv4mcast, I suspect it's multicast video data, since this seems to be the most numerous packet type.

TIS-Stefan, 2017-03-27:
Hi

First of all, the setup for the IP addresses camera-ip-conf-cli and the config program tcam-ctrl are made for our cameras. But since they are available in source code, you can see, where the limitations are. So you can make the modification on your own. But it is not guaranteed, this will work, in case the manufacturer of your camera makes something different. 

Next I recommend to download and install the Aravis 0.4 library. This  implements the GigE Vision protocol, we use in our tcambin library too. This package comes also with a viewer "arv-viewer", which is nice to check, whether you get images at all.  Please refer to https://github.com/TheImagingSource/tiscamera at "Optional Aravis Installation".

I spend a few minutes searching your camera in the internet. I found it, but I did not go any information about the protocol it uses. So I am not sure, whether it supports GigE Vision protocol at all.

So it is a good idea to get in contact with the manufacturer of the camera. I am not able to provide support for this model.

Stefan

closed 2019-07-04 by TIS-Edgar