TheImagingSource/tiscamera #90: Running DFM 27UP006-ML on Beaglebone Black

justynb, 2017-04-13:
I'm trying to use the DFM 27UP006-ML camera on a Beaglebone Black, but can't get the example/c/ applications to work.

`Distributor ID:	Debian
Description:	Debian GNU/Linux 8.6 (jessie)
Release:	8.6
Codename:	jessie

debian@beaglebone:~$ uname -a
Linux beaglebone 4.4.30-ti-r64 #1 SMP Fri Nov 4 21:23:33 UTC 2016 armv7l GNU/Linux

debian@beaglebone:~$ dmesg | grep uvc
[   39.886228] uvcvideo: Found UVC 1.10 device DFM 27UP006-ML (199e:8616)

debian@beaglebone:/dev/v4l$ tree
.
├── by-id
│   └── usb-The_Imaging_Source_Europe_GmbH_DFM_27UP006-ML_10710456-video-index0 -> ../../video0
└── by-path
    └── platform-musb-hdrc.1.auto-usb-0:1:1.0-video-index0 -> ../../video0

2 directories, 2 files

debian@beaglebone:/dev/v4l$ lsusb
Bus 001 Device 011: ID 199e:8616 The Imaging Source Europe GmbH 
`

Everything looks good: it gets identified by the uvcvideo driver, it gets recognized as a USB device, and it appears in /dev/v4l.  When I try to run tcam_properties, I get the following warning:

` ** (tcam_properties:6492): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files`

Then when I hit the "start" button on the GUI, I get the following:

`Starting stream...
Could not find src.

** (tcam_properties:6492): CRITICAL **: tcam_prop_get_tcam_property_names: assertion 'self != NULL' failed
No properties found.`

From cursory Google'ing, it seems like the warning is some GTK thing.
The more urgent part is the "Could not find src".  It looks like get_element("src") returns NULL.  I'll look into it further, but is there something obvious I'm missing?  Am I missing a gstreamer dependency somewhere?

TIS-Stefan, 2017-04-13:
Hello

I must admit, I have no experiences with Beaglebone, except one: It was too weak, even for slow 640x480 video. Thus I can not really help.
In order whether your get a video from the camera, you may launch the standard pipeline:
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
on the command line. 

However,  this even does not work on a PI 3 without problems. So you may use

gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=640, height=480, framerate=7/1  ! videoconvert ! ximagesink

I am not sure about the frame rate. The tcam-ctrl program will list the correct ones.

Stefan

justynbell, 2017-04-13:
Really all I care about is getting 1 image from the camera from an external trigger in c. So I'm trying to get the c example with the uvc extension for external triggering working.

If there's a more direct method, like if you could help me set up a gstreamer pipeline to do that, I really don't need to use the gtk GUI application.

TIS-Stefan, 2017-04-18:
[softwaretrigger.zip](https://github.com/TheImagingSource/tiscamera/files/928573/softwaretrigger.zip)

The sample above is a c++ sample based on the v4ldevice.cpp standard program. With this, you can grab one image. It uses OpenCV for debayering, but you can remove that, if bayer raw data from the sensor is ok for you.

If you have a hardware trigger, you can remove the code regarding the software trigger. 

But imho the Beaglebone is too weak for your camera. You will loose image data packets.

Stefan

justynb, 2017-05-10:
I'm not familiar with how the images are transported under the hood, it's a little surprising that packets could be "lost".  I would assume the transport layer would be something like TCP where packets are guaranteed to arrive, maybe just much later in time.

I know this is outside of the scope of the ticket, or even TIS, but is there an embedded platform you guys might recommend?

TIS-Stefan, 2017-05-11:
Oh no. UVC does not guarantee that. Also the DMA transfer from USB controller into memory can be blocked. Packet resend in UVC is not specified.

We made some good experiences with Odroid, since they have USB 3 on board. Also the NVidia Tegra K1 worked fine, but that is somewhat expensive.

closed 2017-05-11 by TIS-Stefan