TheImagingSource/tiscamera #91: DMK 31AU03   No devices found  camera not work

DragonSmile, 2017-04-14:
Hello
I follow the README.md at [https://github.com/TheImagingSource/tiscamera]
I execute **make install **  and deamon work well .

when I execute **tcam-ctrl -l**  it says **No devices found**, the carmera does't work !

My camera's mode is DMK 31AU03 .

The OS  i used is ubunu16.04.
**By the way,i run the processes on the VMware rather than a real machine. Does it make a difference?
I check the parameters of the camera on the official website，it also uses USB2.0 ,but my usb2.0 disk work well on the VMware .**
 
How to make my carmera work up? 
Thank you very very much!  ~^_^~

TIS-Stefan, 2017-04-18:
Hello 

Please have a look at https://github.com/TheImagingSource/tiscamera/wiki/UDEV-Rules

Stefan

DragonSmile, 2017-04-18:
The carmera  still cant not  work,but I have some gains.

My carmera's infomation should be **Imaging Source Europe DMx 31AU03** ,but when I execute command **lsusb**,it show me **Bus 001 Device 004: ID 199e:8101 The Imaging Source Europe GmbH DFx 21BU04 Camera**

Does it the key of the problem?

TIS-Stefan, 2017-04-18:
Nice. Please let me know the serial number of your camera.

Stefan

DragonSmile, 2017-04-19:
serial  # 43010609

TIS-Stefan, 2017-04-19:
The camera is a DMK 31AU03. Thus I wonder, where your that 21BU04 from.  What is dmesg output? 
Did you install the UDEV rules?

Stefan

DragonSmile, 2017-04-20:
Hi
21BU04 from the command **lsusb**,By the way,I have only one camera (DMK 31AU03).
> root@ubuntu:/home/xjm/Desktop/tiscamera/examples/c# lsusb
Bus 001 Device 002: ID 199e:8101 The Imaging Source Europe GmbH DFx 21BU04 Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

I use **dmesg** ,it looks like a correct result

> 1756 [   27.627646] e1000: ens33 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
1757 [   51.578646] usb 1-1: new high-speed USB device number 2 using ehci-pci
1758 [   51.930025] usb 1-1: New USB device found, idVendor=199e, idProduct=8101
1759 [   51.930030] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
1760 [   51.930033] usb 1-1: Product: DMx 31AU03
1761 [   51.930035] usb 1-1: Manufacturer: The Imaging Source Europe GmbH
1762 [   51.930037] usb 1-1: SerialNumber: 43010609
1763 [   53.161325] media: Linux media interface: v0.10
1764 [   53.179215] Linux video capture interface: v2.00                                                                                       
1765 [   53.229484] usbcore: registered new interface driver uvcvideo
1766 [   53.229488] USB Video Class driver (1.1.1)

I am sure I have installed the  UDEV rules.
Following  these command 

> cd tiscamera
sudo cp data/udev/80-theimagingsource-cameras.rules /etc/udev/rules.d/
sudo service udev restart

Now, I don't really know wether camera driver does't work or driver work well but the way I test camera  is wrong.
The reasons I think my camera can't work are
①use **tcam-ctl -l** and it show "No devices"
②I execute ../example/c/tcam_pipeline , I get  a blank winform and can't see images.

I think it should be the driver's problem,and is there any other solutions except UDEV rules?

TIS-Stefan, 2017-04-20:
Hello

Does the camera run in Windows?

Stefan

DragonSmile, 2017-04-20:
Yes,it work well in Windows.

TIS-Stefan, 2017-04-20:
I have to ask my Linux experts on next Monday. Sorry for that.

DragonSmile, 2017-04-20:
OK,Thanks for your patient guidances! ^_^

TIS-Edgar, 2017-05-15:
Sorry for the late reply, I was only now made aware of this ticket...

The UVC support for our older cameras is somewhat "difficult", as it was never really planned.
The 31 cameras will say they are 21s. That is to be expected. 

Since your kernel is >= 4.4 your uvc driver should be more than recent enough.

At this point there are two possible points of failure:  
1. VMWare does not correctly pass the camera through.
    To test through it would be simplest to check how the camera acts on a native Linux system.
2. The udev rules are not applied correctly
    To debug this I would need the output of several commands. 
    I am unsure where to start with this, so I have to throw a wide net.
    Execute 'sudo udevadm monitor' and connect your camera.
    You will see something along the lines of 
```
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[27064.749628] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11 (usb)
KERNEL[27065.874870] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0 (usb)
KERNEL[27065.875347] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/video4linux/video0 (video4linux)
KERNEL[27065.875363] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/media0 (media)
KERNEL[27065.875416] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.1 (usb)
UDEV  [27065.883286] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11 (usb)
UDEV  [27065.884553] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0 (usb)
UDEV  [27065.884809] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.1 (usb)
UDEV  [27065.885297] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/media0 (media)
UDEV  [27065.894542] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/video4linux/video0 (video4linux)

```

What I need is the output of 'sudo udevadm test <DEV>' whereas <DEV> corresponds to the different device strings from the monitor command (like /devices/pci0000:00/0000:00:14.0/usb3/3-11). This will show if the rules are actually applied.

"High level" tools, such as our examples or tcam-crtrl will show nothing, because how could they, when the camera fails at such an low level.

closed 2017-08-10 by TIS-Arne