TheImagingSource/tiscamera #92: Save frames with slower framerate in UVC mode using gstreamer

JLMARIN, 2017-04-21:
Hi,

I've been playing around with a DMM 42BUC03-ML camera using the latest firmware (version 198) and with UVC mode on. I'm using gstreamer in bash scripts to interface with the camera.

So far I've been able to configure the camera using v4l2-ctl, and using gstreamer (1.0) I've been able to get a live feed and extract and save a single frame. But the next step was to save multiple frames. If I run this command I can save multiple frames using one of the camera's native framerates (in this case, at this resolution, the options are 30, 25, 20 and 10):

`gst-launch-1.0 -e tcamsrc serial=${SERIAL} ! video/x-raw,format=GRAY8,width=1280,height=960,framerate=10/1 ! videoconvert ! jpegenc ! multifilesink location=./test/frame_%04d.jpg`

The above command will save 10 frames per second. Next, I tried using `videorate` to decrease the framerate and save image files at a lower rate:

`gst-launch-1.0 -e tcamsrc serial=${SERIAL} \
! video/x-raw,format=GRAY8,width=1280,height=960,framerate=10/1 \
! videorate ! video/x-raw,format=GRAY8,framerate=1/1 \
! videoconvert ! jpegenc ! multifilesink location=./test/frame_%04d.jpg`

`videorate` has worked for me in the past with other UVC cameras, but it does not work with the TIS camera. I was hoping someone could help me on how to save multiple image files from the TIS camera using gstreamer at a custom rate (lower than the camera's native framerate).

Finally, now that I'm here, I also have a couple of extra questions:
1) What's the difference between using tcambin and tcamsrc in gstreamer. With tcambin there is no need to specify the serial apparently, but how can I communicate with more than one camera.
2) I suppose I can use tcamsrc with the serial of each camera to talk to multiple cameras and it should be fine. Just to confirm.

Thanks.

TIS-Stefan, 2017-04-24:
Hello

its a time stamp bug in the tcamsrc. Sorry for that. If the tcamsrc is exchanged against v4l2src your pipeline will work as expected. It seems the first four frames are too fast, but the following are in time.

The tcambin employs the tcamsrc and if needed the other modules for automatic exposure, white balance and focus.

JLMARIN, 2017-04-24:
I see. Is there any benefit in using tcamsrc or tcambin compared to v4l2src?

TIS-Stefan, 2017-04-25:
If you use a monochrome camera, there is not as much benefit. The tcamsrc/tcambin is created to make a unified interface for USB and GigE cameras. It is also made to make the use of multi camera systems simpler by identifying the cameras by serial number.

TIS-Arne, 2017-08-18:
This issue already got fixed.

closed 2017-08-18 by TIS-Arne