TheImagingSource/tiscamera #94: How to close the auto Gain?

xiat5, 2017-05-15:
The camera i using is DFx 72AUC02. I've found that the camera automatically adjusts brightness as the environment changes, after canceling the automatic exposure.And I found that the gain was automatically increased in the dark environment.

How to close the auto Gain?

help me!

TIS-Edgar, 2017-05-15:
What does your pipeline look like? And what programming language do you use?

If you manually set up your elements:

Set the flag 'auto-gain' to false for tcamautoexposure.

If you use the tcambin it is easier to disable auto-gain through the tcamprop interface.

In c this would be done by:
```c
GValue val = {};
g_value_init(&val, G_TYPE_BOOLEAN);
g_value_set_boolean(&val, FALSE);
tcam_prop_set_tcam_property(tcambin, "Gain Auto", &val);
```

xiat5, 2017-05-15:
Hi Edgar,
thanks for your answer. i add a line in "./tiscamera/examples/showcamera/usbcamera.cpp".
`      case ExposureAuto:
	  return SetAuto("autoexposure", "auto-exposure", value == 1);
	  break;
`
chenged by:
`      case ExposureAuto:
	  SetAuto("autoexposure", "auto-gain", 0);
	  return SetAuto("autoexposure", "auto-exposure", value == 1);
	  break;
`

closed 2017-05-15 by xiat5