TheImagingSource/tiscamera #95: Software or Hardware Trigger

haseeb-uob, 2017-05-24:
Hello,

I am working with three GigE zooming cameras (DFK Z12GP031) connected via a GigE switch to the PC. I have a question regarding the triggering of cameras for synchronization. What would be the best approach for triggering (Hardware or Software)?

TIS-Stefan, 2017-05-26:
I would use a hardware trigger.

haseeb-uob, 2017-05-26:
Thanks. 
Using Aravis I tried to activate the hardware triggering mode and I am getting error: "Could not create image stream"

arv_device_set_string_feature_value(Device, "TriggerMode", "On");
arv_device_set_string_feature_value(Device, "TriggerSource", "Line1");
arv_device_set_string_feature_value(Device, "TriggerActivation", "RisingEdge");
arv_device_set_string_feature_value(Device, "AcquisitionMode", "Continuous");

For triggering, I am generating PWM using Arduino at specific FPS. What can be the reason?

TIS-Stefan, 2017-05-26:
I must admit, I am not able to guess why you get "Could not create image stream" since I do not know, how you open the camera, set video format and frame rate. Does the camera show a live stream whithout handling the trigger parameters?

Stefan

haseeb-uob, 2017-05-28:
yes I am able to get live stream without handling the trigger parameters.

TIS-Stefan, 2017-05-29:
Ok. If the trigger is enable, then you should get one image, if a trigger pulse occurs.  The trigger signal must be a voltage change from 0V to 5V. I hope your Arduino can provide that.

Stefan

haseeb-uob, 2017-05-29:
Yes, arduino digital pin gives output of 5 volts. I am not sure either the problem is in hardware or software. 
Here is how I calculate the triggering pulse. 
T_ON = 12 us.
T_OFF = 1/FPS + Exposure_time + Delay (Td)
T_OFF = (1/15)us + 70000us + 1.4us
Is it correct?

haseeb-uob, 2017-05-29:
Additionally, Since I am using a GigE switch for POE. I used only trig+ (PWM) and trig- (Ground) pins for triggering. I hope I don't need any other source for the optocoupler.

TIS-Stefan, 2017-05-29:
I must admit, I do not know enough about Arduino to justify, whether your code works. 

And no, you do not need an extra power for trigger. The trigger is "powered" by the wires. You could use a battery e.g. 4.5V or 9V for triggering, just to see, whether you get an image.

Stefan

closed 2017-05-29 by haseeb-uob

sxr3455, 2019-09-09:
@haseeb-uob Can you share some thoughts on how did you solve this? I'm trying to do the same, I am using the external trigger python example.

TIS-Stefan, 2019-09-10:
I must admit, there no solution in this issue, therefore, I have no thoughts. 
However, I wonder, why you got the error message "Could not create image stream", if you set the trigger parameters. From our conversation I see, that if you do not set the trigger parameters, you get a nice image stream.

Do you have some source code or better a very small sample project, that shows, what you do and reproduces the issue? Maybe I can find something in the Aravis programming.

Stefan