TheImagingSource/tiscamera #96: DFK AFUJ003-M12

christofschroeter, 2017-06-12:
I am trying to get a DFK AFUJ003-M12 running on Linux Mint 18.

Following the Readme I installed dependencies and compiled the code in the main dir (without ARAVIS support). Then trying with gst-launch

```
> gst-launch-1.0 tcambin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: failed to activate bufferpool
Additional debug info:
gstbasetransform.c(1725): default_prepare_output_buffer (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
failed to activate bufferpool
Execution ended after 0:00:00.716662993
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

Next I tried to compile the example showcamera. Unfortunately it still assumes gstreamer-0.10, so needed a few adjustments, but I got it to compile. When starting it, it shows the correct camera model and serial. Command line shows resolutions and properties. But when I push "Start", there is no image, and this output:

```
Pipeline build start
The pipeline element "tisvideobufferfilter", "bufferfilter" could not be created.

(showcamera:22854): GStreamer-WARNING **: 0.10-style raw video caps are being created. Should be video/x-raw,format=(string).. now.
Module "autoexposure" is null.
Module "bufferfilter" is null.

** (showcamera:22854): CRITICAL **: Unable to link source pipeline to autoexposure
Module "image_sink" is null.

(showcamera:22854): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Module "image_sink" is null.

(showcamera:22854): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(showcamera:22854): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Playing
```

Any idea on what I am doing wrong?

TIS-Edgar, 2017-06-14:
Hi,

Please excuse the late reply. 
What hardware are you using?
Do tools like qv4l2 or guvcview allow you to interact with the camera?
Could you give me the output of the following command:

    gst-launch-1.0 --gst-debug=tcam*:5 tcambin ! videoconvert ! autovideosink

Gstreamer-0.10 is officially not supported. The showcamera will be removed with the upcoming release.

christofschroeter, 2017-06-15:
I am testing the camera on a standard desktop PC, connected to a USB3 port. If you need more specific info, please let me know which details could be useful.

I tried qv4l2 and guvcviewer now, both work (I had to reconnect the camera first).
Still no luck with gstreamer though. I attached a file with the debug output you asked for.

[output.txt](https://github.com/TheImagingSource/tiscamera/files/1076930/output.txt)

TIS-Edgar, 2017-06-15:
Ok, the camera works as expected.
We are dealing with a gstreamer problem.
From your output we can see that the tcamsrc delivers images but videoconvert fails at activating the bufferpool. I have not yet encountered this message and did not find anything related to it, so I have to somewhat guess. We have to use try and error for this.
Could you execute these commandlines for me and give me the output/tell if they work:

    gst-launch-1.0 --gst-debug-level=3 tcambin ! video/x-raw,format=GREY8 ! videoconvert ! ximagesink

    gst-launch-1.0 --gst-debug-level=3 tcambin ! video/x-raw,format=GREY8,framerate=15/1,width=640,height=480 ! videoconvert ! ximagesink

    gst-launch-1.0 --gst-debug-level=3 v4l2src ! bayer2rgb ! videoconvert ! ximagesink

    gst-launch-1.0 --gst-debug-level=3 v4l2src ! videoconvert ! ximagesink

christofschroeter, 2017-06-15:
The first 3 didn't work, the 4th one gives some warnings, but also shows camera images. Full output here:
[output_tests.txt](https://github.com/TheImagingSource/tiscamera/files/1077731/output_tests.txt)

(The error messages on the first and second make me suspect there are errors in the formulation of this pipeline, but my experience with gstreamer is too limited to make a definitive claim.

I also suspect this is rather a gstreamer problem, since meanwhile I managed to get full resolution images from the camera using a cv::VideoCapture. Still, I guess it would be helpful and informative in general to find out the reason for the gstreamer issue.

TIS-Edgar, 2017-06-16:
A mistake on my side. GREY8 != GRAY8 and of course one works while the other doesn't.

    gst-launch-1.0 --gst-debug-level=3 tcambin ! video/x-raw,format=GRAY8 ! videoconvert ! ximagesink

    gst-launch-1.0 --gst-debug-level=3 tcambin ! video/x-raw,format=GRAY8,framerate=15/1,width=640,height=480 ! videoconvert ! ximagesink

Since the third line also causes problems and there are none of our modules involved it is definitely a gstreamer problem. Considering the output of the fourth pipeline (invalid buffer line, etc) there is something weird.

    gst-launch-1.0 --gst-debug=v4l2src:5 v4l2src ! videoconvert ! ximagesink

closed 2017-08-09 by TIS-Arne