TheImagingSource/tiscamera #97: Issue generating Tcam-0.1.gir

SvenHahne, 2017-06-20:
Hi, 
compiling tiscam I get the error: 

`[ 39%] Built target tcamprop
[ 40%] Generating Tcam-0.1.gir
g-ir-scanner: compile: cc -Wno-deprecated-declarations -pthread -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/home/sven/Downloads/tiscamera-master/src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -c -o /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1.o /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1.c
/usr/local/bin/libtool: line 2460: printf: write error: Bad file descriptor
g-ir-scanner: link: libtool --mode=link --tag=CC cc -o /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1 -export-dynamic /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1.o -L. -ltcamprop -L/home/sven/Downloads/tiscamera-master/build/src/gobject -rpath /home/sven/Downloads/tiscamera-master/build/src/gobject -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
libtool: link: cc -o /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1 /home/sven/Downloads/tiscamera-master/src/gobject/tmp-introspectzCBfc_/Tcam-0.1.o -Wl,--export-dynamic -pthread -Wl,--export-dynamic  -L. -ltcamprop -L/home/sven/Downloads/tiscamera-master/build/src/gobject -lgio-2.0 -lgmodule-2.0 -lgobject-2.0 -lglib-2.0 -pthread -Wl,-rpath -Wl,/home/sven/Downloads/tiscamera-master/build/src/gobject
/usr/local/bin/libtool: line 2460: printf: write error: Bad file descriptor
`

compilation finished anyway...

uname -a:
debian 4.9.0-0.bpo.2-amd64 #1 SMP Debian 4.9.18-1~bpo8+1 (2017-04-10) x86_64 GNU/Linux

libglib2.0.0  vers 2.48.0-1~bpo8+1
gstreamer-1.0 vers 1.4.4.2+deb8u1

...everything installed from the debian repository

Any ideas?

cheers,
Sven

TIS-Edgar, 2017-06-22:
Hi,

This is an error from g-ir-scanner and not from us.
It has no influence on the compilation of our libraries and can safely be ignored.
It has also been fixed and should disappear once the Debian repositories are updated.

Here is the commit fixing the issue: https://github.com/GNOME/gobject-introspection/commit/85b1d70b93211415d975deff6050f1401743e72d

closed 2017-08-09 by TIS-Arne

Rasec8401, 2017-08-31:
I am having a similar issue :

[ 40%] Generating Tcam-0.1.gir
g-ir-scanner: compile: cc -Wno-deprecated-declarations -pthread -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/home/cesar/tiscamera/src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -c -o /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.c
g-ir-scanner: link: libtool --mode=link --tag=CC cc -o /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1 -export-dynamic /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o -L. -ltcamprop -L/home/cesar/tiscamera/build/src/gobject -rpath /home/cesar/tiscamera/build/src/gobject -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
/home/cesar/anaconda3/bin/libtool: 1581: /home/cesar/anaconda3/bin/libtool: preserve_args+= --tag CC: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -o: not found
/home/cesar/anaconda3/bin/libtool: 6434: /home/cesar/anaconda3/bin/libtool: compile_command+= -o: not found
/home/cesar/anaconda3/bin/libtool: 6435: /home/cesar/anaconda3/bin/libtool: finalize_command+= -o: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1: not found
/home/cesar/anaconda3/bin/libtool: 5662: /home/cesar/anaconda3/bin/libtool: compile_command+= @OUTPUT@: not found
/home/cesar/anaconda3/bin/libtool: 5663: /home/cesar/anaconda3/bin/libtool: finalize_command+= @OUTPUT@: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -export-dynamic: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o: not found
/home/cesar/anaconda3/bin/libtool: 6312: /home/cesar/anaconda3/bin/libtool: objs+= /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o: not found
/home/cesar/anaconda3/bin/libtool: 6434: /home/cesar/anaconda3/bin/libtool: compile_command+= /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o: not found
/home/cesar/anaconda3/bin/libtool: 6435: /home/cesar/anaconda3/bin/libtool: finalize_command+= /home/cesar/tiscamera/src/gobject/tmp-introspect6z_shN/Tcam-0.1.o: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -L.: not found
/home/cesar/anaconda3/bin/libtool: 6016: /home/cesar/anaconda3/bin/libtool: deplibs+= -L.: not found
/home/cesar/anaconda3/bin/libtool: 6019: /home/cesar/anaconda3/bin/libtool: lib_search_path+= /home/cesar/tiscamera/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -ltcamprop: not found
/home/cesar/anaconda3/bin/libtool: 6077: /home/cesar/anaconda3/bin/libtool: deplibs+= -ltcamprop: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -L/home/cesar/tiscamera/build/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 6016: /home/cesar/anaconda3/bin/libtool: deplibs+= -L/home/cesar/tiscamera/build/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 6019: /home/cesar/anaconda3/bin/libtool: lib_search_path+= /home/cesar/tiscamera/build/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -rpath: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= /home/cesar/tiscamera/build/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 5860: /home/cesar/anaconda3/bin/libtool: rpath+= /home/cesar/tiscamera/build/src/gobject: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -lgio-2.0: not found
/home/cesar/anaconda3/bin/libtool: 6077: /home/cesar/anaconda3/bin/libtool: deplibs+= -lgio-2.0: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -Wl,--export-dynamic: not found
/home/cesar/anaconda3/bin/libtool: 6250: /home/cesar/anaconda3/bin/libtool: arg+= -Wl,--export-dynamic: not found
/home/cesar/anaconda3/bin/libtool: 6251: /home/cesar/anaconda3/bin/libtool: compiler_flags+= -Wl,--export-dynamic: not found
/home/cesar/anaconda3/bin/libtool: 6252: /home/cesar/anaconda3/bin/libtool: linker_flags+= --export-dynamic: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -lgmodule-2.0: not found
/home/cesar/anaconda3/bin/libtool: 6077: /home/cesar/anaconda3/bin/libtool: deplibs+= -lgmodule-2.0: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -pthread: not found
/home/cesar/anaconda3/bin/libtool: 6099: /home/cesar/anaconda3/bin/libtool: compiler_flags+= -pthread: not found
/home/cesar/anaconda3/bin/libtool: 6100: /home/cesar/anaconda3/bin/libtool: compile_command+= -pthread: not found
/home/cesar/anaconda3/bin/libtool: 6101: /home/cesar/anaconda3/bin/libtool: finalize_command+= -pthread: not found
/home/cesar/anaconda3/bin/libtool: 6104: /home/cesar/anaconda3/bin/libtool: new_inherited_linker_flags+= -pthread: not found
/home/cesar/anaconda3/bin/libtool: 5656: /home/cesar/anaconda3/bin/libtool: libtool_args+= -lgobject-2.0: not found
[... cut ...] 0', '-lglib-2.0']' returned non-zero exit status 4
src/gobject/CMakeFiles/create_gobject.dir/build.make:56: recipe for target 'src/gobject/Tcam-0.1.gir' failed
make[2]: *** [src/gobject/Tcam-0.1.gir] Error 1
CMakeFiles/Makefile2:240: recipe for target 'src/gobject/CMakeFiles/create_gobject.dir/all' failed
make[1]: *** [src/gobject/CMakeFiles/create_gobject.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
 

compilation does not finish
Is it an anaconda issue?

TIS-Arne, 2017-08-31:
@Rasec8401 I do not know "anaconda" but the log shows that the libtool binary that resides in a path provided by anaconda fails. Please try using the standard system tools to compile our SDK.

Rasec8401, 2017-08-31:
thanks it worked.