TheImagingSource/tiscamera #98: DFK AFUJ003-M12

makabe0510, 2017-06-21:
I`m trying to use DFK AFUJ003-M12 to use pipeline like

gst-launch-1.0 tcambin ! videoconvert ! autovideosink

Then, I can see the window popping up but that window is white. (I attach the picture of screen shot of that.)

It seems normal log is displayed in terminal.

Using other computer, I did well in this way.  

How can I display movie from DFK AFUJ003-M12 ?

DFK AFUJ003-M12
[image]DFK AFUJ003-M12

TIS-Stefan, 2017-06-22:
Hello

Is the other computer a Linux system too? If yes, what is the difference between them?

Also I would like to know, which computer model you use, on which the camera does not provide a live video.

Thank you in advance!

Stefan

makabe0510, 2017-06-22:
Thank you for your answer.

I use ubuntu 14.04.1 on Intel NUC (NUC7i7BNH).
I did well when I used ubuntu 14.04.5 on Intel NUC(NUC D34010WYK) using same camera.
I think difference between them is minor version of ubuntu.
Is this difference the matter?

TIS-Stefan, 2017-06-23:
May be the difference is, the camera being connected to an USB 2.0 port only? What happens if you try a low resolution video format an a low frame rate? You can use the tcam-ctrl tool in order to query the camera's properties, video formats and frame rates.

Stefan

TIS-Arne, 2017-08-09:
Closing this issue since we need feedback on Stefans questions. Please re-open when neccessary.

closed 2017-08-09 by TIS-Arne