TheImagingSource/tiscamera #99: Error with ./showcamera command

CWEastman, 2017-06-23:
Hi,

I'm trying to get the ./showcamera command to run but I keep getting the error "segmentation fault (core dumped)"   What is a possible cause of this error? 

Advice is much appreciated! Thank You!

TIS-Stefan, 2017-06-26:
Which camera do you use?
If debugging, where does the error occurs?

horstrand, 2017-06-26:
Hi,

I'm having the same issue here. When I run the application (./showcamera) a gui pops up, then I click start, and this is followed by an automatic window closing, and in the terminal it is shown "Segmentation fault".

I'm using a FX10 camera from Specim (with GigE Vision).
Some help would be appreciated.
Thank you!

TIS-Stefan, 2017-06-26:
Hello

showcamera does not handle gige cameras.
Also our software is limited to our camera only.
Therefore, you might check the Aravis Gnome project directly to handle your GigE Vision camera.

Stefan

closed 2017-06-26 by TIS-Stefan

CWEastman, 2017-06-26:
I am using the Imaging Source DMK 22AUC03-F camera. It is a USB2 Camera, not a GigE Vision Camera.  The error occurs as soon as I try to execute the ./showcamera command.  The window pops up like it is going to allow an image to be captured but it fails as soon as any buttons such as "Start" are clicked.

TIS-Stefan, 2017-06-27:
I am sorry, I misinterpreted "FX10 camera from Specim (with GigE Vision)."

I suppose, you use GStreamer 0.10 only. So you may try following pipeline:

`  gst-launch-0.10 v4l2src ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tiscolorize ! tiswhitebalance ! bayer2rgb ! tis_autofocus, auto=1 ! ffmpegcolorspace ! ximagesink `

Maybe you will get the same error, but more detailed.

CWEastman, 2017-06-27:
That's not a problem, misunderstandings happen!

I tried this pipeline and the 'bayer2rgb' comes up with the warning " WARNING: erroneous pipeline: no element "bayer2rgb"  Do you possibly know what is the cause of this?

TIS-Stefan, 2017-06-28:
Hi 

on newer Linux distributions bayer2rgb is no longer part of outdated GStreamer 0.10 packages.

Which Linux Distribution do you use?
Why do you stick to GStreamer 0.10? We already use GStreamer 1.0.