TheImagingSource/tiscamera #100: showcamera program issue

CWEastman, 2017-06-27:
When running the ./showcamera command the program runs and recognizes the camera I have attached (DMK 22AUC03-F) and all the properties can be accessed. However, the screen is black and does not register any images coming through the camera.  There is also the issues of "Error: Could not negotiate format," as well as the error "Error: Device 'dev/video0' is busy." 

Has this happened to anyone before and is there a way to fix this?

CWEastman, 2017-06-27:
[image]

This is what I am seeing when I run the ./showcamera command

CWEastman, 2017-06-27:
[image]

This is what I see when I run the ./showcamera command

TIS-Stefan, 2017-06-28:
Hi again
Try lower frame rates

Try the pipeline

gst-launch-0.10 v4l2src ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tis_autofocus, auto=1 ! ffmpegcolorspace ! ximagesink

The showcamera sample is quite too complex to narrow down issues. 

Now I know, you use an Ubuntu.  So you do not need to answer in issue #99. Which computer model do you use?

However, why do you still use GStreamer 0.10?

CWEastman, 2017-06-28:
Hi again, 

I use a MacBook Pro, and I didn't realize I was supposed to use a different GStreamer until now, I will change that.  I am new to setting up this camera.

TIS-Stefan, 2017-06-28:
MacBook pro with Ubuntu Linux? 

The pipeline above, does it work on the comandline?
`gst-launch-0.10 v4l2src ! video/x-raw-gray ! tisvideobufferfilter ! tis_auto_exposure ! tis_autofocus, auto=1 ! ffmpegcolorspace ! ximagesink`

CWEastman, 2017-06-28:
yes, I am running Ubuntu in a virtual box, sorry I did not mention that earlier.  The pipeline does not fully work, I have gotten the camera to see gray pixilations when I move it around but no clear image.

CWEastman, 2017-06-28:
Never mind, I am back to the camera having the segmentation core dumped error.  I'm not sure what caused it to fail w like this again

TIS-Stefan, 2017-06-28:
If you use "Parallels", you may have some problems with USB forwarding. I had this often on Mac OXs. It seems, you can not receive all image datablocks from the camera. So lets reduce the frame rate:

`gst-launch-0.10 v4l2src ! video/x-raw-gray,width=640,height=480,framerate=15/1 ! tisvideobufferfilter ! tis_auto_exposure ! tis_autofocus, auto=1 ! ffmpegcolorspace ! ximagesink
`

CWEastman, 2017-06-28:
[image]

OK, I used this pipeline but got this error

TIS-Stefan, 2017-06-29:
Seems the device video0 does not exist. So you may disconnect and reconnect the camera.
With 
`ls /dev/vid*`
you will see, whether the camera is "video0" again.

CWEastman, 2017-06-29:
Ok, I've gotten the camera back again after disconnecting it.  It is video0 again and this is what I am seeing.  I can click the start button and it says 'ready' and then I can also click the snap image button and nothing happens. 

[image]

CWEastman, 2017-06-29:
It is still fragile though, occasionally the segmentation error core dumped comes back.

TIS-Stefan, 2017-06-30:
Well, I wanted you to use the last pipeline I had written for you. The showcamera program is far to complex for checking, whether we can get a live video in your environment.

CWEastman, 2017-07-03:
When I used the pipeline you gave me previously, it did not work as well as the ones I have been using.

TIS-Stefan, 2017-07-04:
OK. How to proceed now? 
Other question: Can you use GStreamer 1.0 instead of 0.10? 0.10 is somewhat outdated.
Last point: What exactly do you want to do with the camera and the images? I ask this, because I may can provide a better sample code for you.

Stefan

TIS-Arne, 2017-08-09:
Closing this issue since we need feedback on Stefans last questions. Please re-open when neccessary.

closed 2017-08-09 by TIS-Arne